repetier/RepetierFirmware #349: Pause / Stop are not immediate when printing from SDCard /Host (0.91, 0.92)

luc-github, 2015-01-05:
Pause and Stop commands, when printing from SD Card or Host, have not immediate effect but wait for end of buffer, which can take a while.
On my others printers not using repetier FW, I do not have this behaviour, stop and pause commands are effective immediatly.

Reduce buffer size to reduce response time is a workaround but definitly not great, any chance to improve this ?

Thanks

repetier, 2015-01-05:
No, Stopping before moves are finished is dangerous as it can loose steps on startup or finish. It also would disallow any moves while stopped so we would have to block 50% of the possible functions makeing it useless for most cases. You could even not move the head to a parking position ruining every print.

closed 2015-01-05 by repetier

luc-github, 2015-01-05:
I was more thinking, stop current line, then to bufferize the buffered lines in another location ( they already are in a buffer),  then put buffered lines back to working buffer when pause is done, it allows to do all move in between and do not block anything

repetier, 2015-01-05:
1. There is no memory for that. At least for deltas memory is not existent.
2. Buffered moves have precomputed acceleration/deceleration and can not be simply ignored or interrupted. So if we would try the copy buffer solution we would need some new path planner implementation for correcting exsiting one. All these solutions are much to complicated for my limited time.

luc-github, 2015-01-05:
Ok thanks a lot for your explaination - I understand you have a lot of requests and this one is big work when actually it may not be used or apprecied by a lot people

Thanks for answering

kyrreaa, 2015-01-05:
How about keeping track of path planned buffers expected runtime and not
buffering more than a limit instead of always filling buffer?
This way you have quick response and adequate buffering for both short
complicated moves and longer simple moves that would otherwise generate a
huge delay when pausing.
It would still buffer enough to avoid blk messages too.

repetier, 2015-01-05:
@kyrreaa That would help stopping faster but would make the buffering more useless. One important reason is to have a buffer when you get many small elements or transfer gets slowed down. Since I can not fetch and process new gcode lines when I do not want to fill buffer lets me get poblems with the other cases we added the buffer for reducing print quality. So the price is too higth with that solution I guess. The only way would be buffering without path optimization and then backup buffer on pause where planning stopped. But that is also difficult with timings and I guess quite error prone. Or a second buffer for cached moves and the copy them to the real buffer in time. Would be easiest solution except the in time thing.

luc-github, 2015-01-05:
From another point of view commands launched during pause do not need to be buffered as they are send one by one, and do not need accelleration optimization, they are manually launched and simple : x dir, or y dir, or z dir, or extruder, and only one by one.
So actually just no need any buffer neither optimization for these commands - so current buffer just can be : not used.
But still stay the acceleration of the first path when resume...

luc-github, 2015-01-14:
I am able to stop immediatly when printing from SD Card or Host (by adding custom command)
 
So now thinking about pause and I have a question:
Instead of buffering any command just keep tracking of last GCODE line numbers from SDCard file or send by host.

So when pause command is received it let end current command, and  save coming GCODE line number and then clearing buffer
so when resume : 
-  if printing from host, FW ask host to resend all commands starting from the saved line
-  if printing from SD Card, FW just seek back to saved line in GCODE file and restart from this line

Tracking line numbers would be  lot of less memory consuming than buffering lines/commands - but would be big per your opinion ?

kyrreaa, 2015-01-14:
I see your point.
The parsing and path planning vs the stepping is already using too much
time for this type of printer.
I am working on the new framework for the stm32 which will run FreeRTOS. I
have tested the usb VCP so far and been able to receive over 500kByte/s.
Since it has so much memory, I think I will do the parsing first, then
queue it up to the path-planner which in turn will queue up moves. The
moves are then processed by the step-generator and finally played back by
the DMA system.
By not having a step interrupt, it should be able to run very high speed
moves without slowing down the parsing.
I ofcource hope to support the Repetier binary protocol as well, for even
more improved performance.

CPU runs at 168 MHz with a floatingpoint unitcapable of 1-cycle abs, neg,
add, sub. 3-cycles multiply, multiply-accumulate/subtract,
multiply-accumulate/subtract then negate and finally 14-cycles division and
square root.
I think it will be fun seeing how it scales.

repetier, 2015-01-14:
@kyrreaa Yes, that would be in deed a method for fast interrupts. Also you would get the same pause problem once you have queued up moves you want to move them. And with the speed you have more possibilities then I have even with the due. Especially if you can design hardware to such dma output it gets very good.

@luc-github Storing positions is in deed less memory intensive. But has the drawback of working only with sd cards. And you still get additional problems as you can not simple remove the buffer. You still have to fade out the last moves to allowed speeds without interrupting the move and rewrite the complete gcode handling from input to buffering.  So yes it is possible but also very time intensive and error prone. I guess at least 50 hours or more depending on all the error I would newly introduce. So I'm sorry but immediate pause is simply to heavy to implement, also I know there are possibilities.

kyrreaa, 2015-01-14:
Btw, could a pause be inserted in the already planned queue, causing the
two points on either side of the pause to have their end and start speed
adjusted to 0, to do a accell?
The system would have to look at the queue and see if there is a usefull
spot to insert such a pause ofcource.

repetier, 2015-01-14:
I guess you would need to go from end to beginning and try to do the opposite - meaning find the slowest speed currently possible. Then you would have to prevent using the rest of the buffer or it from being overwritten. With a copy buffer this would be the simplest version. Only the rest can only be used it the extruder is at exactly the same position at start and you also need to backup the printer state parameter like current position after path planning which can change during pause. These are all the things that make it so complicated that I do not dare to do it. I have simply no time for such costly experiments.

The only simple method would be to only buffer up to x seconds and not try to buffer as much as possible. So if we have 4 long moves you could say, ok 1 for printing, one for next, one for start updating again and the next for deceleration. Always assuming deceleration can be done in one move. Then you coud say move 3 is long enough, so I wait until current move is finished. If you have small moves coming you buffer more. That way you still empty buffer but it is finished within a shorter time. Of course you remove the safety you gained with the full buffering.

kyrreaa, 2015-01-14:
Understood.
The only reason this is a problem is really the load the parsing/planning can have on the system when it is also overloaded with high speed stepping.
When a move is very fast, it has to do step interrupt based on the fastest move after quad stepping is estimated right?
If the fastest move is very fast it may still be fastest with quad, and it may be close to the step int per sec limit, where all cpu is occupied by int.
This is why it will be so interesting to see what happens when the system no longer does step interrupts at all. Still a good ways to go before I am there though.
Still working on parsing and stuff before the path planner. I just need something up so I can do the step algo. (A proof of concept would be nice to do before spending loads of time on path planning.)

Anyways, I agree completely, too heavy refactoring or coding to do for a simple pause, but I think limiting the buffering to a reasonable number of commands and seconds could be doable though.
When mine queues up 15 seconds of moves it is starting to get silly...

Kyrre

From: repetier 
Sent: Wednesday, January 14, 2015 4:02 PM
To: repetier/Repetier-Firmware 
Cc: kyrreaa 
Subject: Re: [Repetier-Firmware] Pause / Stop are not immediate when printing from SDCard /Host (0.91, 0.92) (#349)

I guess you would need to go from end to beginning and try to do the opposite - meaning find the slowest speed currently possible. Then you would have to prevent using the rest of the buffer or it from being overwritten. With a copy buffer this would be the simplest version. Only the rest can only be used it the extruder is at exactly the same position at start and you also need to backup the printer state parameter like current position after path planning which can change during pause. These are all the things that make it so complicated that I do not dare to do it. I have simply no time for such costly experiments.

The only simple method would be to only buffer up to x seconds and not try to buffer as much as possible. So if we have 4 long moves you could say, ok 1 for printing, one for next, one for start updating again and the next for deceleration. Always assuming deceleration can be done in one move. Then you coud say move 3 is long enough, so I wait until current move is finished. If you have small moves coming you buffer more. That way you still empty buffer but it is finished within a shorter time. Of course you remove the safety you gained with the full buffering.

DrAnonymous, 2015-01-15:
Can someone give me some more details on what the firmware is doing? I thought the path planning was all done by the slicer software, which took into account accelerations, etc. This was then used to generate the gcode.

From my perspective, if I hit pause, then something is wrong and the ability to resume is just a nice to have. I wouldn't expect a great print if I had stopped things in the middle. The exception being to change filament colors, but really that seems silly. Just get a multi head extruder.

Why couldn't there be a custom pause command that the host sends either as a custom gcode command or as part of the host protocol? When received, the firmware completes the current gcode command and then pauses the print just as though the standard gcode pause had been used?

I'm new to 3d printing, but comfortable with embedded development, so maybe I am ignorantly over simplifying.