repetier/RepetierFirmware #350: Be able to stop and pause host print from printer

luc-github, 2015-01-05:
Is there a way to implement commands to pause and stop from printer when printing from host ? 
My host is in another room and if I am in front of my printer and printing if I have issue or see I need to change filament I need to reset printer to stop it or run to the other room to pause it to be able to fix isssue (clean nozzle, change filament)
Today  I stop by sending a "start" command from printer to host with a customized menu as it is more clean than shutdown printer and avoid to loose usb connection from Host 
But there is no command I know to be able to pause host from printer itself.

Is there a chance to implement such command ? would be great

Thanks,

repetier, 2015-01-05:
Duplicate

closed 2015-01-05 by repetier

luc-github, 2015-01-05:
I did not find it - that is why I posted it - may I know the ID ?
Thanks

repetier, 2015-01-05:
Ok, I thought your othe rposting. Didn't think you meant a host question in firmware.

The host has precise pause, simply add
@pause Message
and host will pause exact there and run host pause commands.

luc-github, 2015-01-05:
well it is a mixed firmware to host command, sorry not being clear.

Here the path example: 
1 - User press Pause on printer LCD menu which send a "pause" command to host. 
2 - Host get "pause" command from serial and then do a pause (same as pressing pause from host or setting @pause in GCODE) and stop sending GCODE waiting for user to press resume from host
3 - Printer is in Pause so user can change filament from printer or do any action.
4 - User finished his action so press Continue from printer LCD menu which send "continue" and host receiving "continue" command and to the resume. 
4b - or  User finished his action so press resume from Host directly
5 - Printer go resume and continue printing

my understanding is @pause is only in GCODE(https://github.com/repetier/Repetier-Host/wiki/Host-Commands) - and not a serial message from printer to host like "start" or "wait" - Am I wrong ? 

Thanks for your time,