repetier/RepetierFirmware #351: Z POSITION DEAD ZONE WHEN HEATED BED ACTIVATED

adamarumjahn, 2015-01-06:
I've noticed this consistent error in both the Mac Host and the PC Host.  If I perform Z-probing with G32 S3 AND have the heated bed activated, the host will auto level however afterward I will not be able to move the nozzle within a certain range (say 0-0.90mm).  Outside of that range is ok and the Z stage will move up and down, however, as soon as I reach this range the Z stage stops moving even though the host displays that it is moving and shows that it is supposedly sending the commands to move the Z-stage.  This stays true if I hit print as it will just move to the closest it can, say 0.90mm for argument sake, and then start printing in the air.

Has anyone else experienced this problem?  or am I the only one?

I have noticed that this behaviour does not happen when I do not use the heated bed, so there is something strange going on that is linking the two.  Again, this has happened in both PC and Mac hosts.  I have spoken to Roland about it and the thinks that it may be a host problem.

The most annoying part is that this bug seems to stick even if I restart my printer and restart the host program.  The only way I can seem to get it to behave normally again is to flash the firmware.  Does anyone know of any other way to get around this in the meantime?

repetier, 2015-01-07:
No I don't think it is a host problem. The question was more if the host sends the moves or not.

What I still find strange is relation to heated bed since there is no line in heated bed code that has influence on moves.

SOme more questions to identify the problem

#define Z_PROBE_BED_DISTANCE 5.0 // Higher than max bed level distance error in mm

does this value determine your dead zone?
After you reset printer can you enter and move in dead zone? After all it stores it in eeprom so you do not need to do it every time. I can move there without problems even with bed turned on.

My guess is that your bed cables influence the endstop maybe crosstalk. Would also explain why it is related to the bed beeing on.

You could change endstop testing in motion.h

    inline void checkEndstops()
    {
        if(isCheckEndstops())
        {
            if(isXNegativeMove() && Printer::isXMinEndstopHit())
                setXMoveFinished();
            if(isYNegativeMove() && Printer::isYMinEndstopHit())
                setYMoveFinished();
            if(isXPositiveMove() && Printer::isXMaxEndstopHit())
                setXMoveFinished();
            if(isYPositiveMove() && Printer::isYMaxEndstopHit())
                setYMoveFinished();
        }
#if FEATURE_Z_PROBE
        if(Printer::isZProbingActive() && isZNegativeMove() && Printer::isZProbeHit())
        {
            setZMoveFinished();
            Printer::stepsRemainingAtZHit = stepsRemaining;
        }
        else
#endif
            // Test Z-Axis every step if necessary, otherwise it could easyly ruin your printer!
            if(isZNegativeMove() && Printer::isZMinEndstopHit())
                setZMoveFinished();
        if(isZPositiveMove() && Printer::isZMaxEndstopHit())
        {
#if MAX_HARDWARE_ENDSTOP_Z
            Printer::stepsRemainingAtZHit = stepsRemaining;
#endif
            setZMoveFinished();
        }
        if(isZPositiveMove() && Printer::isZMaxEndstopHit())
            setZMoveFinished();
    }

add isCheckEndstops() to the z moves and comple with check endstops always disabled.

    inline void checkEndstops()
    {
        if(isCheckEndstops())
        {
            if(isXNegativeMove() && Printer::isXMinEndstopHit())
                setXMoveFinished();
            if(isYNegativeMove() && Printer::isYMinEndstopHit())
                setYMoveFinished();
            if(isXPositiveMove() && Printer::isXMaxEndstopHit())
                setXMoveFinished();
            if(isYPositiveMove() && Printer::isYMaxEndstopHit())
                setYMoveFinished();
        }
#if FEATURE_Z_PROBE
        if(Printer::isZProbingActive() && isZNegativeMove() && Printer::isZProbeHit())
        {
            setZMoveFinished();
            Printer::stepsRemainingAtZHit = stepsRemaining;
        }
        else
#endif
            // Test Z-Axis every step if necessary, otherwise it could easyly ruin your printer!
            if(isZNegativeMove() && Printer::isZMinEndstopHit())
                setZMoveFinished();
        if(isCheckEndstops() && isZPositiveMove() && Printer::isZMaxEndstopHit())
        {
#if MAX_HARDWARE_ENDSTOP_Z
            Printer::stepsRemainingAtZHit = stepsRemaining;
#endif
            setZMoveFinished();
        }
        if(isCheckEndstops() && isZPositiveMove() && Printer::isZMaxEndstopHit())
            setZMoveFinished();
    }

Hope that homing still works so, but it should disable endstop testing during normal moves so no influence from wrong triggered endstops.

adamarumjahn, 2015-02-19:
I believe I have solved this problem.  It has to do with noise or rather electromagnetic interference (EMI).  I had a motor cable and the z-probe cable running in the same harness without any EMI shielding.  The noise causes a false positive when probing.  This also happened when I had another motor cable running in the same cable harness as my Y Endstop.  If anyone else experiences this issue please be sure to keep your motor/power cables and signal/endstop cables separated physically or with EMI sheathing.

closed 2015-02-19 by adamarumjahn