repetier/RepetierFirmware #352: LCD rendering dissapears ocassionaly

mushketer888, 2015-01-13:
Using Due+RADDS
Suddenly LCD data dissapears and doesnt render anything but menu keys are working
what can be the cause
communication also working

repetier, 2015-01-13:
Depends on since when you have the problem. We have reduced some timings recently. but it worked on RADDS with normal delivered cable. Longer cable, crosstalk from WLAN, other cables can cause problems. I have also heard about defect cables.

Does it work when you enable printer and then at some time it starts getting wrong from time to time or does it keep disappeared?

sigma957, 2015-01-13:
A while ago my LCD (RADDS v1.1) started to disappear after some time spent
in printing over USB. Recently I started to observe garbled characters as
well.

repetier, 2015-01-13:
Garbled chars is a result of wrong signals coming to the display. Same might cause a command to disable lcd completely with bad luck. Big question is what causes the signals to get misunderstood knowing that the right signals leave the due. If you get garbled chars try moving the lcd cable around and see if it gets better to test if it's electric noise.

mushketer888, 2015-01-13:
simga957, i have the same problem
when printing over usb after some time lcd dissappears

mushketer888, 2015-01-13:
I think that it dissappears when heatears are on
When in dry run it works fine
Can anyone with this problem also check this?

mushketer888, 2015-01-13:
Keys also stop to work...
So its a software error

mushketer888, 2015-01-13:
Sometimes lcd dissapear when i just put usb cable IN or OUT...

closed 2015-02-14 by mushketer888