repetier/RepetierFirmware #353: Repetier 0.92 - Negative missed steps?

sigma957, 2015-01-22:
Hi,

My setup is: KosselXL (250x320mm) with Repetier 0.92 on Due + RADDS, 0.9 steppers, 16 microsteps, 40mm/s print speed, 60mm/s travel speed. Stepper high delay is set to 2, doubler frequency to 90kHz.

I'm fighting positioning inconsistencies which make my first layer stick poorly or not stick at all.
I have problems calibrating the center position as well (G132) - it always returns different results and therefore cannot print over radius greater than 30mm. Also I cannot level the bed properl (G29/G32), although I have installed 3 FSR sensors connected to JohnSL's FSR endstop.

After deciding to track it down to the root cause I switched on DEBUG_STEPCOUNT. Not surprisingly I saw missed steps reported. What puzzles me is that the firmware reports negative missed steps in the communication stream.

Could someone please explain what this means? I need to understand this better before I continue my search for the root cause.

Thanks in advance.

repetier, 2015-01-23:
Be carefull with missed steps. During homing and calibration we expect missed steps!

It is a software counter that compares expected steps against executed steps. Soit has nothing to do with steppers loosing steps. With z probing we do not know how deep we go, so at some point we trigger probe and stop move thus we get a difference which gets reported!

G132 requires same position/difference in slider positioning when you repeat, which is hard to get. It also depends on how you find the center position, e.g. I use M99 to position sliders at same height which is quite repeatable. After you have done this do not do it again! A small remeining error is not that bad as you compensate it later with G32.

sigma957, 2015-01-23:
Oh, I see, thank you, repetier. I will check if M99 will behave consistently (so far I used a stick to position the sliders as you suggested elsewhere). My biggest problem is that the hotend raises about 0.5mm only in the Y+ direction, which I presumed is due to incorrect Y tower enstop calibration.

I admit I am confused at this point how to continue. If my first layer sticks properly (i.e. small objects) the printer prints nicely and consistently. Can you recommend me how to proceed at this point - repeat the calibration steps but using M99 instead of a stick? I want to be sure I have exhausted all possibilities before I start to examine if the printer symmetrical.

sigma957, 2015-01-23:
One more thing, the firmware reports these negative missed steps during printing as well. On each move I see reported negative missing steps. Is this expected as well?

repetier, 2015-01-23:
Haven't tested missed steps for while since algorithm was working. So I'm not sure if it is ok especially with all the compensation stuff which can really confuse.

M99 is for the sticks meaasurement. It is used to disable one slider motor only so you can move it without the others loosing position. Precision with it should be one full step. M99 is M99 X0 to disable x motor for 10 seconds.

Make sur eif you have Z-correction (distortion correction) enabled, that correction is disabled while calibrating printer. That should only compensate the very rest you did not get right with calibration.

What you do is 
1. distortion off
2. G131/G132 endstop calibration
3. G32 S2 to remove remaining skew. Importnat: Use the 3 points near columns 120° apart. Even with wrong diameter it sets tilt of be dcorrect. Home after G32!
4. Height map and check if it is a dome around center. If so correct horizontal radius and continue with 3.
5. If you see areas which are higher/lower between columns, change angles a bit to minimize. Return 3.
6. Now you can add z correction to get good contact at bottom.

Thats how I always calibarate a delta.

sigma957, 2015-01-24:
Thank you. I am starting the calibration all over again.

Btw, if/when I decide to use Z-correction, do I need to enable babystepping?

repetier, 2015-01-24:
Babystepping is completely independent from anything else, so no.

sigma957, 2015-01-25:
I'm asking because when I tried it out of curiosity (16x16 grid) the
extruder began to move up and down in a fashion that led me to believe that
it is using only grid points and not interpolating between them. Haven't
looked at the code since I'm not interested in z-correction, but that was
the impression. Thanks for clarifying.

Btw, I managed to calibrate the printer finely but only when I modified by
hand the endstop offset. For some reason G132 did not center the extruder
exactly. The nonlinear behaviour disappeared only after my manual fine
tuning. Thanks for all the advices, now I can print PLA on a cold bed.

repetier, 2015-01-25:
You misunderstood G132 if you think it will center the hotend. You need to center it and then G132 learns that position as center. Small but important difference.

Also I found today a bug in z probing. If you started higher then zProbeBedDistance the resulting height was wrong. As a solution update to latest 0.92 or set that value higher then your printer height, so it never gets used.

sigma957, 2015-01-26:
No misunderstanding here, just bad expression on my part. Nevermind.

Yes, I noticed the z-probing behaviour. It was one of the things that made
me believe I have missed steps. So far I used a workaround - G1 Z5, then
G32.

Thank you for the update. However, the servlet now tells me that
Configuration.h was not created with it. Guess I have to configure from
scratch.

repetier, 2015-01-26:
Well if it was not created it is right. It uses a special comment at the bottom to reread the values as it does not understand header files.

sigma957, 2015-01-26:
Yup, I know that and update it by hand on both places. Maybe I've messed up
something. I will investigate this before re-configuring from scratch.

closed 2015-09-26 by repetier