repetier/RepetierFirmware #356: Kossel Delta inverted X or Y axe

Paxy, 2015-01-28:
With both 0.9.1r8 and 0.9.2dev there is an issue with axes. 

Current state:
Homming is done correctly at X,Y,Z 0,0,280 (delta homes at top). 
Printer works correctly with issue that prints are mirrored.
At most of the prints this is not problem, but with precise or prints with text this is huge problem.

Simple solution like reversing direction, at hardware by reversing wires, or at software by enabling Inverse X (or Y) axe, did note solve a problem because after reversing that ax homming on bottom insted of top. This might be solution for non delta printers, but this is not way to solve Deltas problem with reverse ax.

Angle rotation has not resolve issue too. Looking 2D plane, if I set Y ax with positive value to head goes up, X should go with positive value right, with negative left. Right now, If Y positive is up, X positive is left. If X positive is right, Y positive is down.

I should be able to somehow invert just one of the axes (X or Y) to get correct print, but still keep same way of movement for Z ax.

If I manually invert every object in Repetier-host before slicing, print is perfect.

repetier, 2015-01-28:
Hi,

sounds like to need to swap x and y motors. X is the left column, y the 
right. If they are wrong you get a x mirror.

Yours
Roland Littwin

-- 
Mit freundlichen Grüßen/Yours

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

Paxy, 2015-01-29:
You are right, looks like I have not try this combination before. 
Thank you for your help !

closed 2015-01-29 by Paxy