repetier/RepetierFirmware #357: Delta - No Movement

KGardevoir, 2015-02-02:
I am experiencing an issue on the master branch where when the geometry configuration is set to Delta (`DRIVE_SYSTEM = 3`) movement does not occur. It doesn't even appear that the drives ever get enabled. If I set the geometry to Cartesean (`DRIVE_SYSTEM = 0`), the axes enable and move just fine.

Also, if I define `DEBUG_SPLIT`, the drive enable and movement also works as expected. 

I am using an ATMega 2560 with DRV8824 Stepper drivers on a RAMPS 1.4

repetier, 2015-02-02:
Hi,

that's odd as debug split does not change anything. It only outputs var.

Please try 0.92.2 which has many fixes. Not sure what is happening here,
maybe something compiler related. Updating Arduino IDE might work, also
master will not compile with 1.5.8 only 0.92

Yours
Roalnd Littwin

-- 
Mit freundlichen Grüßen

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in
Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

KGardevoir, 2015-02-03:
So I attempted to compile the new 0.92.2 version with Arduino 1.5.8, but arduino complains about global variables taking 99% of dynamic memory. 

I went ahead and uploaded it anyway, but it is highly unstable (probably because of the whole using 99% of the dynamic memory thing, stack collisions and such). Have you tested the Delta Geometry on a ATMega recently?.

repetier, 2015-02-03:
No my delta now runs a due/RADDS. But 99% ram is too much. SHould leave around 900 byte free ram. In that case you need to reduce delta buffers. You can reduce number of subsegmaents per move or number of buffered moves a bit to reduce ram usage. That it also gets stable again.

KGardevoir, 2015-02-04:
Ok, so I got it working. I'm not entirely sure why, but just opening the Configuration file, and then saving it, made the firmware compile correctly. The hotend now doesn't work, but I'll resolve that tomorrow when I have time (extruder stepper isn't being driven, hotend isn't heating up). I'll report back tomorrow.

KGardevoir, 2015-03-13:
Sorry for the extreme lateness, but the extruder being non-funcitonal just turned out that I had the number of extruders set to zero.

closed 2015-03-13 by repetier