repetier/RepetierFirmware #359: i2c custom display config does not work

SamuelBrucksch, 2015-02-08:
When i use 
```
#define FEATURE_CONTROLLER UICONFIG_CONTROLLER
```
and then in the uiconfig.h set up
```
#define UI_DISPLAY_TYPE DISPLAY_I2C
```
Then i get this message:
```
In file included from ui.h:483:0,
                 from Repetier.h:370,
                 from Commands.cpp:22:
uiconfig.h:167:30: error: missing binary operator before token "("
 #define UI_DISPLAY_D5_PIN _BV(1)
                              ^
Repetier.h:375:50: note: in expansion of macro 'UI_DISPLAY_D5_PIN'
   #if (defined(USER_KEY1_PIN) && (USER_KEY1_PIN==UI_DISPLAY_D5_PIN || USER_KEY1_PIN==UI_DISPLAY_D6_PIN || USER_KEY1_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY2_PIN) && (USER_KEY2_PIN==UI_DISPLAY_D5_PIN || USER_KEY2_PIN==UI_DISPLAY_D6_PIN || USER_KEY2_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY3_PIN) && (USER_KEY3_PIN==UI_DISPLAY_D5_PIN || USER_KEY3_PIN==UI_DISPLAY_D6_PIN || USER_KEY3_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY4_PIN) && (USER_KEY4_PIN==UI_DISPLAY_D5_PIN || USER_KEY4_PIN==UI_DISPLAY_D6_PIN || USER_KEY4_PIN==UI_DISPLAY_D7_PIN))
                                                  ^
Fehler beim Übersetzen
```
Which is pretty strange as USER_KEY1_PIN was nowhere defined so it should not get in there. 

If i change it to
```
#ifdef USER_KEY1_PIN
  #if ((USER_KEY1_PIN==UI_DISPLAY_D5_PIN || USER_KEY1_PIN==UI_DISPLAY_D6_PIN || USER_KEY1_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY2_PIN) && (USER_KEY2_PIN==UI_DISPLAY_D5_PIN || USER_KEY2_PIN==UI_DISPLAY_D6_PIN || USER_KEY2_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY3_PIN) && (USER_KEY3_PIN==UI_DISPLAY_D5_PIN || USER_KEY3_PIN==UI_DISPLAY_D6_PIN || USER_KEY3_PIN==UI_DISPLAY_D7_PIN)) || (defined(USER_KEY4_PIN) && (USER_KEY4_PIN==UI_DISPLAY_D5_PIN || USER_KEY4_PIN==UI_DISPLAY_D6_PIN || USER_KEY4_PIN==UI_DISPLAY_D7_PIN))
    #error "You cannot use DISPLAY_D5_PIN, DISPLAY_D6_PIN or DISPLAY_D7_PIN for "User Keys" with character LCD display"
  #endif
#endif
```

I get another error, which is just too much effort to look into myself.

```
In file included from ui.h:483:0,
                 from Repetier.h:370,
                 from ui.cpp:20:
uiconfig.h:166:30: error: missing binary operator before token "("
 #define UI_DISPLAY_D4_PIN _BV(4)
                              ^
ui.cpp:285:5: note: in expansion of macro 'UI_DISPLAY_D4_PIN'
 #if UI_DISPLAY_D4_PIN==1 && UI_DISPLAY_D5_PIN==2 && UI_DISPLAY_D6_PIN==4 && UI_DISPLAY_D7_PIN==8
     ^
uiconfig.h:166:30: error: missing binary operator before token "("
 #define UI_DISPLAY_D4_PIN _BV(4)
                              ^
ui.cpp:308:5: note: in expansion of macro 'UI_DISPLAY_D4_PIN'
 #if UI_DISPLAY_D4_PIN==1 && UI_DISPLAY_D5_PIN==2 && UI_DISPLAY_D6_PIN==4 && UI_DISPLAY_D7_PIN==8
     ^
ui.cpp: In function 'void lcdWriteNibble(uint8_t)':
uiconfig.h:166:32: error: '_BV' was not declared in this scope
 #define UI_DISPLAY_D4_PIN _BV(4)
                                ^
ui.cpp:289:26: note: in expansion of macro 'UI_DISPLAY_D4_PIN'
     uint8_t v=(value & 1?UI_DISPLAY_D4_PIN:0)|(value & 2?UI_DISPLAY_D5_PIN:0)|(value & 4?UI_DISPLAY_D6_PIN:0)|(value & 8?UI_DISPLAY_D7_PIN:0);
                          ^
ui.cpp: In function 'void lcdWriteByte(uint8_t, uint8_t)':
uiconfig.h:159:32: error: '_BV' was not declared in this scope
 #define UI_DISPLAY_RS_PIN _BV(0)
                                ^
ui.cpp:307:23: note: in expansion of macro 'UI_DISPLAY_RS_PIN'
     uint8_t mod = (rs?UI_DISPLAY_RS_PIN:0) | uid.outputMask; // | (UI_DISPLAY_RW_PIN);
                       ^
ui.cpp: In member function 'void UIDisplay::parse(const char*, bool)':
ui.cpp:1187:45: error: 'class TemperatureController' has no member named 'isSensorDefect'
                     if(tempController[eid]->isSensorDefect())
                                             ^
ui.cpp:1192:50: error: 'class TemperatureController' has no member named 'isSensorDecoupled'
                     else if(tempController[eid]->isSensorDecoupled())
                                                  ^
ui.cpp:1198:41: error: 'class TemperatureController' has no member named 'isJammed'
                 if(tempController[eid]->isJammed())
                                         ^
Fehler beim Übersetzen
```
 
Heres my config:
#define FEATURE_CONTROLLER UICONFIG_CONTROLLER
#define UI_DISPLAY_TYPE DISPLAY_I2C
#define UI_HAS_KEYS 0
#define UI_HAS_BACK_KEY 0

Version 0.92 on Arduino 1.5.8

SamuelBrucksch, 2015-02-08:
The errors related to UI_DISPLAY_PIN_XXX could be solved with the macro
```
#define _BV(x) (1 << (x))
```

Still the others remain:
```
ui.cpp: In member function 'void UIDisplay::parse(const char*, bool)':
ui.cpp:1187:45: error: 'class TemperatureController' has no member named 'isSensorDefect'
                     if(tempController[eid]->isSensorDefect())
                                             ^
ui.cpp:1192:50: error: 'class TemperatureController' has no member named 'isSensorDecoupled'
                     else if(tempController[eid]->isSensorDecoupled())
                                                  ^
ui.cpp:1198:41: error: 'class TemperatureController' has no member named 'isJammed'
                 if(tempController[eid]->isJammed())
                                         ^
Fehler beim Übersetzen
```

SamuelBrucksch, 2015-02-08:
I used a fresh copy of repetier 0.92 and now it seems to work. Which is strange, as the version before was clean as well. still the _BV macro needs to be added for arduino due.

repetier, 2015-02-08:
I tested it and using VIKI controller I first got your errors for due. Adding _BV macro which did come from some avr includes it also worked without compilation errors. So next update will have the macro.

Some of your errors above like
ui.cpp:1198:41: error: 'class TemperatureController' has no member named 'isJammed'
                 if(tempController[eid]->isJammed())

correspond to an update I did a few hours ago.  Not sure how you could get parts using them and missing parts defining them from same commit. But seems to be ok now, so I close the issue.

closed 2015-02-08 by repetier