repetier/RepetierFirmware #363: megatronics v2.0 and repetier problems

ta1dr, 2015-02-14:
Hi All
I have megatronics board v2.0 it is new,  load repetier firmware  and make all settings correct
X and Y axes only move when click the x home button and y home button on repetier host program also
not move correct when printting...

I am working all day this problems (load merlin all ok but I dont use merlin)  but I could not solve the problem

whats the wrong...

repetier, 2015-02-15:
Your settings are wrong I guess :-)

Check endstops with M119 and see if they only show h when triggered. They can prevent moves. Make sure software endstops are only enabled for opposite side of hardware endstops. 

Make also sure motor direction is correct for y axis. I guess you have inverted direction and endstop direction so homing works but y moves are to wrong side.

ta1dr, 2015-02-15:
thanks for answe
this my 5 th printer  I used ramps and printrboard I ıse frist time megatronics board...
I feel like a fool..:-(     still I didnt ....

[image]

my end stop like this 

and my confguratiom.h  ....

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**************** READ FIRST ************************

   This configuration file was created with the configuration tool. For that
   reason, it does not contain the same informations as the original Configuration.h file.
   It misses the comments and unused parts. Open this file file in the config tool
   to see and change the data. You can also upload it to newer/older versions. The system
   will silently add new options, so compilation continues to work.

   This file is optimized for version 0.91
   generator: http://www.repetier.com/firmware/v091/

   If you are in doubt which named functions use which pins on your board, please check the
   pins.h for the used name->pin assignments and your board documentation to verify it is
   as you expect.

*/

#define NUM_EXTRUDER 1
#define MOTHERBOARD 701

#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
// ################ END MANUAL SETTINGS ##########################

#define FAN_BOARD_PIN -1
//#define X_MIN_PIN -1
//#define Y_MIN_PIN -1
//#define Z_MIN_PIN -1
//#define X_MAX_PIN -1
//#define Y_MAX_PIN -1
//#define Z_MAX_PIN -1

//#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not beeing compatible!
//#define COMPAT_PRE1

#define DRIVE_SYSTEM 0
#define XAXIS_STEPS_PER_MM 80
#define YAXIS_STEPS_PER_MM 80
#define ZAXIS_STEPS_PER_MM 400
#define EXTRUDER_FAN_COOL_TEMP 50
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_STEPS_PER_MM 370
#define EXT0_TEMPSENSOR_TYPE 5
#define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
#define EXT0_STEP_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE 0
#define EXT0_ENABLE_PIN E0_ENABLE_PIN
#define EXT0_ENABLE_ON 0
#define EXT0_MAX_FEEDRATE 25
#define EXT0_MAX_START_FEEDRATE 10
#define EXT0_MAX_ACCELERATION 3000
[... 206 lines cut ...]

#define FEATURE_Z_PROBE 1
#define Z_PROBE_BED_DISTANCE 20
#define Z_PROBE_PIN 18//-1
#define Z_PROBE_PULLUP false
#define Z_PROBE_ON_HIGH true   // NPN PNP 
#define Z_PROBE_X_OFFSET -75
#define Z_PROBE_Y_OFFSET 0
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 50
#define Z_PROBE_SWITCHING_DISTANCE 1
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_HEIGHT 1.2
#define Z_PROBE_START_SCRIPT "G0 Z10"
#define Z_PROBE_FINISHED_SCRIPT ""
#define FEATURE_AUTOLEVEL true
#define Z_PROBE_X1 10//20
#define Z_PROBE_Y1 10//20
#define Z_PROBE_X2 180//160
#define Z_PROBE_Y2 10
#define Z_PROBE_X3 130
#define Z_PROBE_Y3 180

when disable z probe all axis work  but enable not work...

repetier, 2015-02-15:
Sure pin 18 is right then? Maybe it is the enable signal also I'm not sure about what enable you are talking. You can test z probe like endstops only with G31. If signal changes it is ok.

ta1dr, 2015-02-15:
thaks for help....
I change pin 18 ( z min endstop pin) to z max endstop pin for probe ....now all ok...
regards..

closed 2015-02-15 by ta1dr

kyrreaa, 2015-02-16:
Just a tip though:
The connection you use on the optocoupler will mean that the mcu cannot tell the difference between a non-triggered endstop and unconnected one.
It is normal to wire the setup so that the pullup is always present on the port, and it is being overruled by the connected but not triggered endstop. That way, disconnecting the endstop will trigger it and software can detect that endstop remains triggered when moving away and towards it meaning an error state. I know that makes it a bit more fiddly for your extra LED that you want to light up, but you can do that by inverting the led setup vs the sensor-signal.
Just a hint though, but it’s generally safer.

From: ta1dr 
Sent: Sunday, February 15, 2015 6:12 PM
To: repetier/Repetier-Firmware 
Subject: Re: [Repetier-Firmware] megatronics v2.0 and repetier problems (#363)

thanks for answe
this my 5 th printer I used ramps and printrboard I ıse frist time megatronics board...
I feel like a fool..:-( still I didnt ....

my end stop like this 

and my confguratiom.h ....

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**************** READ FIRST ************************

This configuration file was created with the configuration tool. For that
reason, it does not contain the same informations as the original Configuration.h file.
It misses the comments and unused parts. Open this file file in the config tool
to see and change the data. You can also upload it to newer/older versions. The system
will silently add new options, so compilation continues to work.

This file is optimized for version 0.91
generator: http://www.repetier.com/firmware/v091/

If you are in doubt which named functions use which pins on your board, please check the
pins.h for the used name->pin assignments and your board documentation to verify it is
as you expect.

*/

#define NUM_EXTRUDER 1
#define MOTHERBOARD 701

#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
// ################ END MANUAL SETTINGS ##########################

#define FAN_BOARD_PIN -1
//#define X_MIN_PIN -1
//#define Y_MIN_PIN -1
//#define Z_MIN_PIN -1
//#define X_MAX_PIN -1
//#define Y_MAX_PIN -1
//#define Z_MAX_PIN -1

//#define EXTERNALSERIAL use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not beeing compatible!
//#define COMPAT_PRE1

#define DRIVE_SYSTEM 0
#define XAXIS_STEPS_PER_MM 80
#define YAXIS_STEPS_PER_MM 80
#define ZAXIS_STEPS_PER_MM 400
#define EXTRUDER_FAN_COOL_TEMP 50
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_STEPS_PER_MM 370
#define EXT0_TEMPSENSOR_TYPE 5
#define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
[... 214 lines cut ...]

#define FEATURE_Z_PROBE 1
#define Z_PROBE_BED_DISTANCE 20
#define Z_PROBE_PIN 18//-1
#define Z_PROBE_PULLUP false
#define Z_PROBE_ON_HIGH true // NPN PNP 
#define Z_PROBE_X_OFFSET -75
#define Z_PROBE_Y_OFFSET 0
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 50
#define Z_PROBE_SWITCHING_DISTANCE 1
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_HEIGHT 1.2
#define Z_PROBE_START_SCRIPT "G0 Z10"
#define Z_PROBE_FINISHED_SCRIPT ""
#define FEATURE_AUTOLEVEL true
#define Z_PROBE_X1 10//20
#define Z_PROBE_Y1 10//20
#define Z_PROBE_X2 180//160
#define Z_PROBE_Y2 10
#define Z_PROBE_X3 130
#define Z_PROBE_Y3 180

when disable z probe all axis work but enable not work...

ta1dr, 2015-02-16:
I think megatronics board is not good  idea or it is wrong.  because  I  load that firmware ( only change motherboard number) on my printrbot LC  printer  it is work well...

also  thanks for sharing your idea...

repetier, 2015-02-16:
SImply changing motherboard is not enough if you use special pins like you did with z-probe. These must match the board and every board has them different.