repetier/RepetierFirmware #365: Board Fan not working anymore

errolt, 2015-02-17:
Hi Repetier,

It seems as if the Board Fan code has been broken recently.

If I understand the code correctly then the board fan has been upgraded to support PWM, but nowhere is that PWM value set to anything, thus the fan never starts...

Thank you,
Errol

errolt, 2015-02-17:
I did some more digging and found that the Board Fan PWM does get set to a fixed 255.
But only if you have a Heated Bed defined and then only if it's temp is over 15C?

errolt, 2015-02-18:
I have traced the problem to HAL.cpp, lines 822 and 913.
The offending statements are the same:
#if FAN_BOARD_PIN > -1  && !defined(SHARED_COOLER_BOARD_EXT)

In Repetier.h SHARED_COOLER_BOARD_EXT is always defined, sometimes as 0, sometimes as 1.

The result is that !defined(SHARED_COOLER_BOARD_EXT) always returns false, even if SHARED_COOLER_BOARD_EXT is set to 0.

errolt, 2015-02-18:
Problem was introduced with "Bluetooth support, faster GLCD on AVR, GLCD on Rambo"

errolt, 2015-02-20:
Just a note. This bug is in 0.92.2

repetier, 2015-03-17:
Should be fixed in current release.

closed 2015-03-17 by repetier