repetier/RepetierFirmware #368: SDramps not working

TTN-, 2015-03-02:
I'm coming from marlin where I was using sdramps just fine.

I'm trying to get sdcard printing working on my ramps1.4 setup. I've played with numerous combinations of sdcard detect invert and different pins but nogo. I've asked around on irc and googled, searched for documentation and searched the forums but haven't found anything usefull.

Any help would be much appreciated!

 Here is my configuration:

/*
    This file is part of Repetier-Firmware.

    Repetier-Firmware is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    Repetier-Firmware is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with Repetier-Firmware.  If not, see <http://www.gnu.org/licenses/>.

*/

#ifndef CONFIGURATION_H
#define CONFIGURATION_H

/**************** READ FIRST ************************

   This configuration file was created with the configuration tool. For that
   reason, it does not contain the same informations as the original Configuration.h file.
   It misses the comments and unused parts. Open this file file in the config tool
   to see and change the data. You can also upload it to newer/older versions. The system
   will silently add new options, so compilation continues to work.

   This file is optimized for version 0.91
   generator: http://www.repetier.com/firmware/v091/

   If you are in doubt which named functions use which pins on your board, please check the
   pins.h for the used name->pin assignments and your board documentation to verify it is
   as you expect.

*/

#define NUM_EXTRUDER 1
#define MOTHERBOARD 33

#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
// ################ END MANUAL SETTINGS ##########################

#define FAN_BOARD_PIN -1

//#define EXTERNALSERIAL  use Arduino serial library instead of build in. Requires more ram, has only 63 byte input buffer.
// Uncomment the following line if you are using arduino compatible firmware made for Arduino version earlier then 1.0
// If it is incompatible you will get compiler errors about write functions not beeing compatible!
//#define COMPAT_PRE1

#define DRIVE_SYSTEM 0
#define XAXIS_STEPS_PER_MM 200
#define YAXIS_STEPS_PER_MM 200
#define ZAXIS_STEPS_PER_MM 4000
#define EXTRUDER_FAN_COOL_TEMP 50
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_STEPS_PER_MM 621
#define EXT0_TEMPSENSOR_TYPE 1
[... 601 lines cut ...]
    "xMinPin": "ORIG_X_MIN_PIN",
    "yMinPin": "ORIG_Y_MIN_PIN",
    "zMinPin": "ORIG_Z_MIN_PIN",
    "xMaxPin": "ORIG_X_MAX_PIN",
    "yMaxPin": "ORIG_Y_MAX_PIN",
    "zMaxPin": "ORIG_Z_MAX_PIN",
    "deltaHomeOnPower": "0",
    "fanBoardPin": -1,
    "heaterPWMSpeed": 0,
    "featureBabystepping": "1",
    "babystepMultiplicator": 1,
    "hasMAX6675": false,
    "hasMAX31855": false,
    "hasGeneric1": false,
    "hasGeneric2": false,
    "hasGeneric3": false,
    "hasUser0": false,
    "hasUser1": false,
    "hasUser2": false,
    "numExtruder": 1,
    "version": 91.7
}
========== End configuration string ==========

*/

TTN-, 2015-03-02:
It fails to compile with #define SDCARDDETECT SDCARDDETECT

When using #define SDCARDDETECT SDSS
and I invert sd card detect, it will tell me sd card init failed. but when its not inverted there is no error message. There is a file called AUTO0.G that should start to auto print but it doesn't.

repetier, 2015-03-03:
sdramps has no SDCARDDETECT so you need to set it to -1

TTN-, 2015-03-04:
Oh! ok. Thanks. I saw that it was set to -1 and I assumed it meant not in use, and it didn't start printing auto0.g so thats what I assumed.

Ok so now that is set to -1 and I can see the files on the sdcard from pronterface but it won't autostart auto0.g and I don't see any messages letting me know that sd card init succeeded, which might be a helpful addition imo.

repetier, 2015-03-04:
Autostart file must be named init.g not auto0.g.

TTN-, 2015-03-04:
Cool thanks. I googled it but couldn't find anything on init.g 
Solved cheers :+1:

closed 2015-03-04 by TTN-