repetier/RepetierFirmware #374: error compilation

maralb, 2015-03-06:
I'm using version 0.92 yesterday 03/05/2015
get this error
Extruder.cpp:295:61: error: operator '&&' has no right operand

repetier, 2015-03-06:
You mean this line
#if defined(KILL_IF_SENSOR_DEFECT) && KILL_IF_SENSOR_DEFECT > 0

works for me perfectly. What value is 
KILL_IF_SENSOR_DEFECT in your Configuration.h (should be 0 or 1)

maralb, 2015-03-06:
my value and unknown
#define PDM_FOR_COOLER 1
#define DECOUPLING_TEST_MAX_HOLD_VARIANCE 20
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
#define KILL_IF_SENSOR_DEFECT 
#define RETRACT_ON_PAUSE 2

use the online configurator 0.92.2
where to find the entry for setting is printed ??

repetier, 2015-03-07:
Not sure how you could get

#define KILL_IF_SENSOR_DEFECT

on my tests I never get it empty. That value must be

#define KILL_IF_SENSOR_DEFECT 0

What does "where to find the entry for setting is printed ??" mean? YO mean in the config tool? There it is in temperature tab with the other checkboxes.

maralb, 2015-03-07:
Yes.
have now been added in the configurator
-Kill/reset firmware on defect sensor. 
-Reverse encoder direction. 
Ok now
thank you

closed 2015-03-07 by repetier