repetier/RepetierFirmware #375: 0.92 Heat bed bang bang control triggers Thermistor Decouple

dutchrazor, 2015-03-07:
Dear all,

Today during printing I kept getting "Thermistor decoupled" for the heated bed thermistor. After checking the thermistor, and plotting the temperature in Repetier, all seemed fine.

After changing to following:

#define DECOUPLING_TEST_MIN_TEMP_RISE -10

the problem is gone. 

Is it possible that bang bang control for the heat bed triggers an erroneous thermistor decoupled signal, since Repeater assumes the bed is in heating mode instead of temperature control mode?

luc-github, 2015-03-08:
well test is 
 if(act->currentTemperatureC - act->lastDecoupleTemp < DECOUPLING_TEST_MIN_TEMP_RISE)   // failed test
so putting -10 mean error will happen last decouple test is 10 degres over current temp at starting which won't happen 

and it is at start for hold temperature it use DECOUPLING_TEST_MAX_HOLD_VARIANCE

I do not say it is bug free but I never had issue and when I had some was always Hardware problem connector loose or fan defect

You can try to change from bang bang control to PID control to test also.

dutchrazor, 2015-03-08:
Thanks for the quick reply, I will try PID or dead time instead of bang bang.

If I remember correctly the error during printing did indicate it was in start up decouple mode instead of hold temperature mode, but  I did not save the logging. Is the full log of a print writen to disk somewhere?

repetier, 2015-03-08:
Bang-Bang has in deed no startup phase like PID/deadtime. I'm currently changing that. Also it never happened to me I can imagine that this can cause some error. After all it's the reason I made the difference. So next update should hopefully fix that.

repetier, 2015-03-11:
New version is now online. Hope this now also works with 
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
as supposed to be :-)

luc-github, 2015-03-13:
Since this update my bed is stuck to target temperature - 20, never reach target and go in error
if I set 70 it stop at 50, if 90 stop at 70, etc...
I found  it is linked to 
```pwm_pos[act->pwmIndex] = output; // set pwm signal```
 done after all tests instead of after each output definition within each test like previously

if I remove the final assignation and put back after each output definition like:
```               
 output = constrain((int)pidTerm, 0, act->pidMax);
 pwm_pos[act->pwmIndex] = output; // set pwm signal
```
I have no more issue,
So I guess there is lack in the logic after
```
#if TEMP_PID
        act->tempArray[act->tempPointer++] = act->currentTemperatureC;
        act->tempPointer &= 3;
        uint8_t output = 0;
```
it sets a wrong value for output ( 0) when it should still heat  - sorry I did not dig more

repetier, 2015-03-13:
Excelent analysis. I really like how you look at thinks, making fixing so easy.
The slow bang bang in deed only computes a update every x seconds so in case it does not update it should simply continue with the next heater instead of finishing the loop. Have fixed it now.

closed 2015-03-13 by repetier