repetier/RepetierFirmware #385: 0.92.3 SD Card sometimes cause menu slowdown

hannoHI, 2015-04-06:
I upgraded from 0.92.2 to 0.92.3 and now my SD card seem to slow down (to a crawl) the response of my RADDS LCD on a DUE. Can not print from SD while in this state.
If I unplug the SD card the menu almost instantly start to work again. 
Trying a few times unplugging the SD if the menu response is slow I sometimes get the SD card directory & then I can normally use & print from it. Seems something changed in the SD Card initialization process.
This also happen when I reset (or connect from repetier-host) and cause the printer to respond very slowly to the gcode from the PC.

repetier, 2015-04-07:
No, sd handling has not changed for quite a while. You get this effect when you have sd detect and mounting fails. Then you get in a mount loop that slows down everything. If you are connected to the host you should also see the corresponding messages in the log.

So the question should be more why does sd mounting now start to fail. Something changed on wire positions in the printer?

hannoHI, 2015-04-07:
Nothing changed on the Cabling side. If I go back to the older version that worked before I do not seem to be able to get the same problem.
I noticed that the LCD timings have been reduced. Is the detection of the SD first written & then initialized? I will dig into it a bit later. Trying to get my second machine up.

closed 2015-04-07 by hannoHI