repetier/RepetierFirmware #389: "dec" error @Rumba

Jelwek, 2015-04-14:
Hi! I have some trouble with Repetier at Rumba - every now and then I get the "dec" text where the bed temperature suppose to be - both heater turn off and the extruder stops working but all the other axes are continuing to print. Is it shortcut on the termistor? I didn't find one. Shouldn't it stop the printer if the temperature is out of range due to shortcut?
Thanks in advance

repetier, 2015-04-14:
No, def is defect = shortcut/missing. dec is short for decoupled. Make sure to use latest version is you haven't updated in the last weeks. There was at least one problem that could track a wrong decouple message. Except that check your settings for decoupled if one of them is too narrow for your bed, which reacts much slower then a extruder.

Jelwek, 2015-04-15:
Thank You for that information - it is a bit clearer to me now. If I get it right - can it be caused by bug histeresis of bang-bang heating of the bed? What parameter should I change in the firmware (it's up to date)?

repetier, 2015-04-15:
Decouple variance 20 and min rise 1 which are defaults work without problem for me.

In bed settings Decouple test period must be high enough that you can reach min bed rise in that time. Especially with slow bed that can be a problem.

Jelwek, 2015-04-15:
Got it! Thank You very much!