repetier/RepetierFirmware #390: Delta homing

pkmcnc, 2015-04-14:
Hi,
For my delta I use 3 NC switches at max height http://i.imgur.com/RvmG70A.png
I think that the homing sequence is a bit wrong: all actuators go up, then the first two actuators hit the switch and go down a bit, then all three go up at reduced speed to their switches. Thus the last actuator often goes up a large distance at reduced speed.
Normally the reduced speed should be enabled only after all three actuators hit their switches.
Please check that. Thanks

pkmcnc, 2015-04-14:
Just noticed that in configurator (see the above screenshot) it should be [mm for X,Y and Z] instead of [- for X,Y and Z] after "Endstop distance after homing"

BenOoi, 2015-06-05:
I have my min switches turned off. No need to enable this with a delta

repetier, 2015-06-05:
The sequence is correct. If one stops before the end it has triggered the endstop early, often through cross talk. New 0.92 release should fix that with new endstop handling.

closed 2015-06-05 by repetier