repetier/RepetierFirmware #392: Config Thermistor 100k

bgcop, 2015-04-29:
Hello, i have a ramps 1.4 in my printer. 

The thermistor in my hotend is:

Item:NTC Thermistor
Model:B2-100-3950-1
R @ 25: K 100K-1%
B25/50:K 3950-1%
Element:Thermistor,type MJB2
 
3. Characteristics
(1) Electrical characteristics
A: Resistance value: R(25°C)=100K±1%
B: B Value: B=3950±1% (calculated from resistance value at 25°C and 50C)
C: Insulation resistance 50MΩ or over by DC500V megger (between glass and lead wire)
(2) Thermal time constant (τ): τ≤10~17s (in still air)
(3) Thermal dissipation constant (δ): δ=1.1~1.6mW/°C (in still air)
(4) Operating temperature range:-50~+260°C

But when a set 200°c in repetier host and firmware rise to this temperature, a real measure is 168°c. The thermistor selected is 1. How to generate new table for this thermistor?

repetier, 2015-04-30:
These generic curves are never that exact. Especially since the second reference point was at 50°C the deviation at 200°C will increase. So your values would be good at 0-100°C maybe and the further you leave this area the more deviation you get. So to get better results you should not use the generic data but copy the temperature/resistance in a user defined table with 10-20°C steps. That gives much better temperatures.

The second thing is at high temperatures there is nearly no change in analog values measured, so precision will drop. And last but not least you measure a different temperature then the sensor.

But I guess with the user table you will already come much closer.

onsc, 2015-07-14:
i am using this configuration for that sensor. it seems working.

#define NUM_TEMPS_USERTHERMISTOR0 50
#define USER_THERMISTORTABLE0 {{1*4,8*938}, {31*4,8*314}, {41*4,8*290}, {51*4,8*272}, {61*4,8*258}, {71*4,8*247}, {81*4,8*237}, {91*4,8*229}, {101*4,8*221}, {111*4,8*215}, {121*4,8*209}, {131*4,8*204}, {141*4,8*199}, {151*4,8*195}, {161*4,8*190}, {171*4,8*187}, {181*4,8*183}, {191*4,8*179}, {201*4,8*176}, {221*4,8*170}, {241*4,8*165}, {261*4,8*160}, {281*4,8*155}, {301*4,8*150}, {331*4,8*144}, {361*4,8*139}, {391*4,8*133}, {421*4,8*128}, {451*4,8*123}, {491*4,8*117}, {531*4,8*111}, {571*4,8*105}, {611*4,8*100}, {641*4,8*95}, {681*4,8*90}, {711*4,8*85}, {751*4,8*79}, {791*4,8*72}, {811*4,8*69}, {831*4,8*65}, {871*4,8*57}, {881*4,8*55}, {901*4,8*51}, {921*4,8*45}, {941*4,8*39}, {971*4,8*28}, {981*4,8*23}, {991*4,8*17}, {1001*4,8*9}, {1021*4,8*-27}}

sanchosk, 2016-02-06:
Thanks for the table - I fixed my printer thanks to it ;) My vendor provided the built code for the firmware and some "source code" (junk with unusable values).
I got majority of the printer working, but the bed was outputting 55 degrees when my infra temp measured ~80... Your table fixed it perfectly!

closed 2016-02-07 by repetier