repetier/RepetierFirmware #393: use more extruders at the same time

Erik3000, 2015-04-29:
Hello, 

I am interested in printing with more extruders at the same time. Is it possible with current firmware?

Thank you

repetier, 2015-04-30:
We call this ditto printing. 0.92.3 supports this with up to 4 extruders simultaneously. But all will print the same object. That is a requirement because firmware can only receive one gcode at a time.

You will need this command:
- M280 S<mode> - Set ditto printing mode. mode: 0 = off, 1 = 1 extra extruder, 2 = 2 extra extruder, 3 = 3 extra extruders

Erik3000, 2015-04-30:
thanks, but would it be too difficult to modify firmware, to receive different signal for each extruder? 

For example - first extruder would be called "Ea", second "Eb", and the gcode would look like G1 X50 Y25 Ea15 Eb30?

kyrreaa, 2015-04-30:
What is the difference between this and mixing extruder type config ?
I thought this was already supported...

From: Erik3000 
Sent: Thursday, April 30, 2015 11:51 PM
To: repetier/Repetier-Firmware 
Subject: Re: [Repetier-Firmware] use more extruders at the same time (#393)

thanks, but would it be too difficult to modify firmware, to receive different signal for each extruder? 

For example - first extruder would be called "Ea", second "Eb", and the gcode would look like G1 X50 Y25 Ea15 Eb30?

kyrreaa, 2015-04-30:
Also, Ea Eb etc would break gcode compatibility/standard.
Not good move.

From: Erik3000 
Sent: Thursday, April 30, 2015 11:51 PM
To: repetier/Repetier-Firmware 
Subject: Re: [Repetier-Firmware] use more extruders at the same time (#393)

thanks, but would it be too difficult to modify firmware, to receive different signal for each extruder? 

For example - first extruder would be called "Ea", second "Eb", and the gcode would look like G1 X50 Y25 Ea15 Eb30?

repetier, 2015-05-01:
First GCODE has only one letter per part, so that is not supported by the gcode parser. And second yes it is difficult that way as no slicer creates such code and there is no reason why they should.

As kyrreaa pointed out the mixing extruder feature is meant to driver multiple motors with a set ratio. So that is what you want anyway if it wasn't ditto printing.

closed 2015-05-01 by repetier

kyrreaa, 2015-05-01:
It should also be pointed out that the Ex.xxx notation used by printer should never have been done.
Gcode already supported 9 axis. XYZ, UVW and ABC. These can appear on a single command together and would be executed in paralell. If the original implementations of the extruder code had been treated as just another axis and motor, things would have been much simpler in the code, and much simpler in use.

Erik3000, 2015-05-04:
well, it doesnt strictly have to be Ea and Eb - it was only an example. I dont care that there isnt any slicer that supports this. I just wanted to know whether it is too difficult to modify firmware to control two extruders individually at the same time (for example using mentiond axis ABC)? 
Or could you give me some hints, where to look in code to modify it?
thanks

repetier, 2015-05-05:
You have to rewrite all movement related function and add one or two more dimensions to the bresenham algorithm to interpolate 5 or 6 instead of the 4 values you now have. Not complicated just requires to rewrite many functions and slows down the handling a bit. For that reason it will not be part of the official version. SImply to special as there is no need for this with mixing extruders already working as a alternative.

skwhite16, 2018-08-27:
Could you provide a little more detail about how to add this functionality to the firmware? I'm trying to move two extruders independently, would like to be able to address them as A and B or something similar. I'm not a software/firmware expert but have been working with repetier for a while, just looking for a starting point.

repetier, 2018-08-28:
In V1 they never move independently. You can make 2 or more extruders extrude the same time and with dual x you can have them at 50% distance. Next update for V2 will also support dual x which then can instead also mirror x axis. In V2 you can also have 7 axis control but you will never be able to have 2 independent extruders. That would also require a very special gcode and splitted extrusion values and anti collision logic and more.

skwhite16, 2018-08-28:
My situation is a bit different, I am not building a traditional 3D printer but I am using 3D printing hardware and firmware for a different project. So anti-collision logic and things like that aren't necessary for me. I also write my own G-Code instead of using a slicer. All I would really need would be an additional axis so that I could move the extruder motors independently but simultaneously. So a Gcode line could be G1 Xnnn Ynnn Znnn Ennn Dnnn. Would also need to be able to set the temperature for the 2nd extruder but it would be the same as the 1st one. Is there a simple way to add an additional axis?

repetier, 2018-08-28:
The simples solution is to switch to V2 when next update is out.  Just running some tests for the next update that should make it usable also currently only heaters and motion is working.There you also have axis a,b,c you can control independently. But be aware there is no config tool and description is not up to date.

skwhite16, 2018-08-28:
When will V2 be out?

repetier, 2018-08-28:
It is already out in dev2 tree here on github. But the next update will contain a massive amount of fixes as I currently use 5 axes for dual x axis and found some errors while testing.

skwhite16, 2019-03-04:
Any update on the status of repetier V2? I am still interested in having more controlled axes, was just wondering if some of the bugs have been worked out?

repetier, 2019-03-05:
In general it is working good and many functions have been added. So with due based boards I have already cartesian, dual x and core xy working. Delta is also working but not fully tested.

Only thing is there is only one active extruder, except in ditto mode. You can have a independed extra motor and heating multiple extruders is no problem. So you can move a second extruder completely independent with a,b,c axis if you have a slicer creating the code. Naturally this is completely untested.