repetier/RepetierFirmware #394: REQUEST: SUICIDE_PIN functionality.

errolt, 2015-04-30:
Please add SUICIDE_PIN functionality.

repetier, 2015-04-30:
What is wrong with existing

UI_ACTION_RESET and UI_ACTION_KILL which you can put on any pin you like? With stop current print.

What does "suicide on kill?" mean?

errolt, 2015-04-30:
Sorry about confusion. SUICIDE_PIN is labeled in pins.h as a pin that the main board must keep high. If allowed to drop to 0V then the power supply turns off and the whole printer dies.

#define SUICIDE_PIN    -1  //PIN that has to be turned on right after start, to keep power flowing.

I was able to do the same with the correct settings for PS_ON_PIN, ENABLE_POWER_ON_STARTUP and POWER_INVERTING.

Maybe SUICIDE_PIN should be removed from pins.h?

closed 2015-04-30 by errolt

repetier, 2015-04-30:
Yes, should delete it. Comes from Marlin copying I guess. Never had a meaning for repetier.