repetier/RepetierFirmware #396: Advance algorithm causes motor hiccups

qx1147, 2015-05-03:
Since I had enabled the ADVANCE algorithm a while ago, unhealthy sounds came from the XY motors at the end of the 1st layer skirt loop, at least with some models. I kept ignoring this, but with the model I wanted to print yesterday, this caused the printer to lose steps even if I set the advance parameter to a very small value (!=0 though). It turned out that, for this particular model, the line segment following the retraction at the end of the skirt loop was rather long (>5mm). Without changing the firmware, I could work around the problem either by splitting up this segment into two or by adding a G4 command (dwell time) after the retraction. I suspect that this is a more general problem with moves before and/or after retraction/prime, but line segments are normally too short to cause audible or visible effects. Moreover, this only occurs with ADVANCE being enabled.
In motion.cpp/computeMaxJunctionSpeed(), there is a special USE_ADVANCE case, which looks suspicious to me:
```
if(previous->isEMove() != current->isEMove() && (previous->isXOrYMove() || current->isXOrYMove()))
{
previous->setEndSpeedFixed(true);
current->setStartSpeedFixed(true);
previous->endSpeed = current->startSpeed = previous->maxJunctionSpeed = RMath::min(previous->endSpeed,current->startSpeed);
previous->invalidateParameter();
current->invalidateParameter();
return;
}
```
Not only is the rationale behind the conditional expression rather unclear (checking for isEOnlyMove() seems to be clearer, see below), but also changing and fixing previous->endSpeed and then invalidating the parameters of 'previous' does not look quite right, especially given the code that follows directly after, in updateTrapezoids():
```
if(previous->isEOnlyMove() != act->isEOnlyMove())
{
previous->setEndSpeedFixed(true);
act->setStartSpeedFixed(true);
act->updateStepsParameter();
firstLine->unblock();
return;
}
```
This condition looks similar to the one in computeMaxJunctionSpeed() and is likely true if the other one is, leaving the planning queue in an inconsistent state as the 'previous' parameters have been invalidated but not updated.
My particular problem is "solved" by getting rid of the ADVANCE specific code fragment in computeMaxJunctionSpeed(). Moreover, I think that the call to computeMaxJunctionSpeed() can come after the check involving isEOnlyMove(), because previous->maxJunctionSpeed should not be relevant anyway if the previous->endSpeed is fixed directly after.
However, I am not sure of these changes because I neither see through the details of the path planning nor do I really understand what exactly goes haywire when a retraction is followed by a long line segment. 
Here is the according g-code (KISSlicer, for a Cartesian printer):
```
G1 X36.18 Y130.53 E5.2952
G1 X37.3 Y130.25 E5.3819
G1 X46.25 Y130.25 E6.0514
;
; 'Destring/Wipe/Jump Path', 0.0 [feed mm/s], 10.0 [head mm/s]
G1 E5.0514 F900
; G4
G1 X53.12 Y130.25 E5.0514 F600   <<<<<---- this is where the X motor loses steps, making the according sound
G1 X53.93 Y130.38 E5.0514
```

sigxcpu76, 2016-05-26:
Did you check the latest firmware without this change? I'm asking because I also have random bangs coming from motors on certain moves, but I see that the code in that area changed a bit since you've posted this.

qx1147, 2016-05-26:
No, I haven't checked since. If you are using the ADVANCE algorithm, the bangs might as well be caused by other ADVANCE-related issues - see also https://github.com/repetier/Repetier-Firmware/issues/342.

sigxcpu76, 2016-05-26:
Extruder motor (cartesian printer + bowden) noises are different. The bangs are 100% coming from X or Y, my guess being Y, because it is heavier. I've had this issue in the past happening more often but that was caused by enabling DISABLE_{XYZ} options. The sound is the same, so I'm pretty sure it is a sudden acceleration of one of X or Y axes.