repetier/RepetierFirmware #398: PDM_FOR_COOLER 0 breaks board fan

errolt, 2015-05-13:
Hi Repetier,
If I set PDM_FOR_COOLER=0 then the board fan does not work.

If I set PDM_FOR_COOLER=1 then the board fan works, but I can't get extruder fan to 100%, no matter what I set EXT0_EXTRUDER_COOLER_SPEED to.

Thank you,
Errol

errolt, 2015-05-13:
Never mind. Now it works... :)

errolt, 2015-07-12:
Hmm, don't know why. Seeing same issue again. Due, with firmware from yesterday. Setting PDM_FOR_COOLER=0 prevents board fan from starting up. Setting it to 1 then the board fan works. Switched back and forth now a few times. Still seeing same issue.

PDM does not work on any of my fans. With it set I must use 245 for the fan to barely start, 255 for full speed. And the fan whines.

repetier, 2015-07-12:
I guess it depends on the fan type. You should always use the cheap one without speed control. Speed control requires a high frequency pwm signal I think to set speed over the signal pin. What we do is simply turning power on and off and that is probably not the best way. I wonder why 255 does not work for you. This should keep power permanently on and always work.

errolt, 2015-07-12:
PDM gives me problems on my hotend fan. This is on a Delta and it is a 40mm fan. I'm using a 2 wire fan, possibly of higher quality(it is claimed to be a maglev bearing fan) as I want a decent fan due to all the forces exerted on the head.

With PDM=1 then 255 works, but the fan is too fast. And at 240 the fan does not start at all.

With PDM=0 the hot end fan is great and I can run it reliably at 120, but the board fan does not start at all. I do have the board fan connected to D9 which is usually hotend2 output, don't know if that might be the problem, will investigate more tomorrow...

errolt, 2015-07-12:
Found.

I don't know why PDM=1 works and PDM=0 does not work.

I have a mixing extruder, so heater 2 is not used and not controllable, but it was still defined as a heater pin and was constantly set to 0, even though the fan was setting it to 1.

Changed heater2 pin to -1 now it works with PDM=0... :)

closed 2015-07-12 by errolt

repetier, 2015-07-12:
I understand. The PWM manager does not know it is a mixing extruder so it still has a PWM place for the heater which is the same as fan. So one enables it and the other disables it.