repetier/RepetierFirmware #400: error U8glib.cpp

maralb, 2015-05-16:
with last update I can not fill
Alligator Board define Lcd 11.

Arduino:1.6.4 (Windows 7), Scheda:"Alligator 3D Printer Board R2 (Native USB Port)"

In file included from ui.cpp:681:0:
u8glib_ex.h: In function 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)':
u8glib_ex.h:9022: error: redefinition of 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)'
 static void u8g_com_arduino_init_shift_out(uint8_t dataPin, uint8_t clockPin)
             ^
u8glib_ex.h:6943: error: 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)' previously defined here
 static void u8g_com_arduino_init_shift_out(uint8_t dataPin, uint8_t clockPin)
             ^
u8glib_ex.h: In function 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)':
u8glib_ex.h:9028: error: redefinition of 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)'
 static void u8g_com_arduino_do_shift_out_msb_first(uint8_t val)
             ^
u8glib_ex.h:6949: error: 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)' previously defined here
 static void u8g_com_arduino_do_shift_out_msb_first(uint8_t val)
             ^
redefinition of 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)'

thanks

repetier, 2015-05-17:
In deed. Addition if new drivers caused a double definition. Should be fixed now.

closed 2015-05-17 by repetier

designermx2, 2015-08-17:
I have a similar problem. I added a smart-ramps board on pins.h and set the feature_controller to 11:

Arduino: 1.5.7 (Mac OS X), Board: "Arduino Due (Programming Port)"

In file included from ui.cpp:662:0:
u8glib_ex.h: In function 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)':
u8glib_ex.h:7836:13: error: redefinition of 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)'
 static void u8g_com_arduino_init_shift_out(uint8_t dataPin, uint8_t clockPin)
             ^
In file included from ui.cpp:662:0:
u8glib_ex.h:3812:13: error: 'void u8g_com_arduino_init_shift_out(uint8_t, uint8_t)' previously defined here
 static void u8g_com_arduino_init_shift_out(uint8_t dataPin, uint8_t clockPin)
             ^
In file included from ui.cpp:662:0:
u8glib_ex.h: In function 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)':
u8glib_ex.h:7842:13: error: redefinition of 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)'
 static void u8g_com_arduino_do_shift_out_msb_first(uint8_t val)
             ^
In file included from ui.cpp:662:0:
u8glib_ex.h:3818:13: error: 'void u8g_com_arduino_do_shift_out_msb_first(uint8_t)' previously defined here
 static void u8g_com_arduino_do_shift_out_msb_first(uint8_t val)
             ^

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

repetier, 2015-08-17:
@designermx2 If you add a new board you can not expect it to be known by any controller. Especially 11 is a combination with 2 and it switches pin assignment based on selected supported boards. So you need to add your case there. Without this you might not have set the right variables to make it work properly. See e.g. the SparkLCD which is the same display as used in 11 but on a RADDS board with due and without sd card.

designermx2, 2015-08-17:
But do you know this borad?
The Pins was set but without success.

It can be here or need to be a new inssue post?

The board is:

http://pt.aliexpress.com/item/3D-print-unit-new-reference-SMART-Ramps-For-Arduino-Due-Like-Ramps-FD-RADDS-AZTEEG-MINI/32213624860.html

repetier, 2015-08-18:
No I did not know it. That also explains why the lcd is unknown. Having the same names does not mean the pins are the same. Will add the pins entries soon also I'm not sure they are all correct. At least the ending #endif is missing and it should always be #define not #Define as they say.

designermx2, 2015-08-19:
Sorry about the #Define.

Today i used the repetier online configuration tool and now it has a smartramps as default configuration.

The board was selected and the lcd too... and now i have a error:

Arduino: 1.5.7 (Mac OS X), Board: "Arduino Due (Programming Port)"

In file included from Repetier.h:415:0,
                 from Printer.cpp:19:
Printer.cpp: In static member function 'static void Printer::setup()':
Printer.cpp:735:16: error: 'SDPOWER' was not declared in this scope
     SET_OUTPUT(SDPOWER);
                ^
HAL.h:159:57: note: in definition of macro 'SET_OUTPUT'
 #define SET_OUTPUT(pin) PIO_Configure(g_APinDescription[pin].pPort, PIO_OUTPUT_1, \
                                                         ^
HAL.h:154:41: error: 'DIOSDPOWER_PORT' was not declared in this scope
 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)
                                         ^
HAL.h:155:22: note: in expansion of macro '_WRITE'
 #define WRITE(pin,v) _WRITE(pin,v)
                      ^
Printer.cpp:736:5: note: in expansion of macro 'WRITE'
     WRITE(SDPOWER, HIGH);
     ^
HAL.h:154:77: error: 'DIOSDPOWER_PIN' was not declared in this scope
 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)
                                                                             ^
HAL.h:155:22: note: in expansion of macro '_WRITE'
 #define WRITE(pin,v) _WRITE(pin,v)
                      ^
Printer.cpp:736:5: note: in expansion of macro 'WRITE'
     WRITE(SDPOWER, HIGH);
     ^
HAL.h:154:106: error: 'DIOSDPOWER_PORT' was not declared in this scope
 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)
                                                                                                          ^
HAL.h:155:22: note: in expansion of macro '_WRITE'
 #define WRITE(pin,v) _WRITE(pin,v)
                      ^
Printer.cpp:736:5: note: in expansion of macro 'WRITE'
     WRITE(SDPOWER, HIGH);
     ^
HAL.h:154:140: error: 'DIOSDPOWER_PIN' was not declared in this scope
 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)
                                                                                                                                            ^
HAL.h:155:22: note: in expansion of macro '_WRITE'
 #define WRITE(pin,v) _WRITE(pin,v)
                      ^
Printer.cpp:736:5: note: in expansion of macro 'WRITE'
     WRITE(SDPOWER, HIGH);
     ^

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

The firmware was not modifyed