repetier/RepetierFirmware #402: AD8495 value out of range

robertovic, 2015-05-19:
hello i'm a mechanic of truck not a programmer.
I have flashed a ARDUiNO DUE with repetier firmware v092.
I have a problem with the displayed temperature from repetier host ecc.

In the file Configuration.h  i have selected the temperature sensor number 60. "#define EXT0_TEMPSENSOR_TYPE 60"
When i connect repetier_host i see the temperature celsius over 400.
For to view near correct value i have added in fthe file Extruder.cpp it:

In "void TemperatureController::updateCurrentTemperature()" :
    
case 60: // AD8495 (Delivers 5mV/degC vs the AD595's 10mV)
       // currentTemperatureC = ((float)currentTemperature * 1000.0f / (1024 << (2 - ANALOG_REDUCE_BITS)));
	pizza=(float)currentTemperature / 4.0;    
	valore=pizza * (3.38 /1024.0);        
	adessotemp= (valore - 1.25);	
	currentTemperatureC = adessotemp / 0.005;      
	break;

My Hardware:
http://www.adafruit.com/products/1778
http://www.arduino.cc/en/Main/arduinoBoardDue
Thermocouple type K

I don't understand because the original code not work for me.
Is correct my code?
The solution is add another hardware near ad8495?
Can you help me?

Thank at all,
I'm not english 

Bye

repetier, 2015-05-19:
Hi,

the big question is where you connected the output signal. If you 
connect it to thermistor inputs you will get wrong readings.
You need to connect it to a direct adc channel and then set the channel 
number (not pin number, not arduino analog pin number) as input pin. 
E.g. for the RADDS board there is ADC pin near the thermistor inputs on 
the long side that has Channel 2 also it is Arduino pin A5.

You can get the channels from this

http://www.robgray.com/temp/Due-pinout.svg

Channel number are the one starting with AD

I'm currently testing this myself with the same board. If I see more 
errors then this I will add it soon to firmware.

Yours
Roland Littwin

-- 
Mit freundlichen Grüßen/Yours

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

kyrreaa, 2015-05-19:
Remember that the ADxxxx series need to be connected to a analog pin that
does not have pullup. You cannot use the thermistor inputs. I recommend
using min endstops or aux pins if you can depending on what you have unused.

Kyrre

repetier, 2015-05-21:
Hi,

I had now time to test. The available types had wrong valtage 
computation for thermocouples and there was no type with 1.25v offset. 
So I have corrected that and added type 61 for the adafruit breakout. 
Works with my RADDS setting it to THERMOCOUPLE_0_PIN whcih is 2 for the 
ADC marked pin. Only thing is that it missed my room temperature by 7°C 
so Vref resp. the resistor creating it is not exactly 1.25.

Yours
Roland Littwin

-- 
Mit freundlichen Grüßen/Yours

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

robertovic, 2015-05-24:
I have found a method to resolve my problem  with it :
	#define END_SCALE_ADC 4096
	#define OFFSET_TEMPERATURE -250.0f
	#define MAX_TEMPERATURE	(410.0f - OFFSET_TEMPERATURE)
	#define CALIBRATION 5

in "void TemperatureController::updateCurrentTemperature():"

 case 60: // AD8495 (Delivers 5mV/degC vs the AD595's 10mV)
	currentTemperatureC = ((float)currentTemperature * MAX_TEMPERATURE / END_SCALE_ADC) + OFFSET_TEMPERATURE+CALIBRATION;
	break;

The voltage  DC for the  ad8495 is 3.3v.
thank at all..
Bye

kyrreaa, 2015-05-24:
I disagree.
If there is a pullup involved the error is not a proper offset and it will be wrong to treat it as such. Error will grow as temp rises.

Kyrre

From: robertovic 
Sent: Sunday, May 24, 2015 10:30 PM
To: repetier/Repetier-Firmware 
Cc: kyrreaa 
Subject: Re: [Repetier-Firmware] AD8495 value out of range (#402)

I have found a method to resolve my problem with it :
#define END_SCALE_ADC 4096
#define OFFSET_TEMPERATURE -250.0f
#define MAX_TEMPERATURE (410.0f - OFFSET_TEMPERATURE)
#define CALIBRATION 5

in "void TemperatureController::updateCurrentTemperature():"

case 60: // AD8495 (Delivers 5mV/degC vs the AD595's 10mV)
currentTemperatureC = ((float)currentTemperature * MAX_TEMPERATURE / END_SCALE_ADC) + OFFSET_TEMPERATURE+CALIBRATION;
break;

The voltage DC for the ad8495 is 3.3v.
thank at all..
Bye

repetier, 2015-05-25:
Please use latest version with type 61

    case 61: // AD8495 1.25V Vref offset (like Adafruit 8495 breakout board)
#if CPU_ARCH == ARCH_AVR
        currentTemperatureC = ((float)currentTemperature * 1000.0f / (1024 << (2 - ANALOG_REDUCE_BITS))) - 250.0f;
#else
        currentTemperatureC = ((float)currentTemperature * 660.0f / (1024 << (2 - ANALOG_REDUCE_BITS))) - 250.0f;
#endif
        break;
This only lacks a calibration I guess as my AD8495 has 7°C offset at room temperature but otherwise rises as expected. I guess it is easy to get some 0.035V change in reference on the one or other side.

robertovic, 2015-05-25:
Sorry,
I don't found the modified file Extruder.cpp.
Where i can found it?
Thank very much at all for fast reply and for your work.
Bye

kyrreaa, 2015-05-25:
I just had a look at that board and indeed, they have chosen to lift the reference up by 1.25v to allow negative temperatures.
Further more, note that the k-probe terminals are placed far from the amplifier circuit which will more easily influence the temperature compensation. If amp and connector is not at same temp errors are added.
Be adviced that this board has no output protection and shorting of output, or connection to supply of any kind to output pin can damage the board. It does however have proper common mode filtering etc.

Kyrre

From: repetier 
Sent: Monday, May 25, 2015 9:03 AM
To: repetier/Repetier-Firmware 
Cc: kyrreaa 
Subject: Re: [Repetier-Firmware] AD8495 value out of range (#402)

Please use latest version with type 61

case 61: // AD8495 1.25V Vref offset (like Adafruit 8495 breakout board)
#if CPU_ARCH == ARCH_AVR
currentTemperatureC = ((float)currentTemperature * 1000.0f / (1024 << (2 - ANALOG_REDUCE_BITS))) - 250.0f;
#else
currentTemperatureC = ((float)currentTemperature * 660.0f / (1024 << (2 - ANALOG_REDUCE_BITS))) - 250.0f;
#endif
break;
This only lacks a calibration I guess as my AD8495 has 7°C offset at room temperature but otherwise rises as expected. I guess it is easy to get some 0.035V change in reference on the one or other side.

repetier, 2015-05-25:
It is in the work092 branch.

@kyrreaa Thanks for the info. So better watch to connect it right then if I understand you right. Not that I use it anyway - I have it just for testing.

simonkuehling, 2015-08-19:
@repetier, i just quickly backported your "case 61" code for testing into our RUMBA firmware fork and found a parenthesis mistake in your code for the AVRs - the 1.25V offset isn't included in the 10bit-to-12bit conversion which is resulting in wrong temperature calculations:

Your current code from work092 branch:
```
    case 61: // AD8495 1.25V Vref offset (like Adafruit 8495 breakout board)
        currentTemperatureC = ((float)currentTemperature * 1000.0f / (1024 << (2 - ANALOG_REDUCE_BITS))) - 250.0f;
        break;
```
Fixed:
```
    case 61: // AD8495 1.25V Vref offset (like Adafruit 8495 breakout board)
        currentTemperatureC = ((float)currentTemperature * 1000.0f / ((1024  - 250.0f) << (2 - ANALOG_REDUCE_BITS)));
        break;
```

simonkuehling, 2015-08-20:
@repetier - please skip my comment, i made false assumptions due to a wrong voltage measurement... sorry for that...

repetier, 2015-08-20:
Ok. Was just thinking about the problem and came also to the conclusion that formula was ok:-)