repetier/RepetierFirmware #403: GFX LCD displays strange chars

errolt, 2015-05-27:
Hi Repetier,

I have seen recently that the GFX LCD has strange chars in places. After the "Flow:" text there is a funny "9" symbol, Same after "Mul: 100%" then is a broken "9" symbol.

When moving in the menus the char most left (used to be an arrow?) is now a "y", when changing something like "Flow" the "y" becomes a "p".

I though this was because of my custom displays, but I now see it also happens on a stock RepRapDiscount Full Gfx display.

Seems to only happen on MEGA, not on DUE...

Thank you,
Errol

[image]
[image]

repetier, 2015-05-27:
Very strange. Must be from the additions to u8glib_ex.h I made. But all I did was adding new drivers, haven't deleted existing one and I have surely not changed the font (we use a custom font with additionl characters so in strange chars in menu are most likely from wrong font. Nut sure what the 9 could be - there should be a space or better no char at all. Maybe interpreted as utf-8). Will test tomorrow if I can reproduce the error. Luckily my only glcd hangs on a RAMPS/Mega. 

I guess replacing u8glib_ex.h with old version from 2 weeks ago or older would solve it.

errolt, 2015-05-27:
This has been around for a month or more. Not exactly sure when it started. It might be related to an update related to speeding up the LCD?

The strange thing is that the char after "Flow" moves the whole line on, with the last char of "Z 0.00" falling on the line.

repetier, 2015-05-27:
I haven't changed the lib for a very long time, only last week to add drivers. Except speed changes but I doubt that increased speed (which was I think for due not avr) would produce consistently existing chars especially if you consider that the content is rendered on the avr not the display.

Maybe more likely is a memory overwrite somewhere due to not enough memory, especially if you have a delta where buffers take up nearly all ram and new features took away some more ram. 

How much ram do you have when you start and does reducing line buffer by 1 make a change?

errolt, 2015-05-27:
I see this on two Cartesian printers. One with a driver that I added, ST7565P, and the other standard RepRap Discount.

I think Adruino said I have 3500 bytes remaining. Don't have the printer now to make sure...

repetier, 2015-05-27:
No problem 3500 byte is more then enough. Only for delta printers memory is a rare good. Will try then with my cartesian mega/glcd printer.

errolt, 2015-05-27:
I see in uimenu.h, line 209 and 211 there is a "\xfd" in the menu definition. It aligns with the funny chars. I see on my display...

errolt, 2015-05-27:
Seems it was added 5 March, "Bug fixes, improved ui"

errolt, 2015-05-28:
Removing the "\xfd" from uimenu.h removes the "9" chars. Still can't figure out about the "y" in the menu selection is coming from.

repetier, 2015-05-28:
The \xfd is correct. It shifts the following part by one pixel. In ui.cpp you should have

void u8PrintChar(char c)
{
    switch(c)
    {
    case 0x7E: // right arrow
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0x52);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case CHAR_SELECTOR:
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0xb7);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case CHAR_SELECTED:
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0xb6);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case 253:      //shift one pixel to right
        u8_tx++;
        break;
    default:
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, c);
    }
}
where you see the 253 is that special case and it prevents beeing printed, which works for me.

The "y" in the menu I can also not follow. I have the triangle that is expected. As you do not have the 253 part I guess that your ui.cpp is somehow not form the same version and maybe the same is for the special font. Could you try redownloading from github and try again. If it still persists, please mail me the configuration, so I can try it directly with yours or do you have more changes as well?

errolt, 2015-05-28:
I do have that section. My code base is a fresh pull yesterday.

Have you tried it on a MEGA? Maybe the MEGA's compiler does not like that convention? For some reason char value 253 never ends up in the string. 

That case is not executed.

jamesarm97, 2015-05-28:
FYI I have seen this on my firmware builds for a few months ( weird character next to the menu item instead of the arrow). Mega 2560

errolt, 2015-05-28:
Changing the case to:
case (char)253:      //shift one pixel to right
Fixes that "9"
Adding it to the rest fixes the menu "y" problem and probably other issues...

repetier, 2015-05-28:
Sounds like the problem comes from a new Arduino IDE. I use codeblocks a few years old. char is not defined in standard as signed or unsigned so at least that seems to come from a handling as signed while my compiler handles it unsigned.

What IDE version are you using?

errolt, 2015-05-28:
Arduino 1.6.1

repetier, 2015-05-28:
Ok, just could produce the errors with 1.6.3 as well including y problem. So it is the compiler having changed char handling. Guess I need to find all chars and set them explicitly signed or unsigned.

repetier, 2015-05-28:
Ok that was easy. Here correct routine

void u8PrintChar(char c)
{
    switch((uint8_t)c)
    {
    case 0x7E: // right arrow
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0x52);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case CHAR_SELECTOR:
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0xb7);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case CHAR_SELECTED:
        u8g_SetFont(&u8g, u8g_font_6x12_67_75);
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, 0xb6);
        u8g_SetFont(&u8g, UI_FONT_DEFAULT);
        break;
    case 253:      //shift one pixel to right
        u8_tx++;
        break;
    default:
        u8_tx += u8g_DrawGlyph(&u8g, u8_tx, u8_ty, c);
    }
}

also fixes y this way.

closed 2015-05-28 by repetier