repetier/RepetierFirmware #410: Arduino Due + DRV8825 configuration

zeadope, 2015-06-24:
I was wondering if these settings are correct.  So with step doubler 16000f / 320 steps/mm = 50 mm/s would be the printing speed. 
But what if I would like to print faster and have travel speed of 100 mm/s? 
Can the Due handle a faster step doubler frequency, since it uses a faster clk?

Settings below:

#define XAXIS_STEPS_PER_MM 320
#define YAXIS_STEPS_PER_MM 320

#define STEPPER_HIGH_DELAY 0 // 2 Z3D 2015-06-24 14:56
#define DIRECTION_DELAY 0
#define STEP_DOUBLER_FREQUENCY 16000 // 12000 Z3D 2015-06-24 14:55
#define ALLOW_QUADSTEPPING 1
#define DOUBLE_STEP_DELAY 1 // time in microseconds

I would be very grateful if anyone can provide adequate insight about this.

repetier, 2015-06-24:
Hi,

for due set double frequency to 90000 hz. Latest version 0.92.3 can 
handle 300000 and more in quadstepping mode.

-- 
Mit freundlichen Grüßen/Yours

Roland Littwin

----------------------------------------------
Hot-World GmbH & Co. KG
Knickelsdorf 42
47877 Willich
Deutschland

Tel.: 02154-606518
Fax: 02154-606519

Geschäftsführer: Marcus Littwin
Handelsregister: Amtsgericht Krefeld HRA 5325
Umsatzsteuer-Identifikationsnummer: DE213748632
Persönlich haftender Gesellschafter: Hot-World Medien GmbH mit Sitz in Willich
Handelsregister: Amtsgericht Krefeld HRB 10800

zeadope, 2015-06-24:
Hello repetier,

Thank you very much for your quick reply. I will test these settings shortly.

repetier, 2015-06-24:
If you get lost steps it is because DRV8825 might be a bit slower then the due, so then set
#define STEPPER_HIGH_DELAY 1

zeadope, 2015-06-24:
The 90000 setting caused I think do just about that. Trying delay 1.

repetier, 2015-06-24:
No the 90000 does not change delays only allows higher frequencies. If you have a extra delay you might need to lower the limit to 70000 maybe as you are then taking more time in interrupt.

zeadope, 2015-06-24:
It works now 
Frequency doubler 90000
High delay 1

On custom prusa i3. I've changed to bowden. And also set jerk to 35. On 20 jerk I was getting blobs.