repetier/RepetierFirmware #417: 0.92 - trouble getting MAX6675 thermocouple chip running

CupOfTea, 2015-08-10:
as discussed in  0.91 - trouble getting MAX31855 thermocouple chip running #381.
MAX6675 is working in marlin but in 0.92 reads ok on connect but not on subsequent temp requests.

guelz, 2015-08-10:
Sorry for highjacking! not the same but same trouble.
0.92.3 does not compile with MAX31855 support.
With:
#define MOTHERBOARD 402 (RADDS)
#define EXT0_TEMPSENSOR_TYPE 102
#define EXT0_TEMPSENSOR_PIN 10
I get an error:

C:\..\build6983828421676238874.tmp\HAL.h:154:41: error: 'DIOSS_PORT' was not declared in this scope
 #define  _WRITE(port, v)   do { if (v) {DIO ##  port ## _PORT -> PIO_SODR = DIO ## port ## _PIN; } else {DIO ##  port ## _PORT->PIO_CODR = DIO ## port ## _PIN; }; } while (0)
                                         ^
C:\..\build6983828421676238874.tmp\HAL.h:155:22: note: in expansion of macro '_WRITE'
 #define WRITE(pin,v) _WRITE(pin,v)
                      ^
C:\..\build6983828421676238874.tmp\Extruder.cpp:484:13: note: in expansion of macro 'WRITE'
             WRITE(SS, HIGH); 
and so on..

I dont really know what to do with it! any suggestions?

jamesarm97, 2015-08-10:
Doesn't the MAX31855 chip take more than one pin? It is an SPI device so it needs SPI Data, Clock and Chip select (the other devices it is complaining about). You are just defining one pin and probably the input pin as if it was an analog device. There is a MAX type sensor that takes a thermocouple and converts it to a analog output. I am using that on my RADDS board. You then select a type of MAX597 or something like that (from memory).

jamesarm97, 2015-08-10:
The notes from the other thread show these defines. The temp_0_pin is the CS chip select pin, the others you did not show are the clock and data pin and need to be defined also. Plus you may need the SUPPORT_MAX31855 defined.

#define SUPPORT_MAX31855
#define TEMP_0_PIN         33       // unused Z-MIN endstop pin as CS (chip select)
#define SCK_PIN            52
#define MISO_PIN           50

repetier, 2015-08-10:
@guelz Your problem is in this code
```
            SET_OUTPUT(SS);
            WRITE(SS, HIGH);
```
You haven't defined SS which was needed for AVR systems. Solution is to add
#define SS 10
in configuration.h to make it compile. Setting it twice to output does not hurt.

@CupOfTea The code responsebile is I think this in Extruder.cpp:
```
#ifdef SUPPORT_MAX6675
int16_t read_max6675(uint8_t ss_pin)
{
    int16_t max6675_temp = 0;
    HAL::spiInit(2);
    HAL::digitalWrite(ss_pin, 0);  // enable TT_MAX6675
    HAL::delayMicroseconds(1);    // ensure 100ns delay - a bit extra is fine
    max6675_temp = HAL::spiReceive(0);
    max6675_temp <<= 8;
    max6675_temp |= HAL::spiReceive(0);
    HAL::digitalWrite(ss_pin, 1);  // disable TT_MAX6675
    return max6675_temp & 4 ? 2000 : max6675_temp >> 3; // thermocouple open?
}
#endif
```
I can not say if it is working or not. I do not have such sensors. You could for testing add a write before return to see what you get from the chip and to see if it gets called at all. As you see the code is quite simple - just receiving 16 bit of data and dividing it by 8 (>> 3) if no error is present.

CupOfTea, 2015-08-11:
hiya

   stumbled on it...  from the datasheet

  "Forcing CS low immediately stops any conversion process."

   needs a conversion time....  datasheet says max 0.22 seconds

   Dunno how you wanna handle it,  but in the mean time I've added a 
simple delay and it works...

         cheers H

repetier, 2015-08-11:
You mean increasing the delay to 2200 ? Considering that we call this 10 times per second we would have 2.2s delay per second. Not really what I'd expect to work without new problems. Or did it work with with lower delay? Apperently th eoriginal author thought 100ns were enough delay.

CupOfTea, 2015-08-11:
I didn't try different times,  was just happy it was going.   .22s is the max in the datasheet the typical figure is .17s with a note that says   "Note 2: Guaranteed by design. Not production tested. " whatever that means.

perhaps declaring max6675_temp global and returning the previous value if the conversion is incomplete similar to this snippet from marlin

    #ifdef HEATER_0_USES_MAX6675
    #define MAX6675_HEAT_INTERVAL 250
    long max6675_previous_millis = MAX6675_HEAT_INTERVAL;
    int max6675_temp = 2000;

    int read_max6675()
    {
      if (millis() - max6675_previous_millis < MAX6675_HEAT_INTERVAL)
        return max6675_temp;
     
      max6675_previous_millis = millis();

repetier, 2015-08-12:
Ok, I have now also read the datasheet and came up with this solution:

#ifdef SUPPORT_MAX6675

int16_t read_max6675(uint8_t ss_pin)
{
    static millis_t last_max6675_read = 0;
    static int16_t max6675_temp = 0;
    if (HAL::timeInMilliseconds() - last_max6675_read > 230)
    {
        HAL::spiInit(2);
        HAL::digitalWrite(ss_pin, 0);  // enable TT_MAX6675
        HAL::delayMicroseconds(1);    // ensure 100ns delay - a bit extra is fine
        max6675_temp = HAL::spiReceive(0);
        max6675_temp <<= 8;
        max6675_temp |= HAL::spiReceive(0);
        HAL::digitalWrite(ss_pin, 1);  // disable TT_MAX6675
        last_max6675_read = HAL::timeInMilliseconds();
    }
    return max6675_temp & 4 ? 2000 : max6675_temp >> 3; // thermocouple open?
}
#endif

Would be great if you could test if it works. It is using the same trick as Marlin. Will be included in next update.

pkmcnc, 2015-08-12:
@repetier Great, when are you planning the next update? I'd try it too.

repetier, 2015-08-12:
Just replace in Extruders.cpp around line 1900 the extisting function with this one.

My printer behaves strange with other changes I've done so I want to sort them out before updating. Hoping the next days.

CupOfTea, 2015-08-12:
Nice....   It's working!

closed 2015-08-12 by CupOfTea

pkmcnc, 2015-08-15:
Yes it finally works, thanks!
Just the readings sometimes change a few degrees. Must be my MAX6675 or thermocouple issue?

BTW it's not obvious at all that you must choose D53 as sensor pin for MAX6675. Would be great if the configurator did it.

pkmcnc, 2015-08-15:
How often does it update the temp? Looks like each 2s, can it be faster?

CupOfTea, 2015-08-15:
configuration.h line 250ish

    // Analog input pin for reading temperatures or pin enabling SS for MAX6675

MAX6675_SS is defined in pins.h  so... 

    #define EXT0_TEMPSENSOR_PIN MAX6675_SS

repetier, 2015-08-16:
Update should be every 250ms. That is at least the delay we have included and normally we get called avery 100ms so real update delay is more like 300ms depending on accuracy of temp update being called.