repetier/RepetierFirmware #421: request pause: homing failed:

rhuuk420, 2015-08-23:
I am working on lighting up a new delta printer with the latest bits, pulled as of about 20 minutes from this post. I have confirmed it with the pull that gets packaged from the website, and the latest clone from here. (work 092).

 I am running into an issue with the g132 s1 command not committing the offsets to eeprom. I go through the steps as listed on your site, and watched the video. previously, this worked in older revs, I wanted to get this dilaed in with the latest for an alligator, but I get very strange results.

The host receives a pause and  prompts me to continue printing, even when homing.  I click on continue and
the logs return the following, which appear to be roughly inline with previous numbers (278, 56, respectively): 

13:06:10.488 : N63 G132 S1 *126
13:06:10.489 : ok 63
13:06:11.042 : Tower 1:331
13:06:11.048 : Tower 2:0
13:06:11.048 : Tower 3:76
13:06:11.120 : RequestPause:Homing failed!

EEPROM does not have these values committed. 

now, I can set these manually in eeprom, but the pause check is killing homing somehow, and I can't pin it down to what exactly.  below is my configuration.h  (abridged) as well. any assistance would be most valuable. (NOTE: misc settings moved closer to top as they are more readily accessible)

#define NUM_EXTRUDER 1
#define MOTHERBOARD 501
#define RFSERIAL SerialUSB
#include "pins.h"

// ################## EDIT THESE SETTINGS MANUALLY ################
#define MICROSTEP_MODES { 32,32,32,16,16 } // [1,2,4,8,16]
#define MOTOR_CURRENT_PERCENT { 65,65,65,65,65}
// ################ END MANUAL SETTINGS ##########################

#define FAN_BOARD_PIN -1

// ################# Misc. settings ##################

#define BAUDRATE 250000
#define ENABLE_POWER_ON_STARTUP 1
#define POWER_INVERTING 0
#define KILL_METHOD 1
#define ACK_WITH_LINENUMBER 1
#define WAITING_IDENTIFIER "wait"
#define ECHO_ON_EXECUTE 1
#define EEPROM_MODE 1
#define PS_ON_PIN ORIG_PS_ON_PIN

#define DRIVE_SYSTEM 3
#define XAXIS_STEPS_PER_MM 320
#define YAXIS_STEPS_PER_MM 320
#define ZAXIS_STEPS_PER_MM 320
#define EXTRUDER_FAN_COOL_TEMP 50
#define PDM_FOR_EXTRUDER 0
#define PDM_FOR_COOLER 1
#define DECOUPLING_TEST_MAX_HOLD_VARIANCE 20
#define DECOUPLING_TEST_MIN_TEMP_RISE 1
#define KILL_IF_SENSOR_DEFECT 0
#define RETRACT_ON_PAUSE 5
#define PAUSE_START_COMMANDS ""
#define PAUSE_END_COMMANDS ""
#define EXT0_X_OFFSET 0
#define EXT0_Y_OFFSET 0
#define EXT0_Z_OFFSET 0
#define EXT0_STEPS_PER_MM 698
#define EXT0_TEMPSENSOR_TYPE 1
#define EXT0_TEMPSENSOR_PIN TEMP_0_PIN
#define EXT0_HEATER_PIN HEATER_0_PIN
#define EXT0_STEP_PIN ORIG_E0_STEP_PIN
#define EXT0_DIR_PIN ORIG_E0_DIR_PIN
#define EXT0_INVERSE 0
#define EXT0_ENABLE_PIN E0_ENABLE_PIN
#define EXT0_ENABLE_ON 0
#define EXT0_MAX_FEEDRATE 50
#define EXT0_MAX_START_FEEDRATE 20
#define EXT0_MAX_ACCELERATION 5000
[... 321 lines cut ...]
#define CASE_LIGHT_DEFAULT_ON 1
#define UI_START_SCREEN_DELAY 1000
#define UI_DYNAMIC_ENCODER_SPEED 1
        /**
Beeper sound definitions for short beeps during key actions
and longer beeps for important actions.
Parameter is delay in microseconds and the secons is the number of repetitions.
Values must be in range 1..255
*/
#define BEEPER_SHORT_SEQUENCE 2,2
#define BEEPER_LONG_SEQUENCE 8,8
#define UI_SET_PRESET_HEATED_BED_TEMP_PLA 60
#define UI_SET_PRESET_EXTRUDER_TEMP_PLA   190
#define UI_SET_PRESET_HEATED_BED_TEMP_ABS 110
#define UI_SET_PRESET_EXTRUDER_TEMP_ABS   240
#define UI_SET_MIN_HEATED_BED_TEMP  30
#define UI_SET_MAX_HEATED_BED_TEMP 120
#define UI_SET_MIN_EXTRUDER_TEMP   170
#define UI_SET_MAX_EXTRUDER_TEMP   260
#define UI_SET_EXTRUDER_FEEDRATE 2
#define UI_SET_EXTRUDER_RETRACT_DISTANCE 3

#define NUM_MOTOR_DRIVERS 0

#endif

pkmcnc, 2015-08-23:
I confirm this.
Happens pretty often when homing a delta.
Just need to home again when it happens.

kyrreaa, 2015-08-23:
It's the code that verifies the endstops are switching correctly that is
detecting misbehaving switches. I have switched to hall sensors on mine and
have not had this problem since.

pkmcnc, 2015-08-23:
Ah, interesting. I use NC mechanical endstops. But ordered halls a few days back.
Really, this is intermittent issue, might be some electrical noise.

kyrreaa, 2015-08-24:
Hmm. Thinking about it my initial suggestion for this code does not take bounce into account at all.
I guess it should do the check as normal, then retest if failed after a certain time period and pass if the signal was corrected.
Right now it verifies that endstops have the correct state before and after every move is done. Since first contact before bounce can end a move the check could then see the bounce.

Really though, this is a workaround/detection for crappy or misbehaving endstops and was put into the code simply because it is better to warn than to blindly assume machine is homed after performing a set of moves.
It stops the printer from ramming the head into the printbed when you get a stuck switch also.
I’d like for the printer to issue a buffer flush and abort instead as continuing is really wrong for such a detection.

You can try adding a slight delay before any retest upon failure and you may get around the bouncing issue. However I think your repeatability on homing will be very poor.
I had 0.1mm repeatability with switches, but 0.01mm repeatability with halls since they have built in hysteresis.

Kyrre

From: pkmcnc 
Sent: Sunday, August 23, 2015 10:31 PM
To: repetier/Repetier-Firmware 
Cc: kyrreaa 
Subject: Re: [Repetier-Firmware] request pause: homing failed: (#421)

Ah, interesting. I use NC mechanical endstops. But ordered halls a few days back.
Really, this is intermittent issue, might be some electrical noise.

repetier, 2015-08-24:
That problem should be fixed with one of the latest updates. Problem was a test if after reaching the top all endstops are triggered. Also this was true when the move stopped, this might have changed until all sliders are at the top. Now I use a accumulating switch state and never had it happen again with the delta that I had with the same problem.

pkmcnc, 2015-08-24:
I updated the firmware probably 2weeks ago, hope you fixed it later.

One more detail probably connected with this issue: after (failed) homing before printing, the head starts printing high above the platform (the height looks like "Endstop distance after homing"). Anyone noticed that?

repetier, 2015-08-24:
Yes it was last week I fixed it.

repetier, 2015-08-24:
High printing can be related to it. You normally go down a few mm after homing. If it fails you don't so it starts higher.

pkmcnc, 2015-08-25:
I just uploaded the latest firmware. Once or twice it homed OK, but the next time same problem. First Y tower hit the switch, then Z, and then everything stopped. Next time it went OK.

repetier, 2015-08-25:
Please check Printer.cpp line 1157 should be

	Endstops::fillFromAccumulator();

which indicates it is using the fix. All I can say so far is that I had the problem with a new delta and the fix now works 100% for me. One important thing is that ENDSTOP_Z_BACK_MOVE is high enough to safely disable all endstops again. That part has no allocator so also all switch bouncing bust be finished by then. So if you have the fix try increasing that value.

pkmcnc, 2015-08-25:
Yes it is using the fix.
If you mean parameter "Go back after first hit", it is default 5mm (I prefer to use the config tool).
Homing stopped at the moment when 2 switches were hit and the last was not yet.

repetier, 2015-08-25:
If 3. switch is not hit it is in deed an error and triggering the error message is ok. In that case the question is why did it not make the distance to the endstop. Is your zlength completely wrong? Homing goes 1.5 times zlength up which normally is enough to reach the endstop.

pkmcnc, 2015-08-25:
I'll check it tomorrow, but probably it's just the case when 1.5*zlenght is not enough. Where can I increase this 1.5 ratio?

repetier, 2015-08-25:
Printer.cpp 1118:

void Printer::deltaMoveToTopEndstops(float feedrate)
{
    for (fast8_t i = 0; i < 3; i++)
        Printer::currentPositionSteps[i] = 0;
    Printer::stepsRemainingAtXHit = -1;
    Printer::stepsRemainingAtYHit = -1;
    Printer::stepsRemainingAtZHit = -1;
    setHoming(true);
    transformCartesianStepsToDeltaSteps(currentPositionSteps, currentDeltaPositionSteps);
    PrintLine::moveRelativeDistanceInSteps(0, 0, zMaxSteps * 1.5, 0, feedrate, true, true);

pkmcnc, 2015-08-25:
Thanks! I'll check that tomorrow. The point is that my delta has relatively small zheight now ~170mm, and now I'm sure it happens when the actuator is far from center so one tower is very low.

repetier, 2015-08-25:
You are right. Was already wondering how we can be > 1.0 but going off center moves it even more down and that needs to go up as well. So if your radius is > 85mm this can in deed happen. Will increase factor to 2.0. I think that should work on all deltas then.

pkmcnc, 2015-08-25:
Yes, the radius is 90mm. I think 2.0 should be OK for most deltas.
But actually I can imagine some rare cases when even 2.0 is not enough. I doubt that someone intentionally builds such delta, but if for some reason someone limits zheight and increases "Endstop distance after homing" accordingly, it can be 3 times or more.
The ultimate solution would be parameter "Tower height".

neildmd, 2015-10-23:
I posted this on the forums, but since it seems to be a bug I'm mirroring it here so there are more eyes on it. I'll update both when the solution is found.

I'm having the same problem when running G132 S1 on my Kossel. I'm using hall effect endstops. My actual build height is 317.5 mm and my horizontal radius is 90 mm.

I read in another thread that this issue can happen on tall printers with 300+ mm and (read here) radius greater than 85 mm. I changed horizontal radius to 84 mm just to see but it still fails. I also tried increasing the factor to 2.5, 3.0 and even 4.0 in Printer.cpp on line 1141 (firmware 0.92.5).

I set Z max length lower than actual, tried from 310 mm down to 290 mm as suggested elsewhere, also tried 330 mm (over-sized).

Every time I run G132 S1 there is always one endstop that does not trigger and I get the homing failed message.

17:15:31.757 : Tower 1:0
17:15:31.761 : Tower 2:132
17:15:31.761 : Tower 3:454
17:15:31.892 : RequestPause:Homing failed!

If I decrease Z max length in EEPROM below 300 mm the failed endstop will change so I know it's a soft issue considering homing works perfectly outside the G132 command.

17:36:08.769 : Tower 1:183
17:36:08.769 : Tower 2:0
17:36:08.769 : Tower 3:448
17:36:08.904 : RequestPause:Homing failed!

Increasing Z max length to 330 mm causes 2 endstops to fail:

18:25:19.164 : Tower 1:0
18:25:19.168 : Tower 2:0
18:25:19.168 : Tower 3:196
18:25:19.385 : RequestPause:Homing failed!

My endstop settings should be correct since I have no issues with homing at all other than this one with G132. I'm currently on 0.92.5 but I tried 0.92.3 and have the same problem. Here's the relevant sections of my configuration.h:

// ##########################################################################################
// ##                            Endstop configuration                                     ##
// ##########################################################################################

/* By default all endstops are pulled up to HIGH. You need a pullup if you
use a mechanical endstop connected with GND. Set value to false for no pullup
on this endstop.
*/
#define ENDSTOP_PULLUP_X_MIN false
#define ENDSTOP_PULLUP_Y_MIN false
#define ENDSTOP_PULLUP_Z_MIN false
#define ENDSTOP_PULLUP_X_MAX true
#define ENDSTOP_PULLUP_Y_MAX true
#define ENDSTOP_PULLUP_Z_MAX true

//set to true to invert the logic of the endstops
#define ENDSTOP_X_MIN_INVERTING false
#define ENDSTOP_Y_MIN_INVERTING false
#define ENDSTOP_Z_MIN_INVERTING true
#define ENDSTOP_X_MAX_INVERTING true
#define ENDSTOP_Y_MAX_INVERTING true
#define ENDSTOP_Z_MAX_INVERTING true

// Set the values true where you have a hardware endstop. The Pin number is taken from pins.h.

#define MIN_HARDWARE_ENDSTOP_X false
#define MIN_HARDWARE_ENDSTOP_Y false
#define MIN_HARDWARE_ENDSTOP_Z false
#define MAX_HARDWARE_ENDSTOP_X true
#define MAX_HARDWARE_ENDSTOP_Y true
#define MAX_HARDWARE_ENDSTOP_Z true

//If your axes are only moving in one direction, make sure the endstops are connected properly.
//If your axes move in one direction ONLY when the endstops are triggered, set ENDSTOPS_INVERTING to true here

//// ADVANCED SETTINGS - to tweak parameters

// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
#define X_ENABLE_ON 0
[... 449 lines cut ...]
#define DISTORTION_END_HEIGHT 1.5
/** If your corners measurement points are not measureable with given radius, you can
set this to 1. It then omits the outer measurement points allowing a larger correction area.*/
#define DISTORTION_EXTRAPOLATE_CORNERS 0

/* If your printer is not exactly square but is more like a parallelogramm, you can
use this to compensate the effect of printing squares like parallelogramms. Set the
parameter to then tangens of the deviation from 90° when you print a square object.
E.g. if you angle is 91° enter tan(1) = 0.017. If error doubles you have the wrong sign.
Always hard to say since the other angle is 89° in this case!
*/
#define FEATURE_AXISCOMP 0
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

/** \brief Experimental calibration utility for delta printers
 * Change 1 to 0 to disable
*/
#define FEATURE_SOFTWARE_LEVELING 0

/* Babystepping allows to change z height during print without changing official z height */
#define FEATURE_BABYSTEPPING 1
/* If you have a threaded rod, you want a higher multiplicator to see an effect. Limit value to 50 or you get easily overflows.*/
#define BABYSTEP_MULTIPLICATOR 1

repetier, 2015-10-25:
I have uploaded a new version.

I think the problem comes with new endstop handling. After G132 all endstops might be triggered when homing starts and that seems to confuse home. Endstops should not be triggered before and after homing. So I added a small down move after G132 and before homing what hopefully helps.

Good news is that at the time of the message G132 has already succeeded, it is only the homing afterwards that failes so you could just rehome then. Endstop correction stays stored correct.

2fast4myself, 2015-12-12:
Hi, 
can you please help me,.
a canot get my delta to start autoleveling,..

i get this  error in repetier host:
01:08:54.992 : Info:Autoleveling enabled
01:08:54.992 : Info:Autoleveling disabled
01:08:59.812 : Warning:Invalid delta coordinate - move ignored x:3555 y:3555 z:-84851
01:08:59.828 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
01:08:59.828 : Warning:moveRelativeDistanceInSteps / queueDeltaMove returns error

and here are my settings in config file for z-probing
// #################### Z-Probing #####################

#define FEATURE_Z_PROBE 1
#define Z_PROBE_BED_DISTANCE 10
#define Z_PROBE_PIN ORIG_Z_MIN_PIN
#define Z_PROBE_PULLUP 1
#define Z_PROBE_ON_HIGH 1
#define Z_PROBE_X_OFFSET 0
#define Z_PROBE_Y_OFFSET 0
#define Z_PROBE_WAIT_BEFORE_TEST 0
#define Z_PROBE_SPEED 2
#define Z_PROBE_XY_SPEED 150
#define Z_PROBE_SWITCHING_DISTANCE 1
#define Z_PROBE_REPETITIONS 1
#define Z_PROBE_HEIGHT -0.9
#define Z_PROBE_START_SCRIPT ""
#define Z_PROBE_FINISHED_SCRIPT ""
#define FEATURE_AUTOLEVEL 1
#define Z_PROBE_X1 -69.28
#define Z_PROBE_Y1 -40
#define Z_PROBE_X2 69.28
#define Z_PROBE_Y2 -40
#define Z_PROBE_X3 0
#define Z_PROBE_Y3 80
#define FEATURE_AXISCOMP 1
#define AXISCOMP_TANXY 0
#define AXISCOMP_TANYZ 0
#define AXISCOMP_TANXZ 0

i use firmware v 0.92.3
thank you

repetier, 2015-12-12:
This error occurs if you are moving to invalid coordinates. If it happens during autoleveling it normally means one of your probing points require a move outside your allowed printing circle. Go to eeprom and move the points more inside and check if they match the one from configuration.h. Maybe they are set to other values from first upload.

2fast4myself, 2015-12-12:
Hi,
thank you for the information.

This is what i tried to fix the error.
- correct every eeprom setting and then upload the firmware with corrected settings
- the settings are now 100 % correct and probing is in the printing radius
- try to set larger printing area 
- reducing the points to be in the diameter of 50 (my delta can print 300)

i tried probing with G30 command and get the same error 
21:02:10.795 : Warning:Invalid delta coordinate - move ignored x:0 y:10666 z:-92444
21:02:10.795 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
21:02:10.811 : Warning:moveRelativeDistanceInSteps / queueDeltaMove returns error

this is weird as my delta tries to move to a different position and probe,..

i tried to upload the new firmware and probing an leveling works fine,.. but with the new firmware i get pause on homing and cannot even print anything. 
 
i have tried fixing the pause on home problem but when i correct zmax length in the 1.5 *  margin it doesn't work.. i still get the pause

please help ,... 
:) 
i'm on my last drop of idea what is wrong...

thank you
P.S. sorry for my bad english

repetier, 2015-12-13:
The position is reported in steps so if you divide them by your steps per mm you get the position he wants to reach. Can not compute that as I do not know value. Maybe that gives a hint. Remember that the z probe offset is subtracted from target coordinates. Single probes should be started near bottom and not from top. Top is bad since endstops may be hit when going to side.

I'm always assuming latest 0.92 version, since this is the only version I know how it currently behaves.

The home pause happens if the endstops do not trigger the way we expect it. Since I do not know what exactly happens I can not say more here. It must reach all endstops, release all endstops and hit them all again. Then you should move 10mm or more down to allow some small moves at top level without hitting endstops.

2fast4myself, 2015-12-13:
steps per mm count is 177,778.
i understand for the y axis witch is the offset for the probe, but do not understand the z axis.
is the z coordinate set by me ?

in the v 0.91 the delta homed as described and works but probing does not work.
in v 0.92 it homes and moves down to set position (5mm).
- no mather the position it homes from the results are the same
- the error it gives is request pause: homing failed
I searched the internet for resolving the problem but everything i tried is not producing results.

attached is my config file
[Configuration.txt](https://github.com/repetier/Repetier-Firmware/files/60391/Configuration.txt)

repetier, 2015-12-13:
The z comes from probing. Since we do not know how deep we have to go, we go deeper then printer height and wait for the hit. Difference is then probe distance. So that value is ok.

What exactly is the homing error? If you say it moves 5 mm down, ok. But then it has to go up and then go down 1mm again from your config. Does that happen or is that where it stops working. What does M119 say after homing exits? What kind of endstops do you use? You could change homing code. In 0.91 there was no test for failed endstops which is why it never stops. That does not mean it worked as supposed. If you have mechanical endstops 1mm return might be not enough to untrigger endstops.

2fast4myself, 2015-12-13:
i have now changed the values and set them to
#define ENDSTOP_X_BACK_MOVE 10
#define ENDSTOP_Y_BACK_MOVE 10
#define ENDSTOP_Z_BACK_MOVE 10
#define ENDSTOP_X_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Y_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Z_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_X_BACK_ON_HOME 10
#define ENDSTOP_Y_BACK_ON_HOME 10
#define ENDSTOP_Z_BACK_ON_HOME 10
#define ALWAYS_CHECK_ENDSTOPS 1

after homing it hits every endstop and goes down 10 mm and then it stops.
i use mechanical endstops NO signal when on GND,

witch parameter do i change for them to go farther then 1mm 
thank you

repetier, 2015-12-13:
#define ENDSTOP_Z_BACK_ON_HOME 10
is for the second time down.
What was M119 saying when stopped?

2fast4myself, 2015-12-13:
thank you for helping me,..

14:22:58.319 : RequestPause:Homing failed!
14:22:58.335 : X:0.00 Y:0.00 Z:387.602 E:0.0000
14:22:58.506 : Resend:20
14:23:07.445 : N29 M119 *49
14:23:07.445 : end stops hit: x_max:L y_max:L z_max:L Z-probe state:L

i tested restoring just the void Printer::homeZAxis() // Delta z homing function in printer.cpp to the old v0.91
it seames to work and now my delta homes as it should

somwhere in here lays the problem 
void Printer::homeZAxis() // Delta z homing
{
	bool homingSuccess = false;
	Endstops::resetAccumulator();
    deltaMoveToTopEndstops(Printer::homingFeedrate[Z_AXIS]);
	// New safe homing routine by Kyrre Aalerud
	// This method will safeguard against sticky endstops such as may be gotten cheaply from china.
	// This can lead to headcrashes and even fire, thus a safer algorithm to ensure the endstops actually respond as expected.
	Endstops::report();
	// Check that all endstops (XYZ) were hit
	Endstops::fillFromAccumulator();
	if (Endstops::xMax() && Endstops::yMax() && Endstops::zMax()) {
		// Back off for retest
		PrintLine::moveRelativeDistanceInSteps(0, 0, axisStepsPerMM[Z_AXIS] * -ENDSTOP_Z_BACK_MOVE, 0, Printer::homingFeedrate[Z_AXIS]/ENDSTOP_X_RETEST_REDUCTION_FACTOR, true, false);
                Endstops::report();
		// Check for proper release of all (XYZ) endstops
		if (!(Endstops::xMax() || Endstops::yMax() || Endstops::zMax())) {
			// Rehome with reduced speed
			Endstops::resetAccumulator();
		    deltaMoveToTopEndstops(Printer::homingFeedrate[Z_AXIS] / ENDSTOP_Z_RETEST_REDUCTION_FACTOR);
		    Endstops::fillFromAccumulator();
			Endstops::report();
			// Check that all endstoips (XYZ) were hit again
			if (Endstops::xMax() && Endstops::yMax() && Endstops::zMax()) {
				homingSuccess = true; // Assume success in case there is no back move
#if defined(ENDSTOP_Z_BACK_ON_HOME)
				if(ENDSTOP_Z_BACK_ON_HOME > 0) {
					PrintLine::moveRelativeDistanceInSteps(0, 0, 2 * axisStepsPerMM[Z_AXIS] * -ENDSTOP_Z_BACK_ON_HOME * Z_HOME_DIR,0,homingFeedrate[Z_AXIS], true, false);
					Endstops::report();
					// Check for missing release of any (XYZ) endstop
					if (Endstops::xMax() || Endstops::yMax() || Endstops::zMax()) {
						homingSuccess = false; // Reset success flag
					}
				}
#endif

i assume the micro is to fast for the end stops to disengage. i'm not a programmer so this is my best guess.
correct me please.

thank you for your help

kyrreaa, 2015-12-13:
Isn't this where the endstop tests are added to avoid assuming moves went
correctly?
You may have a sticky endstop if it paused on homing. The bad or sticking
endstop isse us why the extra tests wewe added btw, to avoid smashing
nozzle into bed.

2fast4myself, 2015-12-13:
yes these are the tests, but i have been using my delta for a year now and had no problems with endstops,.. 
endstaps are from Sainsmart

kyrreaa, 2015-12-13:
Endstop switches do experience wear. Maby printing the endstop states and
expected state could help troubleshoot such stops?

simonkuehling, 2015-12-17:
@repetier, @2fast4myself - i am having the same issues with homing here. it is 0.92.6 loaded from the configuration tool.

My setup is a Delta, tested with both Hall-Endstops and mechanical ones.
This is what i get for a G28:

```
Send: G28
Recv: ok 0
Recv: Info:Autoleveling disabled
Recv: endstops hit: x_max:L y_max:L z_max:L Z-probe state:L                ----> REPORT #1
Recv: endstops hit: x_max:L y_max:L z_max:L Z-probe state:L                ----> REPORT #2
Recv: endstops hit: x_max:H y_max:H z_max:H Z-probe state:L                ----> REPORT #3
Recv: endstops hit: x_max:H y_max:H z_max:H Z-probe state:L                ----> REPORT #4
Recv: endstops hit: x_max:H y_max:H z_max:H Z-probe state:L                ----> REPORT #5
Recv: RequestPause:Homing failed!
Recv: FlowMultiply:100
Recv: X:0.00 Y:0.00 Z:271.406 E:0.0000
Recv: Info:Autoleveling enabled
```

I modified <code>homeZAxis()</code> in Printer.cpp with some more endstop reports for debugging:

```
void Printer::homeZAxis() // Delta z homing
{
	bool homingSuccess = false;
  Endstops::report();                                                               // REPORT #1
	Endstops::resetAccumulator();
  Endstops::report();                                                               // REPORT #2
    deltaMoveToTopEndstops(Printer::homingFeedrate[Z_AXIS]);
	// New safe homing routine by Kyrre Aalerud
	// This method will safeguard against sticky endstops such as may be gotten cheaply from china.
	// This can lead to headcrashes and even fire, thus a safer algorithm to ensure the endstops actually respond as expected.
	Endstops::report();                                                               // REPORT #3
	// Check that all endstops (XYZ) were hit
	Endstops::fillFromAccumulator();
  Endstops::report();                                                               // REPORT #4
	if (Endstops::xMax() && Endstops::yMax() && Endstops::zMax()) {
		// Back off for retest
    PrintLine::moveRelativeDistanceInSteps(0, 0, axisStepsPerMM[Z_AXIS] * -ENDSTOP_Z_BACK_MOVE, 0, Printer::homingFeedrate[Z_AXIS]/ENDSTOP_X_RETEST_REDUCTION_FACTOR, true, false);
    Endstops::report();                                                               // REPORT #5
		// Check for proper release of all (XYZ) endstops
    if (!(Endstops::xMax() || Endstops::yMax() || Endstops::zMax())) {
[...]
```

My guess is that the check for endstops after moving down for retest is either to early (e.g. does not wait for the movement first) or the endstop status' did not get updated properly upon release...

I am happy to help with debugging where i can!

kyrreaa, 2015-12-17:
> I modified homeZAxis() in Printer.cpp with some more endstop reports for

There may be some lacking fill from enumerator or such?
17. des. 2015 12.11 p.m.

repetier, 2015-12-18:
Ok, found the problem. I changed the interrupt handler to only update endstop status when always check is enabled or it is explicitly wanted. In th ehoming routine the back moves did not explicitly define endstop tests so endstop status was not updated. Modified the parameter to true and now it should work as expected.

simonkuehling, 2015-12-18:
awesome - i was just about telling that an additional repeating
```
endstops::update();
endstops::update();
endstops::update();
```
before the enstop release checks did work as a temporary workaround.

unfortunately the current 0.92 branch has a compile issue preventing a quick test of the fix - if i may put into this thread quickly:

```
Verifying...
In file included from ui.cpp:20:0:
Repetier.h:421: error: operator '>' has no left operand
 #if FAN_THERMO_THERMISTOR_PIN > -1 && FAN_THERMO_PIN > -1
                               ^
In file included from Repetier.h:784:0,
                 from ui.cpp:20:
Extruder.h:297: error: operator '>' has no left operand
 #if FAN_THERMO_PIN > -1
                    ^
In file included from Repetier.h:813:0,
                 from ui.cpp:20:
Printer.h:341: error: operator '>' has no left operand
 #if FAN_THERMO_PIN > -1
                    ^
In file included from ui.cpp:20:0:
Repetier.h:827: error: #if with no expression
 #if FEATURE_FAN2_CONTROL
                         ^
operator '>' has no left operand
```
```

repetier, 2015-12-18:
You didn't update the config. Just put it through config tool for update. We have now a thermistor controlled fan and a second fan added and these introduced new config variables. Exactly the one you get errors with since they are missing.

simonkuehling, 2015-12-18:
hmm, seems like the configuration tool did not add the new options and therefor included them without proper default values in the code (the defines were there, but no values set)

i quickly fixed this by hand for me...

Regarding your homeing fix: Yep, it is working flawlessly now, thumbs up!

repetier, 2015-12-18:
Mhh. Reuploaded the tool to be sure. Used it locally to convert it and it worked so it must be that I forgot to upload the new tool and only updated firmware. Should contain something like this

/* You can have one additional fan controlled by a temperature. You can set
   set at which temperature it should turn on and at which it should reach max. speed.
*/
#define FAN_THERMO_PIN -1
#define FAN_THERMO_MIN_PWM 128
#define FAN_THERMO_MAX_PWM 255
#define FAN_THERMO_MIN_TEMP 45
#define FAN_THERMO_MAX_TEMP 60
// Analog pin number or channel for due boards
#define FAN_THERMO_THERMISTOR_PIN -1
#define FAN_THERMO_THERMISTOR_TYPE 1

closed 2015-12-18 by repetier

simonkuehling, 2015-12-18:
The defines and variable names were in there, but none of them had a value next to it...

anyway - now that the homing is working, there is one catch i found. Could it be that the ENDSTOP_Z_BACK_ON_HOME movement is accidentally doubled? There is a factor of 2 in the code and it actually moves twice as much as configured as well:

<code>Printer.cpp</code> line 1287:
```
PrintLine::moveRelativeDistanceInSteps(0, 0, 2 * axisStepsPerMM[Z_AXIS] * -ENDSTOP_Z_BACK_ON_HOME * Z_HOME_DIR,0,homingFeedrate[Z_AXIS], true, true);
```

i think this should be
```
PrintLine::moveRelativeDistanceInSteps(0, 0, axisStepsPerMM[Z_AXIS] * -ENDSTOP_Z_BACK_ON_HOME * Z_HOME_DIR,0,homingFeedrate[Z_AXIS], true, true);
```

repetier, 2015-12-18:
The would mean I hadn't uploaded the javascript or you hadn't reloaded the webpage so the variables were missing.

Regarding the double distance you are right. 2 * will double that which is not what should happen. Have corrected it but now leveling is off one distance of course.