repetier/RepetierFirmware #422: Z babystepping

maralb, 2015-08-25:
with the new update
delta Printer
not working well
Please check

the towers are not together control

repetier, 2015-08-25:
What is not working well? What is happening? How to produce the problem?

maralb, 2015-08-25:
the towers do not respond together
do not respect the positive and negative values
only goes up not parallel

have not activated "B Coating"
I do not need this for my delta

repetier, 2015-08-25:
I had a similar problem while testing. The 3 sliders moved into the last set direction. Problem was the stepper driver needing some time to make the direction change taking effect. In my case I had to set
#define DIRECTION_DELAY 50

lower values did not work for me. Maybe you have the same problem. Value depends on driver you use.

maralb, 2015-08-25:
still it does not work with value 50 DIRECTION_DELAY
continued to use the old version 0.92.3

repetier, 2015-08-25:
50 was an example which is needed for my stepper. That does not mean it is enough for all.

BTW: Do you change babysteps before printing or while printing for testing?

maralb, 2015-08-25:
in version 0.92.3 have always zero in this entry DIRECTION_DELAY
I only correct when printed with the baby for the first layer extrusion
I find it very useful for the test print.

I also noticed in version 0.92.4, pausing to press the SD Baby works well.
otherwise it does not comply with the movements of the towers
p.s.
I use Alligator Board

thanks

errolt, 2015-08-29:
I have the same problem. When not printing, and at x0 y0 zmax, baby stepping works, one direction goes up, other goes down. But during a print both directions move the head down. And it shifts the head to one side, and it breaks xy movement in that xy movement is not parallel with the bed anymore.
I also see things like "bsdir:0" and " bsdir:1" coming from the printer.

BTW, this is on a DUE. Don't know if that makes a difference...

errolt, 2015-09-04:
Looking at Printer::zBabystep()

By the looks of it it moves ALL the carriages for a delta one step up or down. Is this correct?

This will work at X=0, Y=0, but not at ANY other location.

repetier, 2015-09-04:
Yes zBabystep for delta MUST increase all steppers up or down. That is the plan. This does not depend on the x/y position. Theproblem is that all carriages have initially a different direction so we might need to change direction do a step and change back to old direction. And if this does not get fast enough throught to the drivers they might do something different.

Previous solution waited for a move that moved all 3 into the right direction and then only added additional steps. The drawback is, that only a few moves move all into same direction.

errolt, 2015-09-07:
Sorry, my mistake. For a pure Z move all the arm angles stay fixed and thus all the carriages must move the same distance...

errolt, 2015-09-24:
I have tried a DIRECTION_DELAY of 50, 150, 1000 and none remove this problem.

errolt, 2015-09-24:
I have moved the zBabystep() function higher and it fixed the problem. Looks like the babystep count display on the display is wrong though. Display reads 60mm to move 20mm...

My modified code from line 2034:
            if (!stepsPerSegRemaining)
            {
                if (cur->numDeltaSegments)
                {
                  if(FEATURE_BABYSTEPPING && Printer::zBabystepsMissing)
                    {
                        // execute a extra babystep
                        Printer::zBabystep();
                    }

                  // Get the next delta segment
                    curd = &cur->segments[--cur->numDeltaSegments];

                    // Initialize bresenham for this segment (numPrimaryStepPerSegment is already correct for the half step setting)
                    cur->error[X_AXIS] = cur->error[Y_AXIS] = cur->error[Z_AXIS] = cur->numPrimaryStepPerSegment >> 1;

                    // Reset the counter of the primary steps. This is initialized in the line
                    // generation so don't have to do this the first time.
                    stepsPerSegRemaining = cur->numPrimaryStepPerSegment;

                    // Change direction if necessary
                    Printer::setXDirection(curd->dir & X_DIRPOS);
                    Printer::setYDirection(curd->dir & Y_DIRPOS);
                    Printer::setZDirection(curd->dir & Z_DIRPOS);
#if defined(DIRECTION_DELAY) && DIRECTION_DELAY > 0
                    HAL::delayMicroseconds(DIRECTION_DELAY);
#endif

              }
              else
                    curd = 0;// Release the last segment

repetier, 2015-09-24:
I will checkt. Should not make a big difference since it should restore direction, which also happens here now after babystepping. So even a wrong direction would get restored this way.

errolt, 2015-09-24:
Are you sure that getXDirection() returns the correct value? It might be inverted? Same with getYDirection() and getZDirection()

repetier, 2015-09-24:
static INLINE void setXDirection(bool positive)
    {
        if(positive)
        {
            WRITE(X_DIR_PIN,!INVERT_X_DIR);
        }
        else
        {
            WRITE(X_DIR_PIN,INVERT_X_DIR);
        }
    }

    static INLINE bool getXDirection()
    {
        return((READ(X_DIR_PIN) != 0) ^ INVERT_X_DIR);
    }

set(get()) should not change stored bit. Here is what I get with the above logic

X dir pin  / INVERT_X_DIR / get returns / set sets
0 / 0 /  0 /  0
0 / 1 / 1 / 0
1 / 0 / 1 / 1
1 / 1 / 0 / 1 

That is the logic I see. You see in theory all combinations set the direction back to original value. If you see an error in the logic, please let me know.

errolt, 2015-09-25:
The logic is definitely sound. Then I don't know why it works when not relying on the restore of the states... :)