repetier/RepetierFirmware #423: UI_ACTION_KILL broken

errolt, 2015-08-31:
Hi Repetier,

I use the UI_ACTION_KILL assigned to a button on one printer, and assigned to a matrix button on the other printer.

But UI_ACTION_KILL has started acting funny in the last few months. I though it was only on the matrix setup but i now see that it also exists on the normal button.

After a reboot from a UI_ACTION_KILL the matrix buttons do not work, I have to remove power and restart the printer.
On the DUE delta with UI_ACTION_KILL connected to a normal button I now saw that the click button on the encoder also does not work, and commands from Repetier host to power up the heaters does wait for the heaters to power up, but the heaters never turn on...

I have also seen this behaviour after a thermocoulpe disconnect event related reboot, that the matrix buttons do not work anymore...

The printer with the matrix buttons is MEGA2550 based, the printer with the normal button is DUE based.

Thank you,
Errol

repetier, 2015-08-31:
That only calls void Commands::emergencyStop() and there are two modes. One does a simple reset so everything works as before (KILL_METHOD 1). The second method disables everything and then hangs around in an endless loop doing nothing. There you would still see lcd but no button will work. Only after a real reset it will start as before.

After a reset everything gets initialized as on first run. All buttons would work as expected.

errolt, 2015-08-31:
The thing is that when I press the kill button then I see the screen blank, then I see the splash screen, then the menu. Then the rotary encoder works, and the printer does talk to Repetier host over USB, but some things are broken.

Just looked, I do use KILL_METHOD 1