repetier/RepetierFirmware #427: Ramps-fd with Due doesn't run on power-up unless reset is pressed

jamesarm97, 2015-09-20:
Running dev 0.92.3
I have two boards that will not run (LCD doesn't show anything) unless the reset button is pressed after a power up. Is there a race condition or something that has been fixed or a delay needed somewhere? After pressing the Reset button everything works.

repetier, 2015-09-21:
I have seen Arduino-Due clones that cause the due to crash on first power up. Even the simple blink program crashes on them on first power up. Only after a reset they start working normally, which sounds exactly like your problem. I think it might start while not powered fully and then cause a crash, because once they work they behave normally. If you unpower and directly power them they sometimes work on first try but letting them rest a while and then powering again makes them crash again.

jamesarm97, 2016-02-05:
Fixed by adding a 10uF capacitor across the reset button lines. Just had it happen on a new board with the RADDS board this week.Still don't know if it is an LCD init problem or the board isn't starting up at all but I am thinking the latter since I have read many messages about the fix being a capacitor.

closed 2016-02-05 by jamesarm97

repetier, 2016-02-05:
It is the board not starting at all. I guess the capacitor changes reset timings so due start up correctly. I guess that is the difference between the working and not working boards. Maybe just cheap capacitors with too much tolerance on the wrong side.