repetier/RepetierFirmware #433: G30: Invalid delta coordinate - move ignored x:-24000 y:-9600 z:-65077

Links2004, 2015-10-25:
I try to get Z-probe / Bed Map to working but if i try to Z probe at negative X or Y coordinates
the system starts to getting real strange reactions.

Firmware: 73abf2600318661df4dbd23ed26bf06a413dbd05
Host: 1.5.6
Printe: Delta self build

[image]

log
```
16:49:12.385 : N16 G28 *4
16:49:13.160 : X:0.00 Y:0.00 Z:213.900 E:0.0000
16:49:22.062 : N20 G1 X-150 Y-60 Z213.9 F4800 *62
16:49:22.071 : N21 G30 P1 *72
16:49:22.167 : N22 G1 X-101.67 Y-60 Z213.9 F4800 *23
16:49:30.449 : Warning:Invalid delta coordinate - move ignored x:-24000 y:-9600 z:-65077
16:49:30.500 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
16:49:30.500 : Warning:moveRelativeDistanceInSteps / queueDeltaMove returns error
16:49:30.501 : Error:Z-probe failed
16:49:30.501 : N23 G30 P0 *75
```

config:
```
16:59:16.832 : EPR:0 1028 0 Language
16:59:16.842 : EPR:2 75 115200 Baudrate
16:59:16.856 : EPR:3 129 -0.120 Filament printed [m]
16:59:16.863 : EPR:2 125 3575 Printer active [s]
16:59:16.871 : EPR:2 79 0 Max. inactive time [ms,0=off]
16:59:16.880 : EPR:2 83 360000 Stop stepper after inactivity [ms,0=off]
16:59:16.892 : EPR:3 11 160.0000 Steps per mm
16:59:16.896 : EPR:3 23 80.000 Max. feedrate [mm/s]
16:59:16.906 : EPR:3 35 80.000 Homing feedrate [mm/s]
16:59:16.919 : EPR:3 39 20.000 Max. jerk [mm/s]
16:59:16.935 : EPR:3 133 -250.000 X min pos [mm]
16:59:16.944 : EPR:3 137 -250.000 Y min pos [mm]
16:59:16.954 : EPR:3 141 0.000 Z min pos [mm]
16:59:16.963 : EPR:3 145 250.000 X max length [mm]
16:59:16.974 : EPR:3 149 250.000 Y max length [mm]
16:59:16.983 : EPR:3 153 213.900 Z max length [mm]
16:59:16.995 : EPR:3 59 100.000 Acceleration [mm/s^2]
16:59:17.007 : EPR:3 71 100.000 Travel acceleration [mm/s^2]
16:59:17.027 : EPR:3 881 560.000 Diagonal rod length [mm]
16:59:17.038 : EPR:3 885 257.550 Horizontal rod radius at 0,0 [mm]
16:59:17.052 : EPR:3 925 300.000 Max printable radius [mm]
16:59:17.060 : EPR:1 891 70 Segments/s for travel
16:59:17.071 : EPR:1 889 180 Segments/s for printing
16:59:17.076 : EPR:1 893 120 Tower X endstop offset [steps]
16:59:17.079 : EPR:1 895 402 Tower Y endstop offset [steps]
16:59:17.084 : EPR:1 897 0 Tower Z endstop offset [steps]
16:59:17.092 : EPR:3 901 210.000 Alpha A(210):
16:59:17.096 : EPR:3 905 330.000 Alpha B(330):
16:59:17.101 : EPR:3 909 90.000 Alpha C(90):
16:59:17.105 : EPR:3 913 0.000 Delta Radius A(0):
16:59:17.108 : EPR:3 917 0.000 Delta Radius B(0):
16:59:17.117 : EPR:3 921 0.000 Delta Radius C(0):
16:59:17.124 : EPR:3 933 0.000 Corr. diagonal A [mm]
16:59:17.127 : EPR:3 937 0.000 Corr. diagonal B [mm]
16:59:17.131 : EPR:3 941 0.000 Corr. diagonal C [mm]
16:59:17.136 : EPR:3 1024 0.000 Coating thickness [mm]
16:59:17.139 : EPR:3 808 25.000 Z-probe height [mm]
16:59:17.143 : EPR:3 929 10.000 Max. z-probe - bed dist. [mm]
16:59:17.149 : EPR:3 812 5.000 Z-probe speed [mm/s]
16:59:17.157 : EPR:3 840 150.000 Z-probe x-y-speed [mm/s]
16:59:17.160 : EPR:3 800 0.000 Z-probe offset x [mm]
16:59:17.164 : EPR:3 804 0.000 Z-probe offset y [mm]
16:59:17.175 : EPR:3 816 -190.000 Z-probe X1 [mm]
16:59:17.179 : EPR:3 820 -70.000 Z-probe Y1 [mm]
16:59:17.186 : EPR:3 824 150.000 Z-probe X2 [mm]
16:59:17.190 : EPR:3 828 190.000 Z-probe Y2 [mm]
16:59:17.193 : EPR:3 832 -70.000 Z-probe X3 [mm]
16:59:17.197 : EPR:3 836 190.000 Z-probe Y3 [mm]
16:59:17.203 : EPR:3 1036 0.000 Z-probe bending correction A [mm]
16:59:17.207 : EPR:3 1040 0.000 Z-probe bending correction B [mm]
16:59:17.210 : EPR:3 1044 0.000 Z-probe bending correction C [mm]
16:59:17.219 : EPR:0 880 0 Autolevel active (1/0)
16:59:17.223 : EPR:0 106 1 Bed Heat Manager [0-3]
16:59:17.227 : EPR:0 107 255 Bed PID drive max
16:59:17.231 : EPR:0 124 80 Bed PID drive min
16:59:17.235 : EPR:3 108 196.000 Bed PID P-gain
16:59:17.240 : EPR:3 112 33.020 Bed PID I-gain
16:59:17.243 : EPR:3 116 290.000 Bed PID D-gain
16:59:17.249 : EPR:0 120 240 Bed PID max value [0-255]
16:59:17.257 : EPR:0 1020 0 Enable retraction conversion [0/1]
16:59:17.260 : EPR:3 992 6.000 Retraction length [mm]
16:59:17.266 : EPR:3 1000 40.000 Retraction speed [mm/s]
16:59:17.270 : EPR:3 1004 0.000 Retraction z-lift [mm]
16:59:17.274 : EPR:3 1008 2.000 Extra extrusion on undo retract [mm]
16:59:17.280 : EPR:3 1016 20.000 Retraction undo speed
16:59:17.287 : EPR:3 200 213.000 Extr.1 steps per mm
16:59:17.291 : EPR:3 204 30.000 Extr.1 max. feedrate [mm/s]
16:59:17.302 : EPR:3 208 10.000 Extr.1 start feedrate [mm/s]
16:59:17.306 : EPR:3 212 4000.000 Extr.1 acceleration [mm/s^2]
16:59:17.311 : EPR:0 216 1 Extr.1 heat manager [0-3]
16:59:17.314 : EPR:0 217 140 Extr.1 PID drive max
16:59:17.323 : EPR:0 245 60 Extr.1 PID drive min
16:59:17.327 : EPR:3 218 24.0000 Extr.1 PID P-gain/dead-time
16:59:17.332 : EPR:3 222 0.8800 Extr.1 PID I-gain
16:59:17.336 : EPR:3 226 80.0000 Extr.1 PID D-gain
16:59:17.339 : EPR:0 230 255 Extr.1 PID max value [0-255]
16:59:17.346 : EPR:2 231 0 Extr.1 X-offset [steps]
16:59:17.352 : EPR:2 235 0 Extr.1 Y-offset [steps]
16:59:17.356 : EPR:2 290 0 Extr.1 Z-offset [steps]
16:59:17.360 : EPR:1 239 1 Extr.1 temp. stabilize time [s]
16:59:17.363 : EPR:1 250 150 Extr.1 temp. for retraction when heating [C]
16:59:17.367 : EPR:1 252 0 Extr.1 distance to retract when heating [mm]
16:59:17.371 : EPR:0 254 255 Extr.1 extruder cooler speed [0-255]
16:59:17.375 : EPR:3 246 0.000 Extr.1 advance L [0=off]
```

repetier, 2015-10-25:
You machine fits into big machine with no 64 bit math. That transformation had a missing test for z probing causing the error. Please try the new version where I think it is corrected now.

closed 2015-10-25 by repetier

Links2004, 2015-10-26:
the error/warning is gone.

but the printer ignores the top endstop at its way up.
now i need to re calibrate the endstop offset, hall effects are bent.
the endstops are tested and working (Homing is ok, M119 too).

log:
```
18:02:40.640 : N44 G30 P1 *75
18:02:40.640 : N45 G1 X-126.67 Y-60 Z213.9 F4800 *19
18:02:40.640 : N46 G30 P0 *72
18:02:40.641 : N47 G1 X-73.33 Y-60 Z213.9 F4800 *33
18:02:40.641 : N48 G30 P0 *70
18:02:40.641 : N49 G1 X-20 Y-60 Z213.9 F4800 *7
18:02:40.659 : N50 G30 P0 *79
18:02:40.787 : N51 G1 X33.33 Y-60 Z213.9 F4800 *15
18:02:49.894 : Z-probe:39.29 X:-180.00 Y:-60.00 zCorr:0.00
18:02:52.752 : N52 G30 P0 *77
18:02:52.997 : N53 G1 X86.67 Y-60 Z213.9 F4800 *2
-- Killed power here to avoid more damage --
```

repetier, 2015-10-26:
G30 goes down and back to start position, why should it go up to the endstops?
I'm a bit confused here. Don't you start this a few mm above the bed? At the top you can not reach the testing area and if ALWAYS_CHECK_ENDSTOPS is 0 it will crash into them, which I guess happens here. If endstops stop it it is in invalid state and no one can say where it will move next. So never start bed height map at top here you can not reach the positions!

On the other side probe return 39 mm so that sound like you started lower. But Z213.9 is definitely high.

Links2004, 2015-10-26:
ALWAYS_CHECK_ENDSTOPS is define as 1, so the printer shut always respect them.

I had home the printer and then i start the mapping from the GUI.
will do some tests with manually move down to the first point before start.

Links2004, 2015-10-26:
ok if i lower the extruder before starting the mapping, it it working.
thanks for the help.

so one question is remaining why it has ignored the endstops?
ALWAYS_CHECK_ENDSTOPS is at 1, and the endstop was triggert to this time (the endstops has LEDs).