repetier/RepetierFirmware #435: Shared Extruder missing for Mixing Extruder

Wurstnase, 2015-11-12:
Hi,

I think you need somewhere something like:
```
#if MIXING_EXTRUDER
  #define EXT1_EXTRUDER_COOLER_PIN EXT0_EXTRUDER_COOLER_PIN
#endif
```

repetier, 2015-11-12:
No I don't think so. Mixing extruder always uses extruder 0 for everything. The 15 other extruders are just virtual. Why do you think that?

Wurstnase, 2015-11-12:
Ah, right, see it now.
Someone has the issue that the cooling fan doesn't run at 100%. After a quick check of that part I tought this could be the issue.
His fan runs at 14V of 24V.

His setup is something like:
```
#define MIXING_EXTRUDER 1
#define PDM_FOR_COOLER 0
#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN_PIN
#define EXT0_EXTRUDER_COOLER_SPEED 255
#define EXT1_EXTRUDER_COOLER_PIN -1
#define EXT1_EXTRUDER_COOLER_SPEED 255
```

What can do this effect then?

repetier, 2015-11-12:
Good question. If that pin is used somewhere else e.g. an other fan it could overwrite the signals. After all all pwm signals get checked and if 2 use same pin first can set and later one can disable earlier. Have seen this several times, but he said no other pin is using it. But maybe he only thinks that. Wanted to test this at the weekend with a new board from angelo as he is also using a radds board.

Wurstnase, 2015-11-12:
I've found this in pins.h
```
#define HEATER_2_PIN 9
#define ORIG_FAN_PIN   9
```

Wurstnase, 2015-11-12:
Yes, this is the issue.
```
#define EXT0_EXTRUDER_COOLER_PIN ORIG_FAN_PIN
#define EXT1_HEATER_PIN HEATER_2_PIN
```

repetier, 2015-11-12:
I guess you are looking in RAMPS where this often happens, because of the only 3 outputs for heater,bed and fan. For RADDS you have this

// Note that on the Due pin A0 on the board is channel 2 on the ARM chip
#define HEATER_0_PIN     13
#define TEMP_0_PIN       7 // Due analog pin #54
#define HEATER_1_PIN     7 
#define TEMP_1_PIN       3 // Due analog pin #58
#define HEATER_2_PIN     12
#define TEMP_2_PIN       6 // Due analog pin #55
#define HEATER_3_PIN     11
#define TEMP_3_PIN       5 // Due analog pin #56
#define TEMP_4_PIN       4 // Due analog pin #57

#define THERMOCOUPLE_0_PIN  2   // Dua analog pin #59 = A5 -> AD 2

#define ORIG_E0_STEP_PIN    61 // A7
#define ORIG_E0_DIR_PIN     60 // A6
#define ORIG_E0_ENABLE_PIN  62 // A8

#define ORIG_E1_STEP_PIN    64 // A10
#define ORIG_E1_DIR_PIN     63 // A9
#define ORIG_E1_ENABLE_PIN  65 // A11

#define ORIG_E2_STEP_PIN    51
#define ORIG_E2_DIR_PIN     53
#define ORIG_E2_ENABLE_PIN  49

#define ORIG_E3_STEP_PIN    35
#define ORIG_E3_DIR_PIN     33
#define ORIG_E3_ENABLE_PIN  37

#define ORIG_E4_STEP_PIN    29
#define ORIG_E4_DIR_PIN     27
#define ORIG_E4_ENABLE_PIN  31

#define SDSUPPORT      1
#define SDPOWER 	   -1
#define SDSS		    4// 4,10,52 if using HW SPI.
#define ORIG_SDCARDDETECT       14
#define SDCARDDETECTINVERTED 0
#define LED_PIN 	   -1
#define ORIG_FAN_PIN 	   9 
#define ORIG_FAN2_PIN           8 

and that should not collide.

Wurstnase, 2015-11-12:
I should spend a minute more with that before posting :)