repetier/RepetierFirmware #439: Dual reference to %oC

luc-github, 2015-11-20:
Hi I see dual reference to %oC is that normal ?
in umenu.h
https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/uimenu.h#L104
```
%oC : Output level current extruder
```
https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/uimenu.h#L110
```
%oC : Bed coating thickness
````
in ui.cpp

https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/ui.cpp#L1424
so this line is never reached 
https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/ui.cpp#L1434

repetier, 2015-11-20:
You are right. I will change bed coating to %BC for next update.

closed 2015-11-20 by repetier

luc-github, 2015-11-20:
ok thank you