repetier/RepetierFirmware #442: Save / Update setting in EEPROM behavior

luc-github, 2015-11-26:
I think there is some inconsistency in the way some settings are saved and on the way to update EEPROM.
Like when changing language
Expected behaviour is to save to EEPROM language change and nothing else
But currently all currents settings are saved to EEPROM because use of 
```EEPROM::storeDataIntoEEPROM(0);``` even the unwished ones
instead of use something like ```update(GCode *com);``` which is more specific

from another point of view, when updating one setting using ```update(GCode *com);```:
Expected behaviour is to change only one setting (the one we target)
But currently after setting is saved to EEPROM, the whole EEPROM is read and reset all others settings in memory because of ```readDataFromEEPROM(includesEeprom);```
I understand this is to verify the EEPROM integrity but it bring some confusion when doing some settings adjustments.

I would like to suggest when saving one setting to use something like
```
//internal usage of update value
void EEPROM:: update(long P,uint8_t T,long S,float X)
{
#if EEPROM_MODE!=0
    if(T>=0 &&T<=3 && P>0 && P<=2048) //SDEEPROM_SIZE =2048 // Minimum size used by Eeprom.cpp
    switch(T)
        {
        case EPR_TYPE_BYTE:
            HAL::eprSetByte(P,(uint8_t)S);
            break;
        case EPR_TYPE_INT:
           HAL::eprSetInt16(P,(int)S);
            break;
        case EPR_TYPE_LONG:
            HAL::eprSetInt32(P,(int32_t)S);
            break;
        case EPR_TYPE_FLOAT:
           HAL::eprSetFloat(P,X);
            break;
        }
    uint8_t newcheck = computeChecksum();
    if(newcheck != HAL::eprGetByte(EPR_INTEGRITY_BYTE))
        HAL::eprSetByte(EPR_INTEGRITY_BYTE,newcheck);
    //readDataFromEEPROM(true);
    //Extruder::selectExtruderById(Extruder::current->id);
#else
    Com::printErrorF(Com::tNoEEPROMSupport);
#endif
}
```
and replace ```readDataFromEEPROM(true);``` by something like ```checkDataFromEEPROM(true);``` which will really check and avoid to reset all settings

Is that a bad idea ?

repetier, 2015-11-26:
Not sure if it is a bad idea, but I guess so. The reason is not only to check integrity. Changing eeprom values should also have the desired effect. If you change acceleration you want the printer to use the acceleration. This is a side effect of readDataFromEEOPROM. If you have changed on of these variables with a non eeprom command you loose that. That is correct and a bit of a dilemma. At one time you want it to stay and one time you want the new eeprom to propagate. The only real solution is to make a special solution for every eeprom value. That will cost much work and memory, which is the bigger problem. With all features enabled I have only 20kb free flash for a mega based board. So I guess this behaviour will stay. After all eeprom settings are not meant to be changed while printing, so these side effects are normally neglible.

luc-github, 2015-11-26:
OK I understand your constraint - Thank you for your feedback
My target is DUE based and has more memory so I will see the impact on it about memory and user experience once implemented,

Thanks again

closed 2015-11-26 by luc-github