repetier/RepetierFirmware #446: Which vanilla Arduino boards is Repetier compatible with?

polkovnikov-ph, 2016-01-02:
It's said that Repetier is compatible with Melzi that uses atmega1284p. What's the size of the binary? Which common Arduino boards can handle RAM/ROM usage of this firmware and have proper clock frequency?

jamesarm97, 2016-01-02:
Repetier is compatible with Melzi as all the wanhao i3's run it and I have compiled a few versions for it. Needs to disable all the non-usable parts like z probe and auto leveling. There are a few changes you will have to make if you are not using the versions I have posted since the melzi configuration seems broke in the newer version (at least the wanhao configuration). The common arduino board used is the Mega2560 but the Due is a better 32 but processor if you can go that route with the appropriate ramps type board (I am running the RADDS on my wanhao).

repetier, 2016-01-02:
Nothing against the melzi board, but binary with all features is over 200KB so it won't fit. Practically all Arduiono AVR boards run 16MHz so speed is the same. Only due boards like RADDS with 32 bit are much faster. You can reduce memory usage be removing features to fit it on smaller boards, also I would not take anything below 128kb and vote for 256MB.

closed 2016-01-02 by repetier