repetier/RepetierFirmware #447: Cicles and short segments (curves) execution delay

thiagopeixoto16, 2016-01-05:
Hello! First of all thanks for this amazing firmware.

## The problem:
I found an issue related to the Transformation Matrix, that's a delay execution on circles or curves.
I know how to reproduce and workaround the problem, below you have pictures and videos about the problem and how I solved. You may get that workaround and have a clue to find the root cause.

This is the problem when running circles:
https://youtu.be/i_Tpj2xyX-k
Can you see these gagging movements?

Occurring both via LCD/SD or Repetier Host print.

## The Workaround:
I discover that if I just resend any EEPROM value related to the Transformation Matrix the problem immediately disappears, like the orthogonal compensation: `M206 T3 P976 X0.009`
So the log will looks like:
[image]

## The result:
After the workaround the execution is fine:
https://youtu.be/Ze75Sm53x4w

This is the before/after the g-code workaround:
[image]

## My setup:
Ramps 1.4 / Arduino Mega 2560 / A4988
Repetier 0.92 get from webtool
Config file and EEPROM: 
https://www.dropbox.com/sh/f6mlvtwl4xpixx0/AAByUYRveU0ym9mSUCeK4h8oa?dl=0

Please let me know if some additional information is required.

Thanks!

jamesarm97, 2016-01-05:
Your YouTube links do not work (on my end)

-James Armstrong

https://youtube/i_Tpj2xyX-k

https://youtube/Ze75Sm53x4w

https://wwwdropboxcom/sh/f6mlvtwl4xpixx0/AAByUYRveU0ym9mSUCeK4h8oa?dl=0

thiagopeixoto16, 2016-01-05:
Hello @jamesarm97, I just tested from an anonymous Chrome window and it worked... anyone else having problems?

repetier, 2016-01-05:
First if you are saying circles you are not talking about G2/G3, right?

While I see the difference in last image I see no difference on the youtube videos.

But my biggest problem is that I do not understand what you did to fix the problem. You show the rotation matrix but
M206 T3 P976 X0.009
changes the skew not rotation. So in other words you mean
#define FEATURE_AXISCOMP 0
would solve the problem. And with enabled compensation it adds some error.

Second did you write the same value that was previously there to make it work again?

repetier, 2016-01-05:
The links in the mail send from github do not work, if you view it on github everything is visible.

jamesarm97, 2016-01-05:
I see the differences. In the bad video the extruder shakes pretty badly when it goes from the left to the right (bad printer design in the first place) and in the "fixed" video that shaking is gone. Is the transform being disabled in the second video somehow? If I understand the transform feature it moves the Z while printing so maybe the extruder moves faster than the Z can move and has to slow down during those moves?

repetier, 2016-01-05:
Just one thought if it is related with axis compensation. It constantly reads the values from eeprom which is slow and maybe if you read to often we somehow change the value (which would explain why setting it again helps). What if you change

#if FEATURE_AUTOLEVEL
void Printer::transformToPrinter(float x,float y,float z,float &transX,float &transY,float &transZ)
{
	#if FEATURE_AXISCOMP
	// Axis compensation:
	x = x + y * EEPROM::axisCompTanXY() + z * EEPROM::axisCompTanXZ();
	y = y + z * EEPROM::axisCompTanYZ();
	#endif
	transX = x * autolevelTransformation[0] + y * autolevelTransformation[3] + z * autolevelTransformation[6];
	transY = x * autolevelTransformation[1] + y * autolevelTransformation[4] + z * autolevelTransformation[7];
	transZ = x * autolevelTransformation[2] + y * autolevelTransformation[5] + z * autolevelTransformation[8];
}

void Printer::transformFromPrinter(float x,float y,float z,float &transX,float &transY,float &transZ)
{
	transX = x * autolevelTransformation[0] + y * autolevelTransformation[1] + z * autolevelTransformation[2];
	transY = x * autolevelTransformation[3] + y * autolevelTransformation[4] + z * autolevelTransformation[5];
	transZ = x * autolevelTransformation[6] + y * autolevelTransformation[7] + z * autolevelTransformation[8];
	#if FEATURE_AXISCOMP
	// Axis compensation:
	transY = transY - transZ * EEPROM::axisCompTanYZ();
	transX = transX - transY * EEPROM::axisCompTanXY() - transZ * EEPROM::axisCompTanXZ();
	#endif
}

by replacing all  axisCompTan.. calls with your value. That would speed up computation and not read it from eeprom every time. If that helps I could modify the code to store the values in ram instead.

repetier, 2016-01-05:
Now that you mention the jitter of printer I see it. Watched more the extruder moves.

As you see in my previous post the rotation matrix and axis compensation are independent. The M206 you showed belongs to axis compensation. That is why I asked myself why a unrelated write would make a difference. I still ask myself how it does.

If you would change rotation to make it more orthogonal the z motor would need to move less so if z correction is a limiting factor depending on angle in circle I would say the jitter comes from that.

thiagopeixoto16, 2016-01-05:
Yes @repetier you got the point. I'll try what you suggested here and let you know soon.

Anyway I'll answer your previous questions:
- I'm not using ARCS, they don't allow Z compensation (BAL) right!?
- Yes I have "FEATURE_AXISCOMP" activated, you can see by the configuration.h on dropbox file.
 - Yes, I wrote the same value as before, just to shake the things there on EEPROM.

thiagopeixoto16, 2016-01-05:
Just did and nothing changed... 
The error persistis and the workaround is still helping to fix.
Here is the changed code:
```
#if FEATURE_AXISCOMP
    // Axis compensation:
    x = x + y * 0.009 + z * -0.025;
    y = y + z * -0.020;
#endif
    transX = x * autolevelTransformation[0] + y * autolevelTransformation[3] + z * autolevelTransformation[6];
    transY = x * autolevelTransformation[1] + y * autolevelTransformation[4] + z * autolevelTransformation[7];
    transZ = x * autolevelTransformation[2] + y * autolevelTransformation[5] + z * autolevelTransformation[8];
}

void Printer::transformFromPrinter(float x,float y,float z,float &transX,float &transY,float &transZ)
{
    transX = x * autolevelTransformation[0] + y * autolevelTransformation[1] + z * autolevelTransformation[2];
    transY = x * autolevelTransformation[3] + y * autolevelTransformation[4] + z * autolevelTransformation[5];
    transZ = x * autolevelTransformation[6] + y * autolevelTransformation[7] + z * autolevelTransformation[8];
#if FEATURE_AXISCOMP
    // Axis compensation:
    transY = transY - transZ * -0.020;
    transX = transX - transY * 0.009 - transZ * -0.025;
#endif
}
```

repetier, 2016-01-05:
1. Arcs also use Z compensation. Arcs are just many tiny line segments, so what works for lines also works for arcs.

2. Your fix is correct and now does not use any eeprom value.

3. You mean sending only M206 T3 P976 X0.009 which is now unused makes a difference? In that case it must be a side effect. After changing eeprom it reads all eeprom values to ram (if used) and updates derived parameter. Then it selects the active extruder to update extruder data as well. This is the same as on first load so there should be no difference, except somehow a function got overwritten with wrong values.

So one question is, is the bad behavior from the start away or does it start at some point?

If it is from start or also if not, can you add a new gcode to test which part makes the difference. I think in mcodes add a new case
case 2000:
  if(com->S == 1) EEPROM::readDataFromEEPROM(true);
if(com->S == 2) Printer::updateDerivedParameter();
if(com->S == 3) Extruder::selectExtruderById(Extruder::current->id);
break;

case 1 includes 2 as well. So only 2 and 3 are fine grained.

An other thing - if you reduce speed M220 S50 to 50% does it get better?

thiagopeixoto16, 2016-01-05:
Thanks for the information on ARCS, I'll consider implement it in near future.

Yes, that command is still fixing the issue. **By the way** I tried with a blank `M206` and that solves the problem too... :astonished:

The issue starts since the first curve movement, on very first layer.

The problem persists even with 50%, it is more difficult to see but it is still there.

I didn't get your M2000 suggestion, can you give some light on this?

repetier, 2016-01-05:
For M2000 go to commands.cpp and search case 400: // M400 Finish all moves
add the code just before that or after the next break. Then you have a new M command you call like
M2000 S1
M2000 S2
M2000 S3

Idea is to simulate the parts M206 does but not completely so we can narrow down where the "defect" parameter is, that causes jittering.

thiagopeixoto16, 2016-01-05:
I changed the code, start the same cylinder object and the problem was there.
Then I run the commands from S3 to S1, so let's go to the results:
M2000 S3 - Does nothing
M2000 S2 - Does nothing
**M2000 S1 -  it does solve the problem !**

Any light?

repetier, 2016-01-06:
Very strange. Reading data from eeprom is the last function called in initalization.

In fact M501 should also solve it as it also calles read eeprom again. Maybe You can use this in combination with the host to find out which parameter causes the problem after all. I'm still not sure how I can make everything going correct just with jittering. So here how we can find the changed values.
In host open eeprom settings and save them to disk.
Send M500, this copies RAM to eeprom. Since the other way around fixes it we can assume that one value is different in ram then eeprom. But now that we have jittering we call M500 to copy the defect values to eeprom. Check them with host again and save to disk. 
Compare the files for differences and then we know which parameter gets changed in your case. You may then want to load old good eeprom data in host and store it back.

thiagopeixoto16, 2016-01-13:
Hello, sorry the delay. Here it go the news:

- I did the upgrade from 0.92.7 to 0.92.8 (today from web) but the problem persists;
- M501 do not solved the problem.
- I just discovered that the M206 is causing another problem, it is erasing the original measured correction matrix (mainly the Bending Correction points), so it is not an workaround anymore.
[image]
- I did the EEPROM comparison you requested, before and after the load from RAM (M500) and the sadly they match:
[image]
The original eeprom files at https://goo.gl/fbHUu0
- Today I tried to change my gcode from G32 S0 to G32 S2 but no luck with that too.
- I'm not sure if it helps but I found a pattern on the error, the glitches always occurs over these points inside the four circles, I'll try to arrange the circles in different positions to check if it changes:
[image]

thiagopeixoto16, 2016-01-13:
It seems to be concentric to the object:
[image]
I tried to put the "M206" before the "G32" but it doesn't fix the issue in this order. Whatever is causing this is after the Bed Leveling.

thiagopeixoto16, 2016-01-13:
Another test I just did:
Re-implemented the M2000 S(1/2/3) and tested again. When using G32 S2 none of the three parameters fix the issue.  So S1 may worked before because I never had written the correction matrix to eeprom before and it got some dummy (almost null) values from there.

thiagopeixoto16, 2016-01-14:
After a batch of tests I was able to determine what configuration is causing this: 
- Values lower than -3 on Z-probe bending correction (tested with A and B points) is causing this issue.
  
I tried to put -4 and the problems appeared, back to -3 it's gone.
Due the design of my printer (Smartrap design + z-min endstop as probe) we require a lot of bending correction.

@repetier, Do you think it is something that can be fixed?

repetier, 2016-01-14:
Your M206 did not change rotation matrix. G32 measures the points and updates the matrix.

Bending correction does nothing during print, all it does is adding a value to z probed heights do compensate printer bending. That in return will cause a different rotation matrix. And a different rotation matrix also explains why M206 does make a difference. It copes the old stored rotation matrix and makes it active undoing your G32. If you do G32 S2 a M206 would not make any difference.

This more or less also explains why you have a problem. Depending on your rotation you need to compensate more or less fast with z. I see values of 2.2% in your log (from rotation matrix). So if you move 100mm/s z hoes 2.2mm/s. If you reach z limits in acceleration or speed your troubles begin I think. So all your "fixes" do is changing the rotation matrix with one less rotated version, so there is no speed reduction due to z moves. So check if you can move z faster then set in firmware or level bed manually a bit better, so less rotation is required and thus less speed.

thiagopeixoto16, 2016-01-14:
No luck, I have increased the Z movement parameters like that:
- Max. Z-jerk from 0.3 to 0.6
- Max feedrate from 4 to 8 (tested and worked fine at max speed)
- Z-acceleration (travel and print) from 100 to 500 (tested too)

But the glitch is still there.

Remembering that it happens only on round objects. I just tested the below case and the "micro-pause" didn't occur:
[image]

It seems to be related to computation since short segments may require more speed on computation than huge segments (4 per turn) like the example above requires less job...

Yesterday I tried to check the free RAM but the debug seems to output that info only at the beginning  of the job and only once, the value I got was 2248, I think it is good, right?

I tried a big single ARC G2 command to make a 360 loop but it causes the same effect as the shorts G1 segments.

I'll keep trying things here (since I'm not good enough in math) but if you have any direction or need any other information from me please let me know.

repetier, 2016-01-14:
If you have a display, what does buffer say? 

The strange thing with working or not working is that the computation it self take the same time.

Short segments are always a problem since we descrease acceleration length and if that happens we might vary speed just for the sake of not being able to reach target speed in our small stored and buffered part. Otherwise with less speed it should get better and it does not if I remember right. And why should bending have an effect on this. My explanation might be wrong for the case but explained et least what happens.

If you have 16 buffer and it is full you might try 24 or 32 buffer and see if it differs. It might also be a bad coincidence with resonance frequency that you hit with some settings. It is getting quite mysterious.

thiagopeixoto16, 2016-01-14:
It says 32 all the time, rarely it shows 31.
On the very first tries I thought that the 32 buffer was the problem so I reduced back to 16, but no luck.

I have very low accelerations values like 350mm/s for XY and I just tried to change to something higher like 2.000 but the issue persists. Anyway I think the path planning will overwrite that...

rcarlyle, 2016-01-14:
Looking at the pictures of circles with red lines... If the pauses/jitters are always oriented 45 degrees or 90 degrees to the XY axes, that seems like an important detail. That might suggest an edge case in the transform code when X speed equals Y speed, for example.

thiagopeixoto16, 2016-01-14:
Good point @rcarlyle, however  I just played with the correction matrix trying to find the maximum inclination it can handle without errors and found that near the edge values the jitter changes it positions. Now we have it on a totally new places, see the vídeo below:
https://youtu.be/tzSPR2eIx1E

thiagopeixoto16, 2016-01-14:
I played with bending options to generate different matrix and check what is causing the issue and what is not, after two hours of real fun I got the below:

**Group of the ok/fine matrix -->**
```
14:57:35.221 : Transformation matrix: 1.000000 0.000009 0.000525 0.000000 0.999838 -0.017997 -0.000525 0.017997 0.999838
14:59:44.144 : Transformation matrix: 0.999999 0.000019 0.001067 0.000000 0.999837 -0.018078 -0.001067 0.018078 0.999836
15:03:31.939 : Transformation matrix: 1.000000 0.000015 0.000792 0.000000 0.999821 -0.018910 -0.000792 0.018910 0.999821
16:04:16.761 : Transformation matrix: 1.000000 0.000009 0.000460 0.000000 0.999816 -0.019169 -0.000460 0.019169 0.999816
16:12:35.750 : Transformation matrix: 1.000000 0.000013 0.000700 0.000000 0.999818 -0.019069 -0.000700 0.019069 0.999818
16:41:01.813 : Transformation matrix: 1.000000 0.000010 0.000540 0.000000 0.999825 -0.018730 -0.000540 0.018730 0.999824
```

**group of the bad guys --->**
```
16:46:30.852 : Transformation matrix: 1.000000 0.000009 0.000452 0.000000 0.999815 -0.019221 -0.000452 0.019221 0.999815  > only externals are bad
16:39:11.597 : Transformation matrix: 1.000000 0.000013 0.000690 0.000000 0.999818 -0.019058 -0.000690 0.019058 0.999818
16:33:53.900 : Transformation matrix: 1.000000 0.000010 0.000535 0.000000 0.999813 -0.019352 -0.000535 0.019352 0.999813  > only externals are bad
16:32:12.414 : Transformation matrix: 1.000000 0.000018 0.000865 0.000000 0.999794 -0.020274 -0.000865 0.020274 0.999794
16:29:55.752 : Transformation matrix: 1.000000 0.000013 0.000660 0.000000 0.999820 -0.018997 -0.000660 0.018997 0.999819  > only externals are bad
16:25:02.511 : Transformation matrix: 1.000000 0.000009 0.000465 0.000000 0.999808 -0.019618 -0.000465 0.019618 0.999807
16:23:30.857 : Transformation matrix: 1.000000 0.000014 0.000702 0.000000 0.999807 -0.019621 -0.000702 0.019621 0.999807
16:19:38.171 : Transformation matrix: 1.000000 0.000009 0.000442 0.000000 0.999810 -0.019482 -0.000442 0.019482 0.999810
16:19:38.171 : Transformation matrix: 1.000000 0.000009 0.000442 0.000000 0.999810 -0.019482 -0.000442 0.019482 0.999810  > 3 external perimeters (188)
16:17:35.719 : Transformation matrix: 1.000000 0.000009 0.000442 0.000000 0.999809 -0.019549 -0.000442 0.019549 0.999809
16:02:42.712 : Transformation matrix: 1.000000 0.000008 0.000437 0.000000 0.999816 -0.019177 -0.000437 0.019177 0.999816
15:05:51.907 : Transformation matrix: 1.000000 0.000014 0.000745 0.000000 0.999814 -0.019302 -0.000745 0.019302 0.999813
```

kyrreaa, 2016-01-14:
I've noticed that my highspeed motion demo will run very jerkily on the delta after it's been on for a while, but reset it, and run demo and it is very smooth.
Could this be related? It's like it micropauses multiple times during long straight moves and some points during circles.

repetier, 2016-01-15:
@kyrreaa As long as we do not know the source it could be the same, also delta has so much more computations that it might be a different thing. Since we have no memory management it is not really to be expected to get changes over time unless there is something changing over time that has influence on moves.

@thiagopeixoto16 The rotation matrix is position invariant. All moves get the same transformation. In this context what do you mean with "only externals are bad"? I assume the circles have all segments with same length, so only difference is ratio x/y steps. Speaking of steps, what is your resolution? Could it be that coarseness and constant changes make it swing. As I see it is a printrbot with a long levered arm and such construct can swing quite easily, but I have no comparable printer so I can not really say how prone it really is.

thiagopeixoto16, 2016-01-15:
Hello @repetier !
No it isn't a matter of swing, the linear movement really having little pauses. The swing is just the consequence of that. I can make a video showing the stepper motor if required.

I have 80 steps per mm, the very standard GT2 pulley 20 teeth. 

Sorry about the no extra explanation about it, these "only external are bad", it means that only the external loop of the test cylinder had the jitters, after the 3rd or 4th round to the center the problem disappeared. It has a concentric infill.

thiagopeixoto16, 2016-01-15:
In this video you can see the problem in all its splendor:
https://youtu.be/DrNLL5Cq8WA

Below a full first layer:
Can you see the lines? This is where the stops occurred.
[image]

repetier, 2016-01-15:
That last video and image are quite interesting. Can you send the gcode for the example. I'd like to test it on my printer also it has no rotation at all. I guess it is a path planning thing combined with communication speed. Also I see always plenty of buffers used it seems that not always enough is buffered to reach target speed, so max. speed varies. Or you block segments while path planning that are needed for current segment. But then you would see a BLK message in log and you did not mention this. So I assume we do not run out of data etc, just vary speed as dictated by data. Having circles with double segment length would help then, but that is part of why I'd like to see the gcode.

thiagopeixoto16, 2016-01-15:
I've just copied the g-code on that dropbox folder:
https://www.dropbox.com/sh/f6mlvtwl4xpixx0/AAByUYRveU0ym9mSUCeK4h8oa?dl=0

thiagopeixoto16, 2016-01-15:
I don't think communication is a problem, as mentioned I have always 32 on LCD and tested that g-code from PC Host and LCD with the same issues. Also tried to reduce speed and no luck.
Yes, I don't have such BLK message in log.
The problem is only suppressed when I reduce the plane inclination by playing with bending options, but in that way the purpose of software correction is useless.
Thanks!

repetier, 2016-01-15:
Just had a first view on the gcode. Nothing special and only around 120 segments per circumsphere. I must try to compile it in a printer simulation with your settings and somehow fake rotation and see what I can see. My feeling says there is enough time to calculate everything needed.

kyrreaa, 2016-01-15:
I'll do some dumps later from system exhibiting symptoms and again after
reset. I have a board with jtag.

thiagopeixoto16, 2016-01-16:
@repetier, do you believe it can be some rounding problem? I mean when the vector is too short it may cause too much calculations by the precision of the involved numbers?

thiagopeixoto16, 2016-01-18:
Any luck on testing?

repetier, 2016-01-18:
Ok, I have now a debug environment for the problem. I have a RAMPS with your configuration (hoping eeprom and configuration.h matches) connected to a logic analyser so I can see speed changes. So far the only acceleration I could associate were direction changes between circles. Also path planner plans all speeds with 35mm/s except at circle switches, so this is as expected.

For testing I used this

M206 T3 P844 X1
M206 T3 P848 X0.000013
M206 T3 P852 X0.000690
M206 T3 P856 X0.000000
M206 T3 P860 X0.999818
M206 T3 P864 X-0.019058
M206 T3 P868 X-0.000690
M206 T3 P872 X0.019058
M206 T3 P876 X0.999818

to simulate a problematic autoleveling matrix. And then I started printing in dry run (no printer connected), but it looks like nothing happens.

Could you make a simple test script that shows the problem using my above autolevel matrix (so you need it after G32 or no G32 in your script at all) in dry run mode. Best with video so I can see where the error should be. With simple I mean max. 1 circle. Then I would try to replay that script in my simulated printer and hope to see something.

Thanks.

kyrreaa, 2016-01-18:
Could you try resetting the printer when you experience this, as I find that resetting printer and rehoming it (the levelling should be stored in eeprom etc) will make mine run smooth.

Kyrre

From: Thiago 
Sent: Monday, January 18, 2016 6:49 PM
To: repetier/Repetier-Firmware 
Subject: Re: [Repetier-Firmware] Circles and short segments (curves) execution delay (#447)

Any luck on testing?

thiagopeixoto16, 2016-01-18:
Hello @kyrreaa,

I just tried what you suggested but the problem persisted after the reset. 

@repetier I'm working on this right now.

thiagopeixoto16, 2016-01-18:
@repetier, I tried with your matrix but that is not generating the problem, then I tried another case I have and that produced the issue on video below:
https://youtu.be/1Pe11pC9nPw

The above issue was caused by this matrix: 
```
M206 T3 P844 X1
M206 T3 P848 X0.000026
M206 T3 P852 X0.000674
M206 T3 P856 X0.000000
M206 T3 P860 X0.999263
M206 T3 P864 X-0.038372
M206 T3 P868 X-0.000675
M206 T3 P872 X0.038372
M206 T3 P876 X0.999263
17:51:30.307 : Transformation matrix: 1.000000 0.000030 0.000670 0.000000 0.999260 -0.038370 -0.000680 0.038370 0.999260
```
Have you noticed that it is somehow rounding the number? 000026 turns to 000030

Anyway, here find below the requested simple gcode (used on the video), one loop only with no G32:
https://www.dropbox.com/s/z9m9i4kx25mwnb4/test-gcode.gcode?dl=0

kyrreaa, 2016-01-18:
Verifying that transformation matrix is in unity may detect it?
18. jan. 2016 10.13 p.m.

repetier, 2016-01-19:
I have added a debug option to check planner entries added/modified. With you latest example I get 4 slowdowns to 8.9mm/s which were not present when autolevel was disabled. So that is at least what you see. Logic analyser also shows the slowdowns, so they are put through the complete process

16:28:13.900 : Planner: 4 F 35.0 - 35.0 / 0.2 - 0.3(0.3,7 / 4.5 - 4.5
16:28:13.909 : Planner: 4 F 0.2 - 0.3 / 4.5 - 4.5(2.1,5 / 5.4 - 5.4
16:28:13.916 : Planner: 3 F 4.5 - 4.5 / 5.4 - 31.1(31.1,3 / 31.1 - 5.7
16:28:13.929 : Planner: 4 F 5.4 - 31.1 / 31.1 - 31.1(31.1,7 / 31.1 - 6.1
16:28:13.956 : Planner: 5 F 31.1 - 31.1 / 31.1 - 32.6(32.6,7 / 32.6 - 6.6
16:28:13.964 : Planner: 6 F 31.1 - 32.6 / 32.6 - 32.6(32.6,7 / 32.6 - 7.0
16:28:13.974 : Planner: 7 F 32.6 - 32.6 / 32.6 - 32.8(32.8,7 / 32.8 - 7.5
16:28:13.982 : Planner: 8 F 32.6 - 32.8 / 32.8 - 32.8(32.8,7 / 32.8 - 8.3
16:28:13.995 : Planner: 10 F 32.8 - 8.3 / 9.3 - 33.2(33.2,7 / 33.2 - 10.0
16:28:14.005 : Planner: 11 F 9.3 - 33.2 / 33.2 - 32.8(35.0,5 / 32.8 - 10.0
16:28:14.014 : Planner: 12 F 9.3 - 33.2 / 33.2 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.023 : Planner: 13 F 35.0 - 35.0 / 35.0 - 32.3(35.0,5 / 32.3 - 10.0
16:28:14.029 : Planner: 14 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.041 : Planner: 15 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 10.0
16:28:14.052 : Planner: 16 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.058 : Planner: 17 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.065 : Planner: 18 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.077 : Planner: 19 F 35.0 - 35.0 / 35.0 - 32.1(35.0,5 / 32.1 - 10.0
16:28:14.104 : Planner: 20 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.111 : Planner: 21 F 35.0 - 35.0 / 35.0 - 32.3(35.0,5 / 32.3 - 10.0
16:28:14.120 : Planner: 22 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.127 : Planner: 23 F 35.0 - 35.0 / 35.0 - 32.6(35.0,5 / 32.6 - 10.0
16:28:14.141 : Planner: 24 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.151 : Planner: 25 F 35.0 - 35.0 / 35.0 - 33.7(35.0,5 / 33.7 - 10.0
16:28:14.160 : Planner: 26 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 8.9
16:28:14.169 : Planner: 28 F 35.0 - 8.9 / 8.0 - 33.9(35.0,7 / 33.9 - 7.4
16:28:14.177 : Planner: 29 F 8.0 - 33.9 / 33.9 - 34.8(35.0,5 / 34.8 - 7.0
16:28:14.198 : Planner: 30 F 8.0 - 33.9 / 33.9 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 6.3
16:28:14.201 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 6.0
16:28:14.364 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.6
16:28:14.373 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.4
16:28:14.521 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.1
16:28:14.536 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.3(35.0,5 / 34.3 - 4.8
16:28:14.806 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.7
16:28:14.820 : Planner: 31 F 35.0 - 35.0 / 35.0 - 33.2(35.0,5 / 33.2 - 4.6
16:28:15.092 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.4
16:28:15.106 : Planner: 31 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 4.3
16:28:15.383 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.2
16:28:15.408 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.6(35.0,5 / 31.6 - 4.1
16:28:15.678 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.1
16:28:15.694 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.4(35.0,5 / 31.4 - 4.1
16:28:15.965 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.0
16:28:15.978 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.9(35.0,5 / 30.9 - 3.9
16:28:16.249 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.261 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.7(35.0,5 / 30.7 - 3.9
16:28:16.530 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.551 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.9(35.0,5 / 30.9 - 3.9
16:28:16.818 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.832 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.0(35.0,5 / 31.0 - 4.0
16:28:17.100 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.0
16:28:17.114 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 4.1
16:28:17.391 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 4.2
16:28:17.395 : Planner: 31 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 4.2
16:28:17.671 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.3
16:28:17.683 : Planner: 31 F 35.0 - 35.0 / 35.0 - 33.5(35.0,5 / 33.5 - 4.5
16:28:17.959 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.6
16:28:17.986 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.3(35.0,5 / 34.3 - 4.8
16:28:18.254 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 5.0
16:28:18.267 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.2
16:28:18.538 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.4
16:28:18.549 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.7
16:28:18.824 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 6.1
16:28:18.843 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.8(35.0,5 / 34.8 - 6.6
[... cut ...]
16:28:26.546 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.3(35.0,5 / 34.3 - 4.7
16:28:26.820 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 5.0
16:28:26.830 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.1
16:28:27.100 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.4

So now I'm searching why the same sequence can result in different speed profiles while buffer stays full. Looks a bit like the speed was set fixed for some reason so planner had no choice. Will keep you informed about the progress.

thiagopeixoto16, 2016-01-19:
That's great! We at least found that bad guy! Congrats! 
I can't wait to see it defeated :stuck_out_tongue_winking_eye:

kyrreaa, 2016-01-20:
The planner will be forced to slow down if it’s not keeping up with execution right?
I assume the number you print right after “Planner:” is the number of moves in queue?
If so that’s not it... Hmm...

Kyrre

From: repetier 
Sent: Tuesday, January 19, 2016 4:38 PM
To: repetier/Repetier-Firmware 
Cc: kyrreaa 
Subject: Re: [Repetier-Firmware] Circles and short segments (curves) execution delay (#447)

I have added a debug option to check planner entries added/modified. With you latest example I get 4 slowdowns to 8.9mm/s which were not present when autolevel was disabled. So that is at least what you see. Logic analyser also shows the slowdowns, so they are put through the complete process

16:28:13.900 : Planner: 4 F 35.0 - 35.0 / 0.2 - 0.3(0.3,7 / 4.5 - 4.5
16:28:13.909 : Planner: 4 F 0.2 - 0.3 / 4.5 - 4.5(2.1,5 / 5.4 - 5.4
16:28:13.916 : Planner: 3 F 4.5 - 4.5 / 5.4 - 31.1(31.1,3 / 31.1 - 5.7
16:28:13.929 : Planner: 4 F 5.4 - 31.1 / 31.1 - 31.1(31.1,7 / 31.1 - 6.1
16:28:13.956 : Planner: 5 F 31.1 - 31.1 / 31.1 - 32.6(32.6,7 / 32.6 - 6.6
16:28:13.964 : Planner: 6 F 31.1 - 32.6 / 32.6 - 32.6(32.6,7 / 32.6 - 7.0
16:28:13.974 : Planner: 7 F 32.6 - 32.6 / 32.6 - 32.8(32.8,7 / 32.8 - 7.5
16:28:13.982 : Planner: 8 F 32.6 - 32.8 / 32.8 - 32.8(32.8,7 / 32.8 - 8.3
16:28:13.995 : Planner: 10 F 32.8 - 8.3 / 9.3 - 33.2(33.2,7 / 33.2 - 10.0
16:28:14.005 : Planner: 11 F 9.3 - 33.2 / 33.2 - 32.8(35.0,5 / 32.8 - 10.0
16:28:14.014 : Planner: 12 F 9.3 - 33.2 / 33.2 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.023 : Planner: 13 F 35.0 - 35.0 / 35.0 - 32.3(35.0,5 / 32.3 - 10.0
16:28:14.029 : Planner: 14 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.041 : Planner: 15 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 10.0
16:28:14.052 : Planner: 16 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.058 : Planner: 17 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.065 : Planner: 18 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.077 : Planner: 19 F 35.0 - 35.0 / 35.0 - 32.1(35.0,5 / 32.1 - 10.0
16:28:14.104 : Planner: 20 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.111 : Planner: 21 F 35.0 - 35.0 / 35.0 - 32.3(35.0,5 / 32.3 - 10.0
16:28:14.120 : Planner: 22 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.127 : Planner: 23 F 35.0 - 35.0 / 35.0 - 32.6(35.0,5 / 32.6 - 10.0
16:28:14.141 : Planner: 24 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 10.0
16:28:14.151 : Planner: 25 F 35.0 - 35.0 / 35.0 - 33.7(35.0,5 / 33.7 - 10.0
16:28:14.160 : Planner: 26 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 8.9
16:28:14.169 : Planner: 28 F 35.0 - 8.9 / 8.0 - 33.9(35.0,7 / 33.9 - 7.4
16:28:14.177 : Planner: 29 F 8.0 - 33.9 / 33.9 - 34.8(35.0,5 / 34.8 - 7.0
16:28:14.198 : Planner: 30 F 8.0 - 33.9 / 33.9 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 6.3
16:28:14.201 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 6.0
16:28:14.364 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.6
16:28:14.373 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.4
16:28:14.521 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.1
16:28:14.536 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.3(35.0,5 / 34.3 - 4.8
16:28:14.806 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.7
16:28:14.820 : Planner: 31 F 35.0 - 35.0 / 35.0 - 33.2(35.0,5 / 33.2 - 4.6
16:28:15.092 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.4
16:28:15.106 : Planner: 31 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 4.3
16:28:15.383 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.2
16:28:15.408 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.6(35.0,5 / 31.6 - 4.1
16:28:15.678 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.1
16:28:15.694 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.4(35.0,5 / 31.4 - 4.1
16:28:15.965 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.0
16:28:15.978 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.9(35.0,5 / 30.9 - 3.9
16:28:16.249 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.261 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.7(35.0,5 / 30.7 - 3.9
16:28:16.530 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.551 : Planner: 31 F 35.0 - 35.0 / 35.0 - 30.9(35.0,5 / 30.9 - 3.9
16:28:16.818 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 3.9
16:28:16.832 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.0(35.0,5 / 31.0 - 4.0
16:28:17.100 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.0
16:28:17.114 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 4.1
16:28:17.391 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 4.2
16:28:17.395 : Planner: 31 F 35.0 - 35.0 / 35.0 - 32.2(35.0,5 / 32.2 - 4.2
16:28:17.671 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.3
16:28:17.683 : Planner: 31 F 35.0 - 35.0 / 35.0 - 33.5(35.0,5 / 33.5 - 4.5
16:28:17.959 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.6
[... 39 lines cut ...]
16:28:23.950 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 4.2
16:28:23.969 : Planner: 31 F 35.0 - 35.0 / 35.0 - 31.9(35.0,5 / 31.9 - 4.2
[... cut ...]
16:28:26.546 : Planner: 31 F 35.0 - 35.0 / 35.0 - 34.3(35.0,5 / 34.3 - 4.7
16:28:26.820 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 35.0(35.0,7 / 35.0 - 5.0
16:28:26.830 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.1
16:28:27.100 : Planner: 31 F 35.0 - 35.0 / 35.0 - 35.0(35.0,7 / 35.0 - 5.4

So now I'm searching why the same sequence can result in different speed profiles while buffer stays full. Looks a bit like the speed was set fixed for some reason so planner had no choice. Will keep you informed about the progress.

repetier, 2016-01-21:
Ok, I think I have found it. The problem rises when z axis becomes dominant axis in bresenham. If the path planner sees the switch it assumed a switch xy move->z move and forced start speeds for that point. In this case it is just a different angle with z from rotation matrix so that pause is wrong. Latest update should now work without that pause.

Thanks for the help and finding the problem. My cartesian printer does mechanical leveling so the problem never arrised.

thiagopeixoto16, 2016-01-30:
I've tested it exhaustively in two different machines and it is solved, I'm closing this.
Thank you very much!

closed 2016-01-30 by thiagopeixoto16

thiagopeixoto16, 2016-02-01:
Just a question: Is this already published on Web version? 
I mean: www.repetier.com/firmware/v092/
Will this have a new release number like 0.92.9? or 8b?

repetier, 2016-02-01:
Yes it is included online.