repetier/RepetierFirmware #450: Temperature setpoints not always showing properly in host

frank16, 2016-01-06:
When I use the M104 command the temperature given is correctly used by the printer (live values show correctly in Repetier host), but I get 235degrees indicated for the setpoint, also in the graph. 
The 'output extruder' and 'output bed' graphs content stop being updated, but the main graph still is. I have seen this behaviour in every version I used so far. 
91.6.0 is what I am using now with Marlin firmware v2.3-k8200 version.
Is it possible M104 is not interpreted by repetier host?

Part of my start G-code:
M104 S180; preheat extruder (non blocking)
M190 S85 ; first layer: set and wait for bed temp to be reached (blocking)
M109 S235; first layer: set and wait for extruder temp to be reached (or higher)
G1 E5 ; prime the nozzle

repetier, 2016-01-06:
According to your script that is correct. You set first 180 and then later 235 °C. Since the host has send the M109 it shows that target temperature already, that is all. At some point when M190 is finished the firmware will also show it.

closed 2016-01-06 by repetier

frank16, 2016-01-06:
But it already shows 235 setpoint when the firmware is still in M190! M109 is not even executed yet...
Could it be repetier host is not interpreting M104? In that case when the heater is activated it would just show the last value for the setpoint which also happens to be 235 in my case (leftover from the last run or perhaps the value from the manual setting).
And what about the graph not being updated anymore? I noticed it starts again later. Probably around when the nozzle starts moving.

repetier, 2016-01-06:
The host sends 2-5 commands in advance. Since it does not know when it gets executed, it assumes it is executed so it shows always some commands in advance of what firmware does.

The graph not being updates is a marlin thing. It does not send temperatures while waiting for heaters and as it is a blocking command the host can not ask for it. Once we can ask again you see the graph.

frank16, 2016-01-06:
Ah,
now it's clear! Thank you kindly for taking the time to explain!