repetier/RepetierFirmware #451: Compile Error on Latest work092 Branch

floyd871, 2016-01-06:
Seems something to to with the Userinterface i have selected (RepRapDiscount Full Grapic...)
When i use no Display all works fine.

Arduino 1.6.7 Error output:
----------------------------------------------------------------------------------------------------------
In file included from sketch\HAL.h:83:0,

                 from sketch\Repetier.h:468,

                 from sketch\ui.cpp:20:

sketch\ui.h: In function 'void uiInitKeys()':

fastio.h:34: error: 'DIOUI_RESET_PIN_DDR' was not declared in this scope

 #define  _SET_INPUT(IO)  do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)

                              ^

sketch\fastio.h:54:26: note: in expansion of macro '_SET_INPUT'

 #define  SET_INPUT(IO)   _SET_INPUT(IO)

                          ^

sketch\ui.h:328:38: note: in expansion of macro 'SET_INPUT'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                      ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

In file included from sketch\HAL.h:83:0,

                 from sketch\Repetier.h:468,

                 from sketch\ui.cpp:20:

fastio.h:34: error: 'DIOUI_RESET_PIN_PIN' was not declared in this scope

 #define  _SET_INPUT(IO)  do {DIO ##  IO ## _DDR &= ~MASK(DIO ## IO ## _PIN); } while (0)

                                                          ^

sketch\fastio.h:17:30: note: in definition of macro 'MASK'

  #define  MASK(PIN)    (1 << PIN)

                              ^

sketch\fastio.h:54:26: note: in expansion of macro '_SET_INPUT'

 #define  SET_INPUT(IO)   _SET_INPUT(IO)

                          ^

sketch\ui.h:328:38: note: in expansion of macro 'SET_INPUT'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                      ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

fastio.h:29: error: 'DIOUI_RESET_PIN_WPORT' was not declared in this scope

 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)

                                       ^

sketch\fastio.h:49:37: note: in expansion of macro '_WRITE'

 #define     PULLUP(IO,v)            _WRITE(IO, v)

                                     ^

sketch\ui.h:328:53: note: in expansion of macro 'PULLUP'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                                     ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

fastio.h:29: error: 'DIOUI_RESET_PIN_PIN' was not declared in this scope

 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)

                                                                    ^

sketch\fastio.h:17:30: note: in definition of macro 'MASK'

  #define  MASK(PIN)    (1 << PIN)

                              ^

sketch\fastio.h:49:37: note: in expansion of macro '_WRITE'

 #define     PULLUP(IO,v)            _WRITE(IO, v)

                                     ^

sketch\ui.h:328:53: note: in expansion of macro 'PULLUP'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                                     ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

fastio.h:29: error: 'DIOUI_RESET_PIN_WPORT' was not declared in this scope

 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)

                                                                                                ^

sketch\fastio.h:49:37: note: in expansion of macro '_WRITE'

 #define     PULLUP(IO,v)            _WRITE(IO, v)

                                     ^

sketch\ui.h:328:53: note: in expansion of macro 'PULLUP'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                                     ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

fastio.h:29: error: 'DIOUI_RESET_PIN_PIN' was not declared in this scope

 #define  _WRITE(IO, v)   do { if (v) {DIO ##  IO ## _WPORT |= MASK(DIO ## IO ## _PIN); } else {DIO ##  IO ## _WPORT &= ~MASK(DIO ## IO ## _PIN); }; } while (0)

                                                                                                                              ^

sketch\fastio.h:17:30: note: in definition of macro 'MASK'

  #define  MASK(PIN)    (1 << PIN)

                              ^

sketch\fastio.h:49:37: note: in expansion of macro '_WRITE'

 #define     PULLUP(IO,v)            _WRITE(IO, v)

                                     ^

sketch\ui.h:328:53: note: in expansion of macro 'PULLUP'

 #define UI_KEYS_INIT_BUTTON_LOW(pin) SET_INPUT(pin);PULLUP(pin,HIGH);

                                                     ^

sketch\ui.h:843:3: note: in expansion of macro 'UI_KEYS_INIT_BUTTON_LOW'

   UI_KEYS_INIT_BUTTON_LOW(UI_RESET_PIN); // Kill pin

   ^

sketch\ui.h: In function 'void uiCheckKeys(uint16_t&)':

fastio.h:27: error: 'DIOUI_RESET_PIN_RPORT' was not declared in this scope

 #define  _READ(IO)     ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))

                                ^

sketch\fastio.h:46:23: note: in expansion of macro '_READ'

 #define  READ(IO)     _READ(IO)

                       ^

sketch\ui.h:334:44: note: in expansion of macro 'READ'

 #define UI_KEYS_BUTTON_LOW(pin,action_) if(READ(pin)==0) action=action_;

                                            ^

sketch\ui.h:854:2: note: in expansion of macro 'UI_KEYS_BUTTON_LOW'

  UI_KEYS_BUTTON_LOW(UI_RESET_PIN,UI_ACTION_RESET);

  ^

fastio.h:27: error: 'DIOUI_RESET_PIN_PIN' was not declared in this scope

 #define  _READ(IO)     ((bool)(DIO ## IO ## _RPORT & MASK(DIO ## IO ## _PIN)))

                                                           ^

sketch\fastio.h:17:30: note: in definition of macro 'MASK'

  #define  MASK(PIN)    (1 << PIN)

                              ^

sketch\fastio.h:46:23: note: in expansion of macro '_READ'

 #define  READ(IO)     _READ(IO)

                       ^

sketch\ui.h:334:44: note: in expansion of macro 'READ'

 #define UI_KEYS_BUTTON_LOW(pin,action_) if(READ(pin)==0) action=action_;

                                            ^

sketch\ui.h:854:2: note: in expansion of macro 'UI_KEYS_BUTTON_LOW'

  UI_KEYS_BUTTON_LOW(UI_RESET_PIN,UI_ACTION_RESET);

  ^

exit status 1
'DIOUI_RESET_PIN_DDR' was not declared in this scope

repetier, 2016-01-07:
Yes that display uses a define UI_RESET_PIN that does not get set by your display/board combination. Check ui.h and set it accordingly. Maybe your combination is not included at all, but recently gave my the pins for megatronics 3 whcih is what you use I think. Please let me know what pin would work as reset pin (it's just the small switch we use as reset). I have no megatronics for testing.

floyd871, 2016-01-07:
That works!! Great, i have added in line 713:
....
#define UI_RESET_PIN -1
.....

But after that i get another compile error.  UI_INVERT_MENU_DIRECTION seems to be double defined so I have to uncomment line 837:
...
//#define UI_INVERT_MENU_DIRECTION 0
...

repetier, 2016-01-07:
Should not cause a error, only a warning. Anyhow you have set it to 1 so you do not want 0. For next update I have set it to
#ifndef UI_INVERT_MENU_DIRECTION
#define UI_INVERT_MENU_DIRECTION 0
#endif
so it doe snot override set values.

closed 2016-01-07 by repetier