repetier/RepetierFirmware #452: Mixing extruder issue

Paxy, 2016-01-13:
Bug with mixing extruder when mixing is enabled and two extruders are set.
By default only first extruder is used and slicer sliced only for first extruder. (G-code in att)
After while, extruder stops extruding (no blockage, just motor stops spinning).
Did not change default weight values in EEPROM (attachment) with guess that way only first extruder would be used.

When I disable MIXING_EXTRUDER print with same G-code is fine.
[12.1.2016.txt](https://github.com/repetier/Repetier-Firmware/files/88831/12.1.2016.txt)
[k800carriage3.gcode.txt](https://github.com/repetier/Repetier-Firmware/files/88832/k800carriage3.gcode.txt)

Version 0.92.8 from 12.1.2016.

repetier, 2016-01-13:
Is there a communication error when it stops working? You might even see
SelectExtruder:<number>

I have that currently with a print and am testing a fix for it. Problem is that some com errors result in a extruder switch( in my case T1). So if it selects an extruder with no weights that might be the same problem.

Paxy, 2016-01-13:
Unfortunately, I have not kept Server's log file before successful print (without mixing), but I do not thing that there was any communication error as X,Y movement was correct. Just Extruder did not move.   

EEPROM default values for extruders looks strange (two extruders with mixing, in attachment). Looks like for E0 it is 10 and for E1 it is -1.  Is this correct value ? Maybe this default values are incorrect ?

Paxy, 2016-01-13:
Let mi show you printed parts. Two of them, bottom, are with same printing issue, when extruder stopped (at same place). Third one (top) is same part with MIXING disabled.
[image]