repetier/RepetierFirmware #453: Z_PROBE_BED_DISTANCE is not used

Nandox7, 2016-01-18:
Hi,

The firmware disregards the value set in Z_PROBE_BED_DISTANCE when doing the probing.
So if I start the process with Z at 35 it will return to 35 to move to the next point instead of to the value
set with Z_PROBE_BED_DISTANCE that in my case is 5 (e believe is also the default)

repetier, 2016-01-18:
Are you speaking about G30 or G32/G33/G29. G32 uses it. G30 does only single probe without extra work I think.

Nandox7, 2016-01-18:
It's G32/G29 (I haven't tried G33).
So far is all G code that makes probing at different points along the bed.

orgemd, 2016-01-20:
I can confirm this for 92.8 configured from the web interface on a cartesian printer.  G32 doing a grid always returns to the height from which G32 was issued.  If I send G32 Sx from 25mm, the Z will lift to 25mm between each probe.  If I send it from 15mm, the Z will lift to 15mm between each probe.

repetier, 2016-01-21:
Did you home before running G32? Without homing z is normally incorrect and may prevent going to right height.

Nandox7, 2016-01-21:
Shouldn't it work independently if the Z is homed or not?
As soon as the probe triggers it gets the min value (same as homing) and can work form there.

repetier, 2016-01-21:
No. z height adjustment is done at the beginning. Afterwards it always goes to start position so all probes measure the same height. So if you start at z=30 without homing printer thinks it is z=0 and handles accordingly. That does not mean you can not use it without homing (also with z max homing you should not use S parameter or you end up with wrong length since you did not home and assumed position was wrong), but if you do you should know about the consequences.

Nandox7, 2016-01-21:
Ok, thanks.

So to be sure, the correct steps are:
1. Home X, Y
2. Home Z (I'm putting it separate as it usually required some extra steps, deploy probe or move to a position above the bed in case X,Y home is a corner)
3. Execute G32/G33/G29 command

I'll give this a try tonight.

Nandox7, 2016-06-29:
Seems that this no longer an issue in the latest dev version.
After doing the first probe point the probe won't raise all the way to the same height as when it started but just a smaller distance that I believe matches what is set in Z_PROBE_BED_DISTANCE.
This makes the all process much much faster.

Taz-Baker, 2016-07-12:
Hi all, I think that I have been looking at this all wrong. To confirm, the z homeing is still done with the limit switch set up (configured as Z_MIN_ENDSTOP) and the Z_Probing is done with a second input - say Z_MAX_ENDSTOP.  I have just started to look into setting this up and had it in my head that the Z_Probing limit does both.

repetier, 2016-07-12:
@Taz-Baker watch out - z min endstop makes no sense if it is not the z probe! It is only allowed to trigger at lowest pos on bed and then height does not match anyway. Use z max for homing (best) or z probe as min endstop so you have z position and xy position to make sense of it (requires xy homing first then).

Closing as the issue is solved in dev version.

closed 2016-07-12 by repetier