repetier/RepetierFirmware #456: ENDSTOP_Z_BACK_MOVE not honored with z-max on cartesian - 92.8

orgemd, 2016-01-20:
My cartesian printer has endstops at X-min, Y-min, and Z-max.  When homing all axis (XYZ order), X and Y do as expected.  Then Z raises until the Z-max endstop is triggered, then backs off a bit, and then returns to the Z-max endstop at a slower speed.  That is all as expected.  Once Z-max is triggered at the slower speed, the Z drops away from the endstop some standard distance.  It is hard to tell what that distance is - probably something between 3mm and 5mm.  The issue is that I have ENDSTOP_Z_BACK_MOVE set to 0 - so there should be no movement away from the endstop after the second trigger.  I also have ENDSTOP_X_BACK_MOVE=0 and ENDSTOP_Y_BACK_MOVE=0, and both of those are working the way I would expect.  The X and Y stop after the second endstop tap without moving away from the endstop.  Configuration was done via the web interface and it is version 92.8.

repetier, 2016-01-21:
ENDSTOP_Z_BACK_MOVE must be > 0 for the retest of endstop. After that it moves away from endstop ENDSTOP_Z_BACK_ON_HOME mm if set.

orgemd, 2016-01-21:
Sorry, I took the variable name from the web configurator.  If you look at the v092 configurator, you will see that it mislabels the distance after homing as ENDSTOP_(X|Y|Z)_BACK_MOVE.  (That should be changed.)  I verified with the config file that the first row is really BACK_MOVE and the third row is really BACK_ON_HOME - just as you say.

The problem still seems to exist - but the ignored variable is ENDSTOP_Z_BACK_ON_HOME.  Through the configurator I have that set to 0 for X, Y, Z.  X and Y are honored - Z is not.  Looking at the relevant portion of the config, all variables seem to work as expected except that one.

// ################ Endstop configuration #####################

#define ENDSTOP_PULLUP_X_MIN true
#define ENDSTOP_X_MIN_INVERTING false
#define MIN_HARDWARE_ENDSTOP_X true
#define ENDSTOP_PULLUP_Y_MIN true
#define ENDSTOP_Y_MIN_INVERTING false
#define MIN_HARDWARE_ENDSTOP_Y true
#define ENDSTOP_PULLUP_Z_MIN true
#define ENDSTOP_Z_MIN_INVERTING false
#define MIN_HARDWARE_ENDSTOP_Z true
#define ENDSTOP_PULLUP_X_MAX true
#define ENDSTOP_X_MAX_INVERTING false
#define MAX_HARDWARE_ENDSTOP_X false
#define ENDSTOP_PULLUP_Y_MAX true
#define ENDSTOP_Y_MAX_INVERTING false
#define MAX_HARDWARE_ENDSTOP_Y false
#define ENDSTOP_PULLUP_Z_MAX true
#define ENDSTOP_Z_MAX_INVERTING false
#define MAX_HARDWARE_ENDSTOP_Z true
#define max_software_endstop_r true

#define min_software_endstop_x false
#define min_software_endstop_y false
#define min_software_endstop_z false
#define max_software_endstop_x true
#define max_software_endstop_y true
#define max_software_endstop_z false
#define ENDSTOP_X_BACK_MOVE 5
#define ENDSTOP_Y_BACK_MOVE 5
#define ENDSTOP_Z_BACK_MOVE 2
#define ENDSTOP_X_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Y_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_Z_RETEST_REDUCTION_FACTOR 3
#define ENDSTOP_X_BACK_ON_HOME 0
#define ENDSTOP_Y_BACK_ON_HOME 0
#define ENDSTOP_Z_BACK_ON_HOME 0
#define ALWAYS_CHECK_ENDSTOPS 1

repetier, 2016-01-21:
You are right regarding the config tool misspelling.

After reading the code for 30 minutes I still see no error. There is one additional factor that is bed coating thickness that changes end position. So maybe you have set in eeprom to non zero.

orgemd, 2016-01-26:
I am afraid I flashed to a different firmware to get back to printing - so testing much else on this one will be difficult - but here is what I checked before changing firmware in case it helps you.

I confirmed that the EEPROM values for the coating thickness were zero, as they were in the config file - so it isn't a non-zero value there.  I also tried setting the Z back move to 0.  I thought maybe that was just being applied a second time after slow homing.  After that change, the Z would go until it hit the max endstop, and then still would immediately drop about 5mm to stop well off of the endstop switch.  In other words, changing back move did not improve the problem - it just controlled what it is supposed to control.