repetier/RepetierFirmware #462: Massive retraction during warmup - unit confusion

JBKingdon, 2016-01-24:
Hi,

After using the online configuration tool and uploading the new firmware (with a new eeprom set), during warmup my 5mm wait retract turned into a 900mm one, which took quite a while! It seems the config tool puts the requested retract distance times the steps/mm setting into the config, where as the firmware is expected a value in mm. Editing the eeprom value with r-host is a work around.

Regards,

repetier, 2016-01-25:
Are you using latest 0.92.8 version? I remember having changed it sometime, but checking the code it nowhere has a conversion. It's always mm.

JBKingdon, 2016-01-25:
Yes, sorry, I should have given that. 0.92.8 downloaded using the config tool on Jan 21 2016.

The issue I think must be in the config tool, as the large value is in the downloaded configuration.h:

#define EXT0_WAIT_RETRACT_UNITS 925

Regards,

repetier, 2016-01-25:
Ok, that helped. It was the config tool still doing the multiply while firmware was already fixed to accept mm instead of steps. Should now be fixed for next download.

closed 2016-01-25 by repetier

JBKingdon, 2016-01-25:
Great work - many thanks.