repetier/RepetierFirmware #464: Check Z-Probe before attempting each Measurement in Autolevel Procedure

simonkuehling, 2016-02-03:
Hi Roland,

i am extensively testing the autolevel routine with the latest 0.92 branch.
Due to the specific design of my sensor, it can happen that the probe signal is already HIGH before attempting the first z-measurement, or does not reset to LOW successfully before the next measurement. This is not due to bad design, but a sign for misalignment or dirt on exposed contacts (maloperation or the like)

I propose to add a safety check before each measurement for a valid sensor state (LOW in my case). Otherwise the system currently skips the point and assumes a measurement of "0", which is always wrong - either because the probe is already touching the build surface or because the sensor isn't ready.
The proposed behavior should be some kind of dry-run/full-stop to reliably kill the print job.

repetier, 2016-02-03:
The tests itself are already there like
    if(Endstops::zProbe()) {
        Com::printErrorFLN(PSTR("z-probe triggered before starting probing."));
        return ILLEGAL_Z_PROBE;
    }

which does the test at beginning you mention. This skips complete G32/G33 operation with a error message.

Problem is more what should happen after a failed autolevel command. Here options will always differ as it depends. I never use them inside gcode programs so for me a notice that it failed is all I want. Inside a program it is more critical as the rest would want to execute none the less. Now what should we do? Kill printer so it hangs and does nothing? Would get many complains that firmware hangs especially from those not reading log messages. Send a fatal warning, so host could stop sending data - well we already may have buffered commands that could be a problem. But ignoring all following commands would be as bad as kill. Maybe the best solution would be to introduce what Marlin does with heater problems - ignore all commands until M999 resets fault state and return a error message for each requested line. That would allow reenabling without reset and at some point users might notice the reason and no damage should occur as printer stops doing things. I think this would be my favourite solution. What do you think?

simonkuehling, 2016-02-03:
Oh, i missed the commit where you added these checks recently - i should never say "latest version" without double checking, i guess... 

Agreed, let's turn to the interesting problem then. I'd like the auto levelling routine to be robust enough to be performed at the beginning of each gcode - so that the machine can handle whatever print surface (of different thickness’s for example) you throw at it.
The levelling is totally awesome already and i get more than 99% success ratio from the auto level routine. The only failures come from maloperation or if anything gets stuck or mechanically blocked during levelling - which will in any case require a user intervention.

I really like the Marlin behaviour you describe above - also for heater problems this would be a much cleaner way to handle these error states. In that case the host can abort the print job, prominently display an error screen, calling the user to remedy the issue and offer an "ok, continue" button to reactivate the firmware without resetting. Definitely my favorite solution as well.

repetier, 2016-02-06:
Ok, fatal error is now (theoretical) included. Send M999 S1 to force a fatal error for testing your host. Simply scan response for M999 to stop your running job.

closed 2016-02-06 by repetier