repetier/RepetierFirmware #467: Compile failure with FEATURE_CONTROLLER set

PxT, 2016-02-06:
I'm unable to compile with the Display controller set to any option > 0.  Configuration below, and compile time errors beneath.  If I set FEATURE_CONTROLLER to 0 then everything compiles fine. I am using the latest github master branch.

`{
    "editMode": 2,
    "processor": 0,
    "baudrate": 250000,
    "bluetoothSerial": -1,
    "bluetoothBaudrate": 250000,
    "xStepsPerMM": 80,
    "yStepsPerMM": 80,
    "zStepsPerMM": 80,
    "xInvert": "1",
    "xInvertEnable": 0,
    "eepromMode": 2,
    "yInvert": "0",
    "yInvertEnable": 0,
    "zInvert": "1",
    "zInvertEnable": 0,
    "extruder": [
        {
            "id": 0,
            "heatManager": 1,
            "pidDriveMin": 40,
            "pidDriveMax": 230,
            "pidMax": 255,
            "sensorType": 97,
            "sensorPin": "TEMP_0_PIN",
            "heaterPin": "HEATER_0_PIN",
            "maxFeedrate": 100,
            "startFeedrate": 20,
            "invert": "1",
            "invertEnable": "0",
            "acceleration": 6500,
            "watchPeriod": 1,
            "pidP": 31.36,
            "pidI": 2.18,
            "pidD": 112.9,
            "advanceK": 0,
            "advanceL": 0,
            "waitRetractTemp": 150,
            "waitRetractUnits": 25,
            "waitRetract": 25,
            "stepsPerMM": 92.4,
            "coolerPin": 7,
            "coolerSpeed": 255,
            "selectCommands": "",
            "deselectCommands": "",
            "xOffset": 0,
            "yOffset": 0,
            "zOffset": 0,
            "xOffsetSteps": 0,
            "yOffsetSteps": 0,
            "zOffsetSteps": 0,
            "stepper": {
                "name": "Extruder 0",
                "step": "ORIG_E0_STEP_PIN",
                "dir": "ORIG_E0_DIR_PIN",
                "enable": "ORIG_E0_ENABLE_PIN"
            },
            "advanceBacklashSteps": 0,
            "decoupleTestPeriod": 12,
            "jamPin": -1,
            "jamPullup": "0"
        }
    ],
    "uiLanguage": 0,
    "uiController": 0,
    "xMinEndstop": 0,
    "yMinEndstop": 0,
    "zMinEndstop": 0,
[... 504 lines cut ...]
Using previously compiled file: /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/HAL.cpp.o
Using previously compiled file: /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/Printer.cpp.o
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI" "/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/Repetier.ino.cpp" -o "/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/Repetier.ino.cpp.o"
Using previously compiled file: /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/SDCard.cpp.o
Using previously compiled file: /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/SdFat.cpp.o
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/mega" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI" "/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp" -o "/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp.o"
In file included from /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/HAL.h:33:0,
                 from /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/Repetier.h:480,
                 from /var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp:24:
/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp: In static member function 'static void GCode::reportFatalError()':
ui.h:1995: error: initializer fails to determine size of '__c'
 #define UI_ERROR_UPD(status) {uid.setStatusP(PSTR(status),true);uid.refreshPage();}
                                              ^
/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp:1044:2: note: in expansion of macro 'UI_ERROR_UPD'
  UI_ERROR_UPD(fatalErrorMsg)
  ^
ui.h:1995: error: array must be initialized with a brace-enclosed initializer
 #define UI_ERROR_UPD(status) {uid.setStatusP(PSTR(status),true);uid.refreshPage();}
                                              ^
/var/folders/rt/b3z5zfgx1g95gtqkq161ppt40000gn/T/build7bddd69c652611a4e8cc813cf7b7cb1a.tmp/sketch/gcode.cpp:1044:2: note: in expansion of macro 'UI_ERROR_UPD'
  UI_ERROR_UPD(fatalErrorMsg)
  ^
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SPI 
exit status 1
initializer fails to determine size of '__c'

PxT, 2016-02-07:
Sorry for the wall of text above, github refused to let me format it.

I tried to use git bisect, and it looks like this error was introduced with commit ce05b5f5acf7000e93937f4448e08d6293799baf

repetier, 2016-02-07:
Thanks for the info. Compile issue should be fixed now.

closed 2016-02-07 by repetier