repetier/RepetierFirmware #469: Delta Z tower homes intermittently

rcarlyle, 2016-02-08:
I'm stuck on this one. I think I have ruled out all possible hardware causes. 

When I first power on or power-cycle my delta, the X and Y towers will home and jog while the Z tower remains stationary. The Z stepper is enabled/disabled normally, and the endstop is not triggering, but the Z tower doesn't move at all in either direction.

If I power cycle and then press reset (via GLCD button) before doing anything, the Z tower continues to not function.

If I power cycle, attempt any kind of move (jog or home) with the Z tower not working, and THEN press reset, **the Z tower will start working** after the reset. It will then continue working properly through any number of resets until the printer is power-cycled. Power cycling reverts to the non-functioning Z tower.

So, a couple things come to mind:
1) Something is initializing differently when the board powers up versus doing a commanded reset
2) Commanding a move is doing something to the firmware state that allows it to move **after** a reset but not before

Printer info:
- Due/RAMPS-FDv2
- Issue occurs with and without an OctoPi hooked up
- Issue occurs with both homing and jogging
- Issue occurs with both 0.92.3 and 0.92.8
- Issue occurs with both mechanical endstops and RADDS-style Hall-effect endstops
- Endstop cables are shielded, with the drain grounded at the board end
- Motor and heater wiring is twisted and not parallel to endstop cables
- M119 reports proper values at all times
- Issue stays with Z axis if I switch motors, drivers, or endstops
- I am switching the AC side of the PSU when I power cycle
- RepRapDiscount Full Graphic LCD
- I am using a button on the GLCD to reset the board to get it working
- Does not matter if the SD card is installed
- Watchdog is enabled
- Z probing is enabled via Z-min endstop, and works fine when the Z tower is able to move

[Configuration.txt](https://github.com/repetier/Repetier-Firmware/files/121164/Configuration.txt)

repetier, 2016-02-08:
Just to make it clear - the firmware does not detect reset and always does the same - I mean for power on or reset. Always the same routine. A known issue is that some due boards do not start properly on power up as they start too early. So they hang completely on first power and start working after a reset. Sounds like it could be the same problem just with a slightly better timing but still the arm not ready 100% when firmware starts. The solution seems to solder a small capacitor to reset to extend time in power up. Or just as you did hitting reset after first power up.

rcarlyle, 2016-02-08:
I will try that, but it doesn't explain why:
1) If I power cycle and reset without moving anything, Z does not work and will never work no matter how many times I reset
2) If I power cycle and jog or home, and then reset, Z starts working and will always work no matter how many times I reset (until power cycled again)

I can't see any way that moving Z affects the board's reset, but it does. 

I would suspect something with the 3v3-5v level shifters on the RAMPS-FDv2 board, but Z-ENABLE always works fine (which is level-shifted) and STEP is not level-shifted.

repetier, 2016-02-08:
You are right. The reset problem I mentioned would not really explain 1). All outputs get defined right startup so in an very early stage. You might try adding a delay in Printer::setup at the beginning
HAL::delayMilliseconds(100);
I guess the reason it does not work is the step/dir might be still set as inputs also that code has run. Most stepper driver would work with 3.3 for all 3 inputs.

As a last resort, try using a unused extruder socket for z axis instead.

rcarlyle, 2016-02-08:
Ok, I tried 100 and 1,000 us delays here:
```
void Printer::setup()
{
    HAL::stopWatchdog();
    HAL::delayMilliseconds(1000); // ADDED 2/8/16 TO TEST Z TOWER NOT MOVING
```
Aside from slowing down the LCD screen coming on, it had no effect. 

I will try swapping driver sockets next.

rcarlyle, 2016-02-08:
I just discovered something new -- if I plug in to USB (or a USB charger) and let the firmware load, THEN turn on the 24v PSU, the Z tower works immediately!  So that is a big piece of evidence for an issue with voltages at initial power on. 

To summarize:
1) Turn on PSU: Z doesn't work
2) Turn on PSU and reset: Z doesn't work
3) Turn on PSU then move/jog then reset: Z DOES work
4) Plug in USB power source then turn on PSU: Z DOES work
5) Plug in USB power, turn on PSU, remove USB power: Z DOES work
6) Turn on PSU, then plug in USB: Z doesn't work
7) Turn on PSU, then plug in USB, then reset: Z doesn't work

So there are two ways to make it work, either start with USB 5v or jog the motors and then reset. 

I can understand why the clean 5v power supply at startup would help, but I do not understand why moving the motors is required to make it reset and work from the PSU. That is very strange.

rcarlyle, 2016-02-08:
I switched the Z tower to the E1 and E2 driver sockets and the problem persisted. I then switched Y and Z sockets and the problem switched to Y. So it's tied to some particular Due pins, I guess? Very strange.

Probably unrelated, but my E1 socket will only run the stepper in one direction, but I don't see any reason why. Acts like DIR is not toggling. I'm not gonna worry about that right now. Aside from the DIR issue, it acts the same as the Z and E2 sockets. 

Any other suggestions before I unwire everything to start soldering on capacitors and trying different boards?

repetier, 2016-02-08:
Since delay makes no difference I can not think the capacitor will help.  
That switching Y and Z makes Y fail shows that is more bound to pins/sockets of the board then firmware initialization. Here the initalization:

    //Initialize Step Pins
    SET_OUTPUT(X_STEP_PIN);
    SET_OUTPUT(Y_STEP_PIN);
    SET_OUTPUT(Z_STEP_PIN);
	endXYZSteps();
    //Initialize Dir Pins
#if X_DIR_PIN > -1
    SET_OUTPUT(X_DIR_PIN);
#endif
#if Y_DIR_PIN > -1
    SET_OUTPUT(Y_DIR_PIN);
#endif
#if Z_DIR_PIN > -1
    SET_OUTPUT(Z_DIR_PIN);
#endif

    //Steppers default to disabled.
#if X_ENABLE_PIN > -1
    SET_OUTPUT(X_ENABLE_PIN);
    WRITE(X_ENABLE_PIN, !X_ENABLE_ON);
#endif
#if Y_ENABLE_PIN > -1
    SET_OUTPUT(Y_ENABLE_PIN);
    WRITE(Y_ENABLE_PIN, !Y_ENABLE_ON);
#endif
#if Z_ENABLE_PIN > -1
    SET_OUTPUT(Z_ENABLE_PIN);
    WRITE(Z_ENABLE_PIN, !Z_ENABLE_ON);
#endif

as you see z is always last, so if processor is ready for x/y why should it not be ready for z/E1/E2. But you already proved it is more a hardware thing since powering order makes also a difference. But that is not my field, i'm only a programmer.

rcarlyle, 2016-02-08:
Yep, I'm guessing it's something in the Due, I will try some hardware changes. Thank you for your help!

closed 2016-02-08 by rcarlyle

rcarlyle, 2016-02-17:
Quick update, I swapped to RADDS and a Due R3-E and the problem went away. I don't know whether the previous Due R3 or RAMPS-FDv2 was causing the issue, but in either case, it looks like hardware was confirmed as the problem. Thanks for the assistance.