repetier/RepetierFirmware #470: M205 returns NAN for Filament Printed

errolt, 2016-02-09:
After upgrading to the latest version the M205 command suddenly returns NAN for Filament Printed. This gives me a red box on the EEPROM editing screen in Repetier Host...

10:14:26.635 : EPR:0 1028 0 Language
10:14:26.635 : EPR:2 75 250000 Baudrate
10:14:26.639 : EPR:3 129 NAN Filament printed [m]
10:14:26.641 : EPR:2 125 1754556 Printer active [s]
10:14:26.641 : EPR:2 79 600000 Max. inactive time [ms,0=off]

Thank you,
Errol

repetier, 2016-02-09:
NaN = Not a Number. Set it to 0 and it should be gone. Might be from a result from failed eeprom upgrade. Please report if it comes back after a print.

errolt, 2016-02-09:
Hmm, I have set it to 0 every time I edited the EEPROM. Didn't work. Setting it to 1 did the trick though... :)

closed 2016-02-09 by errolt

repetier, 2016-02-09:
That is a problem I will test. All printers start with 0 m printed length so that special value should work as well.

errolt, 2016-02-09:
I don't know if Repetier Host maybe didn't write "0" to the eeporm because it thought it "was" zero, and thus needed no update. I wrote 1 to the field which worked fine, then I wrote 0 and it also wrote fine...

Thank you,
Errol