repetier/RepetierFirmware #472: Autoleveling doesn't work - Z-probe:NAN

mikozak, 2016-02-14:
**My configuration:**
- Kossel Mini
- RAMPS 1.4 with Arduino Mega
- Repetier 0.92.8

**Issue**
Recently I've installed z-probe using simple switch.
Configured autoleveling using web interface (3 points autoleveling)
When I issue G32 it doesn't work, procedure stops and printer reports an error. Here are the logs copied from repetier host (note line with Z-probe:NaN):

> 20:07:02.838 : Printer reset detected - initalizing
> 20:07:02.841 : start
> 20:07:02.842 : Info:External Reset
> 20:07:02.851 : Transformation matrix: 1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000
> 20:07:05.509 : Free RAM:984
> 20:07:05.509 : Autoretract:0
> 20:07:05.513 : X:0.00 Y:0.00 Z:0.000 E:0.0000
> 20:07:05.513 : SelectExtruder:0
> 20:07:05.537 : FIRMWARE_NAME:Repetier_0.92.8 FIRMWARE_URL:https://github.com/repetier/Repetier-Firmware/ PROTOCOL_VERSION:1.0 MACHINE_TYPE:Delta EXTRUDER_COUNT:1 REPETIER_PROTOCOL:3
> 20:07:05.538 : Printed filament:0.00m Printing time:0 days 0 hours 0 min
> 20:07:05.538 : PrinterMode:FFF
> 20:07:05.556 : X:0.00 Y:0.00 Z:0.000 E:0.0000
> 20:07:05.556 : DebugLevel:6
> 20:07:05.556 : SelectExtruder:0
> 20:07:05.556 : Begin file list
> 20:07:05.559 : End file list
> 20:07:05.560 : DebugLevel:6
> 20:07:05.562 : SelectExtruder:0
> 20:07:24.576 : EPR:0 1028 0 Language
> 20:07:24.578 : EPR:2 75 115200 Baudrate
> 20:07:24.598 : EPR:3 129 0.000 Filament printed [m]
> 20:07:24.601 : EPR:2 125 0 Printer active [s]
> 20:07:24.610 : EPR:2 79 0 Max. inactive time [ms,0=off]
> 20:07:24.612 : EPR:2 83 360000 Stop stepper after inactivity [ms,0=off]
> 20:07:24.614 : EPR:3 11 99.5000 Steps per mm
> 20:07:24.617 : EPR:3 23 150.000 Max. feedrate [mm/s]
> 20:07:24.619 : EPR:3 35 90.000 Homing feedrate [mm/s]
> 20:07:24.621 : EPR:3 39 10.000 Max. jerk [mm/s]
> 20:07:24.623 : EPR:3 133 0.000 X min pos [mm]
> 20:07:24.625 : EPR:3 137 0.000 Y min pos [mm]
> 20:07:24.627 : EPR:3 141 0.000 Z min pos [mm]
> 20:07:24.629 : EPR:3 145 200.000 X max length [mm]
> 20:07:24.632 : EPR:3 149 200.000 Y max length [mm]
> 20:07:24.634 : EPR:3 153 219.443 Z max length [mm]
> 20:07:24.636 : EPR:3 59 1000.000 Acceleration [mm/s^2]
> 20:07:24.638 : EPR:3 71 100.000 Travel acceleration [mm/s^2]
> 20:07:24.641 : EPR:3 881 212.300 Diagonal rod length [mm]
> 20:07:24.644 : EPR:3 885 105.280 Horizontal rod radius at 0,0 [mm]
> 20:07:24.646 : EPR:3 925 75.000 Max printable radius [mm]
> 20:07:24.649 : EPR:1 891 70 Segments/s for travel
> 20:07:24.651 : EPR:1 889 180 Segments/s for printing
> 20:07:24.655 : EPR:1 893 29 Tower X endstop offset [steps]
> 20:07:24.657 : EPR:1 895 0 Tower Y endstop offset [steps]
> 20:07:24.660 : EPR:1 897 69 Tower Z endstop offset [steps]
> 20:07:24.662 : EPR:3 901 210.000 Alpha A(210):
> 20:07:24.665 : EPR:3 905 330.000 Alpha B(330):
> 20:07:24.667 : EPR:3 909 90.000 Alpha C(90):
> 20:07:24.670 : EPR:3 913 0.000 Delta Radius A(0):
> 20:07:24.672 : EPR:3 917 0.000 Delta Radius B(0):
> 20:07:24.675 : EPR:3 921 0.000 Delta Radius C(0):
> 20:07:24.677 : EPR:3 933 0.757 Corr. diagonal A [mm]
> 20:07:24.679 : EPR:3 937 0.581 Corr. diagonal B [mm]
> 20:07:24.682 : EPR:3 941 0.000 Corr. diagonal C [mm]
> 20:07:24.684 : EPR:3 1024 0.000 Coating thickness [mm]
> 20:07:24.687 : EPR:3 808 5.910 Z-probe height [mm]
> 20:07:24.696 : EPR:3 929 2.000 Max. z-probe - bed dist. [mm]
> 20:07:24.699 : EPR:3 812 2.000 Z-probe speed [mm/s]
> 20:07:24.702 : EPR:3 840 50.000 Z-probe x-y-speed [mm/s]
> 20:07:24.705 : EPR:3 800 30.200 Z-probe offset x [mm]
> 20:07:24.707 : EPR:3 804 -6.670 Z-probe offset y [mm]
[... 104 lines cut ...]
> 20:17:01.850 : Warning:Invalid delta coordinate - move ignored x:-5917 y:647 z:-2147483648
> 20:17:01.854 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
> 20:17:01.863 : Warning:moveRelativeDistanceInSteps / queueDeltaMove returns error
> 20:17:01.863 : Info:Autoleveling enabled
> 20:17:01.863 : Error:G32
> 20:17:01.867 : X:NAN Y:NAN Z:NAN E:0.0000
> 20:17:01.870 : Info:Autoleveling disabled
> 20:17:01.961 : RequestPause:Homing failed!
> 20:17:01.962 : Error:A hit floor
> 20:17:01.964 : Error:A hit floor
> 20:17:01.971 : Warning:Invalid delta coordinate - move ignored x:0 y:0 z:2147483579
> 20:17:01.971 : Warning:in queueDeltaMove to calculateDeltaSubSegments returns error.
> 20:17:01.978 : Warning:moveRelativeDistanceInSteps / queueDeltaMove returns error
> 20:17:01.978 : Error:A hit floor
> 20:17:01.981 : SelectExtruder:0
> 20:17:01.985 : X:0.00 Y:0.00 Z:-21582750.000 E:0.0000
> 20:17:01.985 : Info:Autoleveling enabled
> 20:17:01.994 : DebugLevel:6
> 20:17:01.994 : SelectExtruder:0
> 20:18:08.343 : Error:isPositionAllowed
> 20:18:08.343 : X:NAN Y:NAN Z:NAN E:0.0000
> 20:18:08.347 : Error:isPositionAllowed
> 20:18:08.348 : X:NAN Y:NAN Z:NAN E:0.0000
> 20:18:08.351 : Error:isPositionAllowed
> 20:18:08.355 : X:NAN Y:NAN Z:NAN E:0.0000

mikozak, 2016-02-14:
For the example I decribed here I used 3 points probing.
Now I just reconfigured it to use grid 2x2 and it works without issues.

repetier, 2016-02-15:
Strange. The part that fails is z probing and that is the same for 3 point and 4 point measurement. It is also the same for G30 just as a test.  To fail with NaN one of the used values must be NaN, normally a eeprom variable, but they seem all correct. So I assume the difference you did betwenn made the problem vanish.

mikozak, 2016-02-15:
I discovered in code that function that corrects calculated z-probe distance by bending returns NaN

repetier, 2016-02-15:
I see. Your 3 points are on a line and that is not allowed. That creates a division by zero causing the NaN. For 3 point measurement these must be as wide aparat as possible in different corners. For 4 point also but I guess yo changed them to build the 2 axis fixing that.

closed 2016-02-15 by repetier