repetier/RepetierFirmware #473: Grid bed level coordinates exceeding bed size

frazierjason, 2016-02-16:
I am using RAMPS 1.4 controller with Repetier 0.92.8 downloaded on 11 February 2016, flashed with Arduino 1.6.5.

I recently switched from three point autoleveling to 5x5 grid leveling on my Cartesian WeldingRodBot (modified Cobblebot).  My effective movement area is 382 X, 371.5 Y, 387.91 Z with a dual nozzle and E1 is offset from E0 by 20.26mm X and -0.075mm Y.  The induction sensor is offset from E0 by 8.47 X 65.31 Y.

I originally started with a three point measurement at 15 X 15 Y, 350 X 15 Y, 290 X 350 Y.  This was working well.  I switched to grid measurement, and now the printer eventually moves outside of the bed area in the X axis in the ~385mm range, where no sensor measurement can be taken.  I have to kill the controller to prevent failure to sense and subsequent damage.

Here is the process I follow.  I boot/reset the printer, and separately send X home, Y home, Z home (to Z-max at the top of the printer).  Then I run G32 S3.

Just to troubleshoot the issue isn't me making a wrong calculation about sensor position offset from E0 or usable bed size, I severely reduced the grid measurement footprint to 5 X 5 Y, 5 X 220 Y, 220 X 220 Y.  I rebooted the controller, and I ensured to set these same values in EEPROM.  I have not attempted an EEPROM bank swap/purge.  The problem persists -- the firmware ignores the sizes I've selected, and attempts to seek far outside of the grid and eventually off the bed again.  Here are the coordinates the printer chooses to read, in order, with the reduced 220 by 220 grid bounds (X,Y):
15,15
66.25,66.25
117.5,117.5
168.75,168.75
220,220
66.25,15
117.5,66.52
168.75,117.5
220,168.75
271.25,220
117.5,15
168.75,66.25
220,117.5
168.75,66.25
220,117.5
271.25,168.75
322.5,220
168.75,15
220,66.25
271.25,117.5
322.5,168.75
373.75,220 <<< sensor now off the bed in the X axis by about 10mm, so I hit reset button

If I instead hover a metal razor under the sensor and let it continue read, it will keep sweeping in the same diagonal pattern as above until it takes a couple more readings hanging off the glass (can't see coords because I'm trying to avoid damage), then it stores the values.  It never actually reads the other diagonal half towards the 0 X and max Y of the bed at all.

Seems there is an error in a nested loop somewhere?

The configuration I expected to work:
http://www.neveroddoreven.com/cobblebot/Configuration_WeldingRodBot.h

The reduced grid configuration used for the coordinate sequence reproduced above:
http://www.neveroddoreven.com/cobblebot/Configuration_WeldingRodBot_reduced_grid.h

repetier, 2016-02-16:
You problem seems to be the control point
#define Z_PROBE_X1 5
#define Z_PROBE_Y1 5
#define Z_PROBE_X2 350
#define Z_PROBE_Y2 15
#define Z_PROBE_X3 350
#define Z_PROBE_Y3 290

They build a parallelogram of control points to measure with X1 = origin and X1->X2 and X1->X3 the other axis. So in this case you should use
#define Z_PROBE_X1 5
#define Z_PROBE_Y1 5
#define Z_PROBE_X2 350
#define Z_PROBE_Y2 5
#define Z_PROBE_X3 5
#define Z_PROBE_Y3 290

to cover a rectangular area.

frazierjason, 2016-02-18:
Thanks, this detail allowed me to update my values and enable matrix leveling.

For what it's worth, my previous set of coordinates also technically define a rectangular area (the Z_PROBE_Y2 was a typo, actually 5 in EEPROM), and they worked just fine for three-point measuring.  I chose them in that order [X@origin Y@origin, X@origin Y@max, X@max Y@max] because I am right handed and that positioning allowed me to reach the sensor points most easily.  It wasn't clear to me that the rectangular aspect is only considered by the firmware in a [X@origin Y@origin, X@max Y@origin, X@origin Y@max] orientation, and that the firmware couldn't understand a parallelogram defined in a different order.

The diagram and description in the Repetier Firmware configuration tool only talk about Side 1 an Side 2, without tying them to the X or Y axis.  They also leave off half way through the description they already have:  "The rectangle is computed as follows: Point 1 = origin, Side 1: Point 1 <-> Point 2, Side 2: Point 1 <-> "  <<< there is something missing at the end!!!

The diagram and description for matrix leveling could make things clearer, with hard requirements that P2 must be parallel to P1 along the X axis, and P3 must be parallel to P1 along the Y axis.  The statement that "Point 1 = origin" doesn't seem accurate, it is allowed to be offset in both X and Y from the actual 0X,0Y origin point.

repetier, 2016-02-18:
Thanks for the missing info hints. I have fixed it so that it is hopefully more understandable (and P3 is inserted).

closed 2016-02-18 by repetier