repetier/RepetierFirmware #474: Da vinci 2.0a duo Heater decoupled msg (if ony extruder is off)

immunity20, 2016-02-16:
Hello, i have a Davinci 2.0a duo and installed repetier 0.92
I slice using simplify3d my object and if i add temperature (240C abs) for only one extruder
printer will stop with heater decoupled message when it will be ready to print  bed and extruder1 temp reached but extruder2 temp is set to 40 . 
If i set both extruder1 and extruder2 temp 240 print will go correct (although filament in extruder2 will be heated with no reason :/ )

luc-github, 2016-02-16:
I always use one extruder on my duo and do not have this issue.

is this happen with specific extruder or both ?

you can disable decouple test in firmware before flashing

immunity20, 2016-02-16:
i think yes it happens on both (for sure on T0 ) (right filament).
i use simplify3d to create the gcode (printer config has T0 T1) i will try
removing T1 from printer settings to see what happens

repetier, 2016-02-16:
40°C for second extruder is bad.In that range temperatures swing a lot and it makes no sense. Set it 0 or to a temperature that can extrude and therefore does not swing that badly.

immunity20, 2016-02-16:
I have removed from gcode file the T1 (temp , wait for temp) and it seems to work OK.

Do you have any gcode script for cleaning nozzles before start the printing ? (2.0a DUO )

repetier, 2016-02-16:
I have no DaVinci, so no cleaning scripts. All I do is printing a skirt and maybe use my tweezers.

immunity20, 2016-02-16:
question was for Luc :P (he has a 2.0 DUO)

luc-github, 2016-02-16:
@immunity20 I think you use wrong issue tracker - I did not realize earlier, sorry 
for cleaning nozzles on Davinci there is custom GCODE: https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/src/ArduinoDUE/Repetier/Repetier.ino#L93

immunity20, 2016-02-16:
oh i realize it too (trying to find the post through 0.92 issues panel but there wasn't).
Thanks Luc for the tip (M100)