repetier/RepetierFirmware #475: Z_HOME Position only for HOME_ORDER_ZXYTZ?

locomarco, 2016-02-18:
Does Z_HOME_X_POS and Z_HOME_Y_POS only affect homing position when HOME_ORDER_ZXYTZ is used?

The explanation doesn't clarify that:
// If your bed might bend while probing, because your sensor is the extruder tip
// you can define a predefined x,y position so beding is always the same and
// can be compensated.

Currently i have set up my printer to use the hotend tip as probe, but it needs to touch the bed at a specific position.
Home order ZXYTZ wouldn't be useful for me, as the hotend needs to be at this position before first touch.

In general ZXYTZ doesn't make any sense to me. Why not just XYTZ? The first probe seems useless to me, if it isn't precise anyway.

repetier, 2016-02-18:
This was made exactly for your case. We want to go up a bit for heating so temp. sensitive beds with e.g. BuildTak do not get destroyed by having a hot nozzle on the same spot. So first z move is just to find coarse z=0 and the go up to z height for heating. The second z probe is done at your specific position for highest accuracy and then goes to homing position to wait for next commands.

The position is only activated for that homing order.

locomarco, 2016-02-18:
Okay, that makes sense. I wasn't thinking about heat sensitive beds.

But it's not exactly for my case, as i cannot home the z-axis regardless of x and y position.
The hotend needs to be somewhere near my piezo sensor, otherwise the bed just crashes into the hotend.
What would work for me is X -> Y  -> go to positon -> Z (-> T -> Z).

repetier, 2016-02-19:
You can omit the first home if you are sure never to start at top

#elif HOMING_ORDER == HOME_ORDER_ZXYTZ
{
    float actTemp[NUM_EXTRUDER];
    for(int i = 0;i < NUM_EXTRUDER; i++)
        actTemp[i] = extruder[i].tempControl.targetTemperatureC;
    if(zaxis) {
        homeZAxis();

remove the homeZAxis(); there it is the first one. Next line would be to go up a bit.

repetier, 2016-02-19:
For the next update I have added HOME_ORDER_XYTZ which then would be what you want.

locomarco, 2016-02-19:
Thanks for the hint. I'll try duplicating your ZXYTZ code and changing it to XYZTZ.